pyvisa/pyvisa-py #131: ValueError: The device has no langid

greyltc, 2018-04-04:
```
In [1]: import visa

In [2]: rm = visa.ResourceManager('@py')

In [3]: rm.list_resources()
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-defa3779928e> in <module>()
----> 1 rm.list_resources()

/usr/lib/python3.6/site-packages/pyvisa/highlevel.py in list_resources(self, query)
   1569         """
   1570 
-> 1571         return self.visalib.list_resources(self.session, query)
   1572 
   1573     def list_resources_info(self, query='?*::INSTR'):

/usr/lib/python3.6/site-packages/pyvisa-py/highlevel.py in list_resources(self, session, query)
    229 
    230         resources = sum([st.list_resources()
--> 231                          for key, st in sessions.Session.iter_valid_session_classes()], [])
    232 
    233         resources = rname.filter(resources, query)

/usr/lib/python3.6/site-packages/pyvisa-py/highlevel.py in <listcomp>(.0)
    229 
    230         resources = sum([st.list_resources()
--> 231                          for key, st in sessions.Session.iter_valid_session_classes()], [])
    232 
    233         resources = rname.filter(resources, query)

/usr/lib/python3.6/site-packages/pyvisa-py/usb.py in list_resources()
    201                                   manufacturer_id=dev.idVendor,
    202                                   model_code=dev.idProduct,
--> 203                                   serial_number=dev.serial_number,
    204                                   usb_interface_number=intfc))
    205         return out

/usr/lib/python3.6/site-packages/usb/core.py in serial_number(self)
    828         """
    829         if self._serial_number is None:
--> 830             self._serial_number = util.get_string(self, self.iSerialNumber)
    831         return self._serial_number
    832 

/usr/lib/python3.6/site-packages/usb/util.py in get_string(dev, index, langid)
    312 
    313     if 0 == len(langids):
--> 314         raise ValueError("The device has no langid")
    315     if langid is None:
    316         langid = langids[0]

ValueError: The device has no langid
```
As per https://github.com/pyvisa/pyvisa-py/issues/84, this happens when a user doesn't have full permissions on some USB device. I think it's quite rare that any user besides root has the permissions required for `rm.list_resources()` to complete without error.

Can this easily be changed to fail more gracefully? Maybe throwing a warning for USB devices without the required permissions instead of going down in flames? @hgrecco, might you take a PR from me if I tried to fix this?

MatthieuDartiailh, 2018-04-04:
I believe that latest master already PR #126. Could you check ?

greyltc, 2018-04-05:
It doesn't appear to be solved in master:
```
$ pyvisa-shell -b py

Welcome to the VISA shell. Type help or ? to list commands.

(visa) list
The device has no langid
(visa) exit
```

MatthieuDartiailh, 2018-04-05:
Looking at the master branch we only catch `NotImplementedError` but not `ValueError`. Could you post the full traceback on master and make a PR adding `ValueError` if necessary ?

greyltc, 2018-04-07:
The full traceback is here: https://github.com/pyvisa/pyvisa-py/issues/131#issue-311160018

This should be fixed by https://github.com/pyvisa/pyvisa-py/pull/134

closed 2018-04-07 by bors[bot]