pyvisa/pyvisa-py #138: The last time we had a release tag was on Aug 25, 2015

greyltc, 2018-04-08:


MatthieuDartiailh, 2018-04-09:
I am well aware of this situation and I was planning to release a month ago. But https://github.com/pyvisa/pyvisa/issues/319 popped up and I am trying to fix before releasing. As I cannot reproduce locally it takes a bit more time than I would have liked but we will get there. Once it is fixed I will release quickly.

greyltc, 2018-04-10:
Cool.
I'll try to find some time to help fix https://github.com/pyvisa/pyvisa/issues/319
I should have some equipment available that I can use to see the error myself.

awattenhofer, 2018-06-05:
A new release is absolutely necessary.   I am using pyvisa-py to interface with an Agilent DMM over USBTMC on Raspbian.  A change was made to the UCBTMC.py between the last release and master that fixed a bug that I think screwed up closing the resource.  I was able to communicate with the device once per power cycle before usbtmc.py was throwing all kinds of errors.  It has since been fixed and is working quite well with the latest master.  

I've sifted through the source code enough, if any help is needed with bugs preventing a new release, I can try to assist.

MatthieuDartiailh, 2018-06-05:
The one thing holding the release for now is #139. I did not yet find time to test it. Once tested and merged I will do my best to release within a week.

MatthieuDartiailh, 2018-08-17:
The next release is nearly ready. But I would like a second pair of eyes on the last changes and if somebody else can test the content of the last PR it would be great. In the absence of feedback I will cut the release around the end of the month.

greyltc, 2020-03-06:
The last time we had a release tag was on Sep 6, 2018 ;-)

MatthieuDartiailh, 2020-03-06:
My bad. I am never really satisfied with how much I can test this library and as a consequence I am always pushing back releases... I will cut a release in a week or two at most. Ping me if I forget.

greyltc, 2020-03-06:
Cool. Thanks!