pyvisa/pyvisa-py #143: USB Timeout TypeError

svenuniku, 2018-04-30:
Hi,

when I am running a keysight scope via usb and the master branch of pyvisa-py, I'm running in a Exception because the timeout type is a float instead of an unsigned int:
`ctypes.ArgumentError: argument 6: <class 'TypeError'>: wrong type`
in 
>   File "/usr/local/lib/python3.6/dist-packages/usb/backend/libusb1.py", line 933, in __read

in pyvisa I'm setting the timeout to 12345 ms. The libusb is awaiting an integer in ms as timeout value[1].
but pyvisa-py is setting the timeout magically in seconds and in floats[2].

my hack at the moment is to overwrite the Line 80[2] with:
`self.interface.timeout = int(value)`
This then sets the current timeout as an integer in ms. I think this is not be clean solution. Hopefully somebody who is more experienced in pyvisa-py can fix this issue.

[1] [https://github.com/pyusb/pyusb/blob/master/usb/core.py#L716](https://github.com/pyusb/pyusb/blob/master/usb/core.py#L716)
[2] [https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/usb.py#L80](https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/usb.py#L80)
My system is running on debian testing.
usb.__version__ = '1.0.2' from debian testing repo
visa.__version__ = '1.10.0.dev0' directly from github

MatthieuDartiailh, 2018-04-30:
The handling of timeout is a bit inconsistent within pyvisa-py partially because support for it on some interfaces was added only recently. I will try to fix this issue before the next release which I plan for next week. I will let you know when you can test again.

svenuniku, 2018-05-03:
The Fix #144 seems to work for the usb timeout. I didn't got timeouts any more and no TypeErrors.

commit 7d876c5e (2018-05-24): Merge #130 #144

130: Attempt to fix an issue in rpc in which a select timeout cause a EOFError r=MatthieuDartiailh a=MatthieuDartiailh

The basic idea is that compared to the old version (pre PR 67), we can now get an empty read_data because the select timed out, which by itself may no mean anything, and we need to wait to see if we get a real timeout rather than complaining about EOFError early.

@svanan77 could you review and test. Hopefully this should fix https://github.com/pyvisa/pyvisa/issues/319 and given the nature of the issue, it is possible I could not easily reproduce locally.

14 [... cut]
PR #144 (2018-05-24): sessions: improve the handling of timeout (and fix it on usb)
 Fixes #143

@svenuniku could check it does fix your issue ?

@greyltc @tivek if you have time to review/test, please feel free to do so.
closed 2018-05-24 by bors[bot]