pyvisa/pyvisa-py #145: Owon SDS7102: Even simplest command hangs indefinitely

JohnCC330, 2018-05-12:
I'm a recent owner of an SDS7102E. LAN communication has resulted *very* difficult. I now installed pyvisa-py, and can ran these instructions from the python prompt:

`import visa
rm = visa.ResourceManager('@py')
scope = rm.open_resource('TCPIP0::192.168.9.120::3000::SOCKET')
scope.query('*IDN?')
`

Looking at wireshark's logging, I see that the instruction (*IDN?) really appears, but it is followed by CR and LF (DOS line ending). If I do this on a home-brew (non-pyvisa) test program (i.e. adding \r\n), the command isn't recognized either.

I have the impression that the SDS7102 LAN implementation is quite flaky. 

Still, is it possible to disable the line-endings in pyvisa[-py]?

MatthieuDartiailh, 2018-05-12:
You can set the write_termination (read_termination) to an empty string.

JohnCC330, 2018-05-12:
Thanks Matthieu! It doesn't seem to make any difference - no reply. I'll find out why later... Have another job to finish first.

Is there any documentation which reflects pyVISA's API? Consult VISA's docs?

MatthieuDartiailh, 2018-05-12:
Pyvisa is documented on readthedocs. I would also suggest to check your instrument manual about the message format. Good luck

JohnCC330, 2018-05-15:
Thanks for the links, Matthieu. I visited pyvisa-py on readthedocs multiple times, but didn't realize the docs were really on pyvisa... my apologies!

The instrument manual on the other hand, only describes the instrument itself. I had already discovered an SCPI manual from Owon, which describes the syntax. And yet another manual (mostly unreadable) which describes communication over USB. It has one line which states that TCP/IP communication is the same as the USB protocol. With possibilities of TCP packets being fragmented and concatenated, it is not the same as USB..

Eg. the scope returns scope channel sensitivity (`:CHAN1:SCAL?`) as `5.00mV`. How do I detect the end of the string? `.query_ascii_values` hangs, as I cannot define a packet termination - there is none - and the string length can vary (eg. `1V` also occurs). According to [examples from Keysight](https://www.keysight.com/main/editorial.jspx?cc=CM&lc=eng&ckey=2798637&nid=-11143.0.00&id=2798637), termination on TCPIP would normally be a `\n`, but Wireshark doesn't show any newlines.

I tried 'V' as terminator, but that causes a float error on 200mV (which is detected as 200m). I tried to add a converter 'callable', but that never gets called (found no examples of use of callebles).

MatthieuDartiailh, 2018-05-15:
I will try to look at tge docs you provide tomorrow. To try to debug thus further, you can use the `read_bytes` to read only a fixed number of bytes. However it us true that usually instrument should send a termination character.

MatthieuDartiailh, 2018-05-16:
Looking at the scpi documentation of your instrument (page 6) it is mentionned that to use SCPI with something that is not the company software one must first send ':SDSLSCPI#' and 'SCPION' as an answer. Could you try that ?

JohnCC330, 2018-05-16:
Of course I did that. Here's the code I wrote to detect if the scope had SCPI enabled or, if not, enable it:

    def enter_SCPI_TCP(self):
        try: # First check if already in SCPI. *ESE? doesn't reply w/o SCPI
            self.socket.send(b"*ESE?")
            r = self.socket.recv(BFF_SIZE)
            print("Was already in SCPI mode. Clearing registers...")

        except self.timeout: # Was not in SCPI mode, so enter now
            self.send(b":SDSLSCPI#\n")
            r = self.socket.recv(BFF_SIZE)
            print("Entered SCPI mode")

This works fine.

MatthieuDartiailh, 2018-05-17:
In that case I still think that your best shot is to read one answer one byte at a time to get the termination character out. Wireshark is nice but because of the VXI11 the dara you get can be cryptic.

JohnCC330, 2018-05-17:
On Wed, 16 May 2018 22:34:26 -0700
Matthieu Dartiailh <notifications@github.com> wrote:

> In that case I still think that your best shot is to read one answer
> one byte at a time to get the termination character out. Wireshark is
> nice but because of the VXI11 the data you get can be cryptic.

Matthieu, why do you mention VXI11? As far as I know, the scope does not
use VXI - there is no mention in the manuals. In fact, I don't think the
scope is capable of VXI.

John

MatthieuDartiailh, 2018-05-17:
Sorry, I posted that quickly and forgot your instrument was a simple socket instrument ....
Could you post the wireshark capture or a byte per byte reading of the instrument answer so that we can try to figure out the termination character.

JohnCC330, 2018-05-18:
On Thu, 17 May 2018 14:34:30 +0000 (UTC)
Matthieu Dartiailh <notifications@github.com> wrote:

> Sorry, I posted that quickly and forgot your instrument was a simple
> socket instrument .... Could you post the wireshark capture or a byte
> per byte reading of the instrument answer so that we can try to
> figure out the termination character.

pysocket_connect_and_commands.pcapng :

	This capture is from my first (non-pyVISA) python program just using
	the native python socket library. There are several - critical - delays
	in the program to get this to work. Eg. before each command, a 300ms
	delay has to be inserted. Even so, at line 32, the scope does not
	reply at all.

Lines	What:
1-3	Connection PC -> scope
4	PC sends *ESE? to see if already in SCPI mode
5	Scope answers with 0x34, so it is already in SCPI
6	ACK the reply
7	PC sends *CLS to reset state variables
8	but needs retransmission for scope to ACK
9	ACK confirming CLS (#7)
10	Get the trigger mode:  :TRIG:MODE?
11	Scope answers          AUTO
12	ACK
13	Get the ACQ mode       :ACQ:TYPE?
14	Scope ACKs and says    SAMPle
15	ACK that
16	Get the memory depth   :ACQ:MDEP?
17	Scope ACK and reply    10K
18	ACK that
19	Get averaging count    :ACQ:AVER?
20      Scope ACK and reply    4
21	ACK that
22	Get Chan2 sensitivity  :CHAN2:SCAL?
23	Scope ACKs and says    500mV
24	ACK that
25	Get Chan2 coupling     :CHAN2:COUP?
26	Scope ACKs and says    DC
27	ACK that
28	Get Chan2 probe atten  :CHAN2:PROB?
29	Scope ACKs and says    10X
30	ACK that
31	Get Chan1 sensitivity  :CHAN1:SCAL?
32	No ACK from scope: resend
33	ACK from scope but no answer!
34-	After timeout, disconnect

visa_connect_and_command

	This was a simple test to check if the pyVISA library could solve the
	problems above.

3-5	Connection to the scope
6	Ask for the memory depth        :ACQ:MDEP?
7	ACK and Reply from scope        10K
8	ACK the reply

As you can see in both captures, there is no termination character.

John

MatthieuDartiailh, 2018-05-18:
Not sure where the capture/scripts are ? Where can I find them ?

JohnCC330, 2018-05-18:
On Fri, 18 May 2018 07:59:06 -0700
Matthieu Dartiailh <notifications@github.com> wrote:

> Not sure where the capture/scripts are ? Where can I find them ?

The previous message I sent has two attachments with the two files - at least 
here the message is marked with two attachments. The have to be read
by Wireshark to show. I'll attach them again. Do you need anything else?

John

MatthieuDartiailh, 2018-05-19:
Sorry but Github does not show the attachments. You will have to either copy paste the logs or put them in a gist.

JohnCC330, 2018-05-19:
On Sat, 19 May 2018 02:30:55 -0700
Matthieu Dartiailh <notifications@github.com> wrote:

> Sorry but Github does not show the attachments. You will have to
> either copy paste the logs or put them in a gist. 

Oh Ok... I used the reply link at the bottom of the messages, so I didn't see
Github throwing them away. I'll upload them now.

John

JohnCC330, 2018-05-19:
[pysocket_connect_and_commands.pcapng.gz](https://github.com/pyvisa/pyvisa-py/files/2019496/pysocket_connect_and_commands.pcapng.gz)
[visa_connect_and_command.pcapng.gz](https://github.com/pyvisa/pyvisa-py/files/2019497/visa_connect_and_command.pcapng.gz)

JohnCC330, 2018-05-19:
On Sat, 19 May 2018 02:30:55 -0700
Matthieu Dartiailh <notifications@github.com> wrote:

> Sorry but Github does not show the attachments.

Was that because of the extension (.pcapng)? Testing with a gzipped file of same.