pyvisa/pyvisa-py #148: Write to a serial instrument

clavay, 2018-06-28:
Hi,

I have an issue to send a write with a serial instrument. (robot to make it move)

When I try this it works :
import serial
ser = serial.Serial(port='/dev/ttyS0', baudrate=9600, parity=serial.PARITY_NONE, stopbits=serial.STOPBITS_ONE, bytesize=serial.EIGHTBITS)
ser.write('B+100')
ser.close()
-> The robot moves.

But this doesn't work : 
import visa
rm = visa.ResourceManager('@py')
list = rm.list_resources()
my_inst = rm.open_resource(list[0], baud_rate=9600, data_bits=8, parity=visa.constants.Parity.none, stop_bits=visa.constants.StopBits.one)
my_inst.write('B+100')

-> I have this output : (7, <StatusCode.success: 0>)
-> and list say : (u'ASRL/dev/ttyS0::INSTR',)
-> But nothing append.

python -m visa info : 
Machine Details:
   Platform ID:    Linux-4.9.0-6-amd64-x86_64-with-debian-9.4
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.13
   Compiler:       GCC 6.3.0 20170516
   Bits:           64bit
   Build:          Nov 24 2017 17:33:09 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

If you have any idea of to control serial instrument with pyvisa-py as using pyserial...

cla

MatthieuDartiailh, 2018-06-28:
Pyvisa does not close the connection to your robot after you write to it, which from your pyserial example is required for your robot to move. Try calling the `my_inst.close` method. Pyvisa assumes that the instrument expect a particular character (`write_temination`) to consider a command complete and execute it, which the most common case for VISA instrument, which is why it does not close the connection.

clavay, 2018-06-28:
The robot moves before the close. It is not necessary, I put it only to close the connection before to try it with pyvisa.

clavay, 2018-06-29:
It's ok with this code : 

import visa
rm = visa.ResourceManager('@py')
list = rm.list_resources()
my_inst = rm.open_resource(list[0], baud_rate=9600, data_bits=8, parity=visa.constants.Parity.none, stop_bits=visa.constants.StopBits.one, write_termination='')
my_inst.write('B+100')

Thank you for your help.

closed 2018-06-29 by clavay

MatthieuDartiailh, 2018-06-29:
I am happy you found a solution !