pyvisa/pyvisa-py #149: Having Difficulty Reading from TCIP socket

AlteredWildebeest, 2018-06-29:
Hello,
First of all I apologize, I'm a new user to this library so I may be making a silly blunder, if so please kindly point it out to me. I'm trying to use the pyvisa-py library to communicate with my Siglent instruments over Ethernet. The instrument in this case is the Siglent SDG2042 function generator. It seems I'm able to open a socket with an instrument, but I dont think I'm getting the correct response. Some basic info (using ubuntu pc):
$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.4.71-ELAR-Systems-armv7l-with-Ubuntu-18.04-bionic
   Processor:      armv7l

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.15rc1
   Compiler:       GCC 7.3.0
   Bits:           32bit
   Build:          Apr 15 2018 21:51:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

$python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> import sys
>>> print (sys.path)
['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-arm-linux-gnueabihf', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/tinker/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/gtk-2.0']
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-06-28 22:58:34,454 - pyvisa - DEBUG - Reusing ResourceManager with session 3730723
>>> device = rm.open_resource("TCPIP::192.168.250.110::23::SOCKET")
2018-06-28 22:58:54,199 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - opening ...
2018-06-28 22:58:54,205 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - is open with session 1282565

When I issue this command, immediately the line Status Code.success is returned, then waits about 1 min before printing the traceback:
>>> print(device.query("*IDN?"))
2018-06-28 22:59:24,031 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)

When issuing the below command adding the newline character, the success line is returned immediately:
>>> print(device.query("*IDN?\n"))
2018-06-28 23:01:51,649 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

>>> 
Reading the Siglent manual, the response to the *IDN? command is supposed to be in the form:
*IDN ,<device id>,<model>,<serial number>,<software version>,<firmware version>
As an example in the Siglent manual is this:
*IDN SDG SDG1022,00-00-00-13-22,1.01.01.10R1,20.234.3.

Is there a mistake I'm making somewhere? Am I not reading the response correctly? Also, I'm new to python, but a veteran C writer, so I could be making a python error as well.

Thanks for any help you can provide. Cheers

AlteredWildebeest, 2018-06-29:
Also, sometimes the device thinks I'm trying to telnet into it:
>>> print(device.query("*IDN?\n"))
2018-06-28 23:21:49,317 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

Login incorrect
(none) login: 
Login timed out after 60 seconds

>>> 
_______________________________________
Did a port scan of the device using nmap:
$ nmap 192.168.250.110

Starting Nmap 7.60 ( https://nmap.org ) at 2018-06-28 23:27 PDT
Nmap scan report for 192.168.250.110
Host is up (0.00038s latency).
Not shown: 998 closed ports
PORT    STATE SERVICE
23/tcp  open  telnet
111/tcp open  rpcbind

Nmap done: 1 IP address (1 host up) scanned in 0.20 seconds
________________________________________
Then tried opening on a different socket, but receive the same results:
>>> device = rm.open_resource("TCPIP::192.168.250.110::111::SOCKET")
2018-06-28 23:23:41,412 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - opening ...
2018-06-28 23:23:41,413 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - is open with session 6963028
>>> print(device.query("*IDN?\n"))
2018-06-28 23:23:45,029 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

>>> 
________________________________________________

If attempting to open the device as an instrument, this is the output:
>>> device = rm.open_resource("TCPIP::192.168.250.110::23::INSTR")
2018-06-28 23:25:46,207 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::INSTR - opening ...
2018-06-28 23:25:46,207 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 111)
2018-06-28 23:25:46,208 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa-py.protocols.rpc.PortMapperPacker instance at 0xb512c0d0>>, <bound method PortMapperUnpacker.unpack_uint of <pyvisa-py.protocols.rpc.PortMapperUnpacker instance at 0xb512c198>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 868, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 741, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0x86 in position 14: ordinal not in range(128)
Logged from file rpc.py, line 296
2018-06-28 23:25:46,210 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb50c6458>: '\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xb5'
2018-06-28 23:25:46,211 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-06-28 23:25:46,211 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 693)
2018-06-28 23:25:46,212 - pyvisa - DEBUG - Make call 10, (2000378720L, 0, 10000, u'23'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb50c1fa8>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb50c1f58>>
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 868, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 741, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0xaf in position 15: ordinal not in range(128)
Logged from file rpc.py, line 296
2018-06-28 23:25:46,214 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb50c6420>: '\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00'
2018-06-28 23:25:46,215 - pyvisa - DEBUG - TCPIP0::192.168.250.110::23::INSTR - closing
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1655, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1612, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 191, in open
    sess = cls(session, resource_name, parsed)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 184, in __init__
    self.after_parsing()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 62, in after_parsing
    raise Exception("error creating link: %d" % error)
Exception: error creating link: 3

MatthieuDartiailh, 2018-06-29:
Could you try to update to the current master branch ? A number of issues have been fixed since 0.2 was released (hopefully I will be able to relrase 0.3 soon).

AlteredWildebeest, 2018-06-30:
Thank you for your assistance Matthieu,
I upgraded using this command:
$ pip install -U https://github.com/hgrecco/pyvisa-py/zipball/master

Then, running python, I received an error when trying to import visa:
$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import socket
>>> import visa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/__init__.py", line 18, in <module>
    import pkg_resources
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 36, in <module>
    import email.parser
  File "/usr/lib/python2.7/email/parser.py", line 12, in <module>
    from email.feedparser import FeedParser
  File "/usr/lib/python2.7/email/feedparser.py", line 27, in <module>
    from email import message
  File "/usr/lib/python2.7/email/message.py", line 16, in <module>
    import email.charset
  File "/usr/lib/python2.7/email/charset.py", line 13, in <module>
    import email.base64mime
  File "/usr/lib/python2.7/email/base64mime.py", line 40, in <module>
    from email.utils import fix_eols
  File "/usr/lib/python2.7/email/utils.py", line 29, in <module>
    import urllib
  File "/usr/lib/python2.7/urllib.py", line 862, in <module>
    class ftpwrapper:
  File "/usr/lib/python2.7/urllib.py", line 866, in ftpwrapper
    timeout=socket._GLOBAL_DEFAULT_TIMEOUT,
AttributeError: 'module' object has no attribute '_GLOBAL_DEFAULT_TIMEOUT'
>>> 

What should I do next?

AlteredWildebeest, 2018-06-30:
This error also occurs if I dont import socket first:
$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/__init__.py", line 18, in <module>
    import pkg_resources
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 36, in <module>
    import email.parser
  File "/usr/lib/python2.7/email/parser.py", line 12, in <module>
    from email.feedparser import FeedParser
  File "/usr/lib/python2.7/email/feedparser.py", line 27, in <module>
    from email import message
  File "/usr/lib/python2.7/email/message.py", line 16, in <module>
    import email.charset
  File "/usr/lib/python2.7/email/charset.py", line 13, in <module>
    import email.base64mime
  File "/usr/lib/python2.7/email/base64mime.py", line 40, in <module>
    from email.utils import fix_eols
  File "/usr/lib/python2.7/email/utils.py", line 29, in <module>
    import urllib
  File "/usr/lib/python2.7/urllib.py", line 862, in <module>
    class ftpwrapper:
  File "/usr/lib/python2.7/urllib.py", line 866, in ftpwrapper
    timeout=socket._GLOBAL_DEFAULT_TIMEOUT,
AttributeError: 'module' object has no attribute '_GLOBAL_DEFAULT_TIMEOUT'
>>>

AlteredWildebeest, 2018-06-30:
Ok I was able to import visa after rebooting:
$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.4.71-ELAR-Systems-armv7l-with-Ubuntu-18.04-bionic
   Processor:      armv7l

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.15rc1
   Compiler:       GCC 7.3.0
   Bits:           32bit
   Build:          Apr 15 2018 21:51:34 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

Now when I attempt to run the commands, this is the output received:
>>> import visa
>>> import sys
>>> import binascii
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-06-29 23:00:11,299 - pyvisa - DEBUG - Reusing ResourceManager with session 1173664
2018-06-29 23:00:11,299 - pyvisa - DEBUG - Reusing ResourceManager with session 1173664
>>> dmm = rm.open_resource('TCPIP::192.168.250.110::23::SOCKET')
2018-06-29 23:02:44,797 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - opening ...
2018-06-29 23:02:44,797 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - opening ...
2018-06-29 23:02:44,799 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - is open with session 6535500
2018-06-29 23:02:44,799 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - is open with session 6535500
>>> print(dmm.query("*IDN?\n"))
2018-06-29 23:02:48,948 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-06-29 23:02:48,948 - pyvisa - DEBUG - TCPIP::192.168.250.110::23::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-06-29 23:03:44,843 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - closing
2018-06-29 23:03:44,843 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - closing
2018-06-29 23:03:44,844 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - is closed
2018-06-29 23:03:44,844 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - is closed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xff in position 0: ordinal not in range(128)
>>>

AlteredWildebeest, 2018-06-30:
UPDATE:
I was able to verify the underlying network connection works by connecting to the socket directly, using the following commands I found in Siglent tutorial:

$ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import socket
>>> s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> ip = "192.168.250.110"
>>> port = 5024
>>> s.connect((ip, port))
>>> s.sendall(b'*IDN?')
>>> s.sendall(b'\n')
>>> print s.recv(4096)
Welcome to the SCPI instrument 'Siglent SDG2042X'
>>Siglent Technologies,SDG2042X,SDG2XCAQ1R1528,2.01.01.23R5
>>

AlteredWildebeest, 2018-06-30:
Here's another error I often receive after the tring of commands:
tinker@ELAR-Systems:~/Documents/SCP/scripts$ python2.7
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> import time
>>> import binascii
>>> import sys
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-06-30 00:49:40,781 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-06-30 00:49:40,827 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-06-30 00:49:40,842 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-06-30 00:49:40,844 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-06-30 00:49:40,844 - pyvisa - DEBUG - Created library wrapper for unset
2018-06-30 00:49:40,845 - pyvisa - DEBUG - Created ResourceManager with session 5994359
>>> device = rm.open_resource("TCPIP::192.168.250.110::111::SOCKET")
2018-06-30 00:49:56,759 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - opening ...
2018-06-30 00:49:56,766 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - is open with session 4810401
>>> device.term_chars = "\n"
>>> print(device.query("*IDN?"))
2018-06-30 00:50:32,690 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-06-30 00:50:34,703 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 323, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>>

MatthieuDartiailh, 2018-07-01:
Where did you find the connection info you are using ? Port 23 is reserved for telnet so it cannot work for a raw VISA socket. 111 if for RPC which in the context of VISA most likely means that you should attempt to connect using the INSTR protocol of VISA not the socket one. I believe you should always use 5024 for RAW socket VISA communication..
Another thing that appears off is that term_chars does not exist anymore (deprecated in 1.5). You can now send independently the read and write termination using the  `read_termination` and `write_termination` attributes.

AlteredWildebeest, 2018-07-01:
Hi Mathieu,
Thank you for your very good information. I stopped using the term_chars and instead inserted \n into the command string. I tried as you suggested, and changed to port 5024. I still seem to be receiving some type of timeout error:
python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> import time
>>> import sys
>>> import binascii
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-07-01 09:51:59,718 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-07-01 09:51:59,766 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-07-01 09:51:59,788 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-07-01 09:51:59,789 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-07-01 09:51:59,790 - pyvisa - DEBUG - Created library wrapper for unset
2018-07-01 09:51:59,790 - pyvisa - DEBUG - Created ResourceManager with session 2155461
>>> device = rm.open_resource("TCPIP::192.168.250.110::5024::SOCKET")
2018-07-01 09:52:43,001 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - opening ...
2018-07-01 09:52:43,003 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - is open with session 5837562
>>> print(device.query("*IDN?"))
2018-07-01 09:53:04,587 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-01 09:53:06,604 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
2018-07-01 09:53:06,604 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::INSTR - closing
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 323, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> print(device.query("*IDN?\n"))
2018-07-01 09:53:38,664 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-01 09:53:40,679 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 323, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

AlteredWildebeest, 2018-07-01:
It is odd though, after sending the command I'll immediately receive a response, followed by a couple second wait, then the timeout error:
>>> print(device.query("*IDN?\n"))
2018-07-01 10:01:08,504 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
[Waits a couple seconds]
[Then delivers the timeout error]

AlteredWildebeest, 2018-07-01:
I was using port 23 and 111 due to my own lack of knowledge. I saw them as the only open ports using nmap. Fo some reason nmap did not detect port 5024 as being open

AlteredWildebeest, 2018-07-01:
If attempting to use port 111 as INSTR, this is the response I receive:
>>> device = rm.open_resource("TCPIP::192.168.250.110::111::INSTR")
2018-07-01 10:04:49,615 - pyvisa - DEBUG - TCPIP::192.168.250.110::111::INSTR - opening ...
2018-07-01 10:04:49,615 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 111)
2018-07-01 10:04:49,617 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa-py.protocols.rpc.PortMapperPacker instance at 0xb526b0a8>>, <bound method PortMapperUnpacker.unpack_uint of <pyvisa-py.protocols.rpc.PortMapperUnpacker instance at 0xb526b0f8>>
2018-07-01 10:04:49,617 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb525c8b8>: '\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2018-07-01 10:04:49,619 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb525c8b8>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xb5')
2018-07-01 10:04:49,619 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-07-01 10:04:49,619 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 693)
2018-07-01 10:04:49,620 - pyvisa - DEBUG - Make call 10, (1166704935L, 0, 10000, u'111'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb5260c60>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb5260c10>>
2018-07-01 10:04:49,621 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb525c8f0>: "\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00E\x8a\x81'\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x03111\x00"
2018-07-01 10:04:49,622 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb525c8f0>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1655, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1612, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 193, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 209, in __init__
    self.after_parsing()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 56, in after_parsing
    raise Exception("error creating link: %d" % error)
Exception: error creating link: 3
>>>

MatthieuDartiailh, 2018-07-01:
PyVISA tries to read till it encounter the termination character, so you should set it using read_termination (note also that using write_termination is less error prone that manually adding it). If you do not know the termination character used by your instrument you can try to use the `read_bytes` method to read a fix number of bytes and identify the term char.
When using INSTR, you SHOULD not pass the port simply used TCPIP::192.168.250.110::INSTR.

AlteredWildebeest, 2018-07-01:
Thank you very much Matthieu, I see improvement! Now I can get part of the response, but no the whole thing. This is what I see:
>>> import visa
>>> import time
>>> import binascii
>>> import sys
>>> visa.log_to_screen()
>>> rm=visa.ResourceManafer('@py')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'ResourceManafer'
>>> rm=visa.ResourceManager('@py')
2018-07-01 13:55:13,162 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-07-01 13:55:13,211 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-07-01 13:55:13,228 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-07-01 13:55:13,229 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-07-01 13:55:13,230 - pyvisa - DEBUG - Created library wrapper for unset
2018-07-01 13:55:13,230 - pyvisa - DEBUG - Created ResourceManager with session 5906780
>>> device = rm.open_resource("TCPIP::192.168.250.110::5024::SOCKET")
2018-07-01 13:55:29,797 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - opening ...
2018-07-01 13:55:29,805 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - is open with session 6964008
>>> device.read_termination='\n'
>>> print(device.query("*IDN?"))
2018-07-01 13:55:56,604 - pyvisa - DEBUG - TCPIP::192.168.250.110::5024::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
**Welcome to the SCPI instrument 'Siglent SDG2042X'**

If I do the query via the socket interface I receive this response:
**Welcome to the SCPI instrument 'Siglent SDG2042X'
Siglent Technologies,SDG2042X,SDG2XCAQ1R1528,2.01.01.23R5**

What can I do to read the whole response?

AlteredWildebeest, 2018-07-01:
Also I'm not familiar with the read_bytes method, what is the syntax for it?

I tried what you suggested for the INSTR interface, not specifying a port, and it ends up trying to open port 111 anyways, then opens port 693:2018-07-01 14:07:38,070 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - opening ...
2018-07-01 14:07:38,071 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 111)
2018-07-01 14:07:38,079 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa-py.protocols.rpc.PortMapperPacker instance at 0xb5259738>>, <bound method PortMapperUnpacker.unpack_uint of <pyvisa-py.protocols.rpc.PortMapperUnpacker instance at 0xb5b45b20>>
2018-07-01 14:07:38,081 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb5b24688>: '\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2018-07-01 14:07:38,083 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5b24688>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xb5')
2018-07-01 14:07:38,084 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-07-01 14:07:38,086 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 693)
2018-07-01 14:07:38,087 - pyvisa - DEBUG - Make call 10, (507220917L, 0, 10000, u'inst0'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb5c1df58>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb5b455a8>>
2018-07-01 14:07:38,089 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb52567a0>: "\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1e;\x93\xb5\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x05inst0\x00\x00\x00"
2018-07-01 14:07:38,091 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb52567a0>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00')
2018-07-01 14:07:38,092 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - is open with session 5640064
2018-07-01 14:07:38,143 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - closing
2018-07-01 14:07:38,144 - pyvisa - DEBUG - Make call 23, 0, <bound method Vxi11Packer.pack_device_link of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb5c1df58>>, <bound method Vxi11Unpacker.unpack_device_error of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb5b455a8>>
2018-07-01 14:07:38,145 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb52567a0>: '\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x17\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2018-07-01 14:07:38,146 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb52567a0>: bytearray(b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2018-07-01 14:07:38,147 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-07-01 14:07:38,147 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - is closed

2018-07-01 14:05:50,361 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb5b597d8>: "\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x88\x00\x00\x00\x07*IDN?\r\n\x00"
2018-07-01 14:05:50,362 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5b597d8>: bytearray(b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07')
2018-07-01 14:05:50,363 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "./mathue.py", line 13, in <module>
    print(device.query("*IDN?"))
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 318, in read
    ret = self.sessions[session].read(count)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 105, in read
    start_time = time.monotonic()
AttributeError: 'module' object has no attribute 'monotonic'
2018-07-01 14:05:50,421 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - closing
[... cut ...]
2018-07-01 14:05:50,423 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb5b597d8>: '\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x17\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
2018-07-01 14:05:50,424 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5b597d8>: bytearray(b'\x00\x00\x00\x03\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2018-07-01 14:05:50,424 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-07-01 14:05:50,425 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - is closed

AlteredWildebeest, 2018-07-01:
I'm not familiar with read_bytes, what is the syntax for it?

Using the INSTR interface without specifying a port still fails:
>>> import visa
>>> import time
>>> import binascii
>>> import sys
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2018-07-01 14:11:17,373 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-07-01 14:11:17,422 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-07-01 14:11:17,443 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-07-01 14:11:17,444 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-07-01 14:11:17,444 - pyvisa - DEBUG - Created library wrapper for unset
2018-07-01 14:11:17,445 - pyvisa - DEBUG - Created ResourceManager with session 6773778
>>> device = rm.open_resource("TCPIP::192.168.250.110::INSTR")
2018-07-01 14:11:17,446 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - opening ...
2018-07-01 14:11:17,447 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 111)
2018-07-01 14:11:17,451 - pyvisa - DEBUG - Make call <PortMapperVersion.get_port: 3>, (395183, 1, 6, 0), <bound method PortMapperPacker.pack_mapping of <pyvisa-py.protocols.rpc.PortMapperPacker instance at 0xb5a9ce18>>, <bound method PortMapperUnpacker.unpack_uint of <pyvisa-py.protocols.rpc.PortMapperUnpacker instance at 0xb5a9ce40>>
2018-07-01 14:11:17,452 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb5a73b58>: '\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x01\x86\xa0\x00\x00\x00\x02\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x06\x00\x00\x00\x00'
2018-07-01 14:11:17,453 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb5a73b58>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\xb5')
2018-07-01 14:11:17,454 - pyvisa - DEBUG - RawTCPClient: closing socket
2018-07-01 14:11:17,454 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (192.168.250.110, 693)
2018-07-01 14:11:17,455 - pyvisa - DEBUG - Make call 10, (621311688L, 0, 10000, u'inst0'), <bound method Vxi11Packer.pack_create_link_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb5b70fd0>>, <bound method Vxi11Unpacker.unpack_create_link_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb5a90cb0>>
2018-07-01 14:11:17,455 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb51a87a0>: "\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\n\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00%\x08v\xc8\x00\x00\x00\x00\x00\x00'\x10\x00\x00\x00\x05inst0\x00\x00\x00"
2018-07-01 14:11:17,456 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb51a87a0>: bytearray(b'\x00\x00\x00\x01\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00')
2018-07-01 14:11:17,458 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - is open with session 4519833
>>> device.read_termination='\n'
>>> print(device.query("*IDN?"))
2018-07-01 14:11:58,092 - pyvisa - DEBUG - Make call 11, (0, 2000, 10000, 136, '*IDN?\r\n'), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb5b70fd0>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb5a90cb0>>
2018-07-01 14:11:58,092 - pyvisa - DEBUG - Sending record through <socket._socketobject object at 0xb51a87a0>: "\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\xd0\x00\x00'\x10\x00\x00\x00\x88\x00\x00\x00\x07*IDN?\r\n\x00"
2018-07-01 14:11:58,095 - pyvisa - DEBUG - Received record through <socket._socketobject object at 0xb51a87a0>: bytearray(b'\x00\x00\x00\x02\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07')
2018-07-01 14:11:58,095 - pyvisa - DEBUG - TCPIP::192.168.250.110::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 564, in query
    return self.read()
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 318, in read
    ret = self.sessions[session].read(count)
  File "/home/tinker/.local/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 105, in read
    start_time = time.monotonic()
AttributeError: 'module' object has no attribute 'monotonic'

MatthieuDartiailh, 2018-07-01:
Looking at the response you get when reading everything I wonder if the first part of the message is not written as soon as you open the connection. It may be said in the documentation. Could you try the following ?
- call `read` right after opening the connection and then query the idn
- query the idn and then call `read`
- do one of the two above and then query idn again to see if you get one line or two

For the INSTR, I believe you found a bug under Python 2. I will try to fix it ASAP.

AlteredWildebeest, 2018-07-01:
Ok sounds good thanks! I'll try right now

MatthieuDartiailh, 2018-07-02:
Can you try the other combinations I was suggesting ? Also can you try with a different query such as (SYST:COMM:LAN:IPAD?) ? Looking at the documentation of your instrument, they never discuss that extra first line you get, so it is a bit weird.
Also can you try to uninstall and re-install pyvisa-py ? The error you get using TCPIP:INSTR, should not be present on master since we never call time.monotonic, so your install is a bit weird.

AlteredWildebeest, 2018-07-03:
Hi Matthieu, certainly I would love to reinstall pyvisa-py. What are the sequence of commands I should use for that? I would like to do it exactly as you would like, so we have a common starting point.

As soon as the software is re-installed I will try with the other commands

MatthieuDartiailh, 2018-07-04:
Sorry for the delay.
To uninstall you should run the following command:
`pip uninstall pyvisa-py`
Then re-install using:
`pip install https://github.com/pyvisa/pyvisa-py/zipball/master`

Note that the repo moved to an organization by installing from hgrecco, you probably picked some old version.

MatthieuDartiailh, 2018-07-04:
Actually your install is fine (my local master branch was outdated...) I fixed the bug you were seeing. So you can test again.

closed 2019-03-21 by MatthieuDartiailh