pyvisa/pyvisa-py #151: PyVisa Exception Ignored in : <object repr() failed>

khalid17890, 2018-07-13:
Hi All,

I have a RIGOL DP832A Power Supply and the instrument does not close gracefully and ignores an exception:

> Exception ignored in: <object repr() failed>Traceback (most recent call last):
>  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 108, in __del__ 
> File "C:\Program Files (x86)\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 236, in close
> File "C:\Program Files (x86)\Python36-32\lib\logging\__init__.py", line 1293, in debug
> File "C:\Program Files (x86)\Python36-32\lib\logging\__init__.py", line 1546, in isEnabledFor
> TypeError: '>=' not supported between instances of 'int' and 'NoneType'

While debugging the error when I call psu.close() function it calls the close function of **"C:\Program Files (x86)\Python36-32\Lib\site-packages\pyvisa\resources\resource.py"** and retrieves this message: **Unable to get repr for <class 'pyvisa.resources.tcpip.TCPInstrument'>** from line 239 **"logger.debug('%s - is closed', self._resource_name, extra=self._logging_extra)"**. I have multiple instruments connected apart from PSU with the same resource manager but first I need exit the process with PSU gracefully. I have Rohde and Shwarz FSP13 connected via the following command: 
`self.interface` = visa_rm.open_resource(f"RSIB::{address}::INSTR", resource_pyclass=MessageBasedResource)`
code:
```
self.interface1 = visa_rm.open_resource(f"TCPIP::{address}::INSTR")
self.interface1.close()
def cleanup(self, msg_q):
    self.dut.close()
    self.psu.set_output(DUT_PSU_CHANNEL, False)
    self.psu.close()
```

System: 
Machine Details:
   Platform ID:    Windows-10-10.0.17134-SP0
   Processor:      Intel64 Family 6 Model 63 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Program Files (x86)\Python36-32\python.exe
   Version:        3.6.5
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Mar 28 2018 16:07:46 (#v3.6.5:f59c0932b4)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672

Any help will be appreciated.
Thanks in advance.

MatthieuDartiailh, 2018-07-13:
Can you try updating to PyVISA 1.9 ?

khalid17890, 2018-07-13:
Same error as soon as the PSU is closed.

```
pydev debugger: process 7928 is connecting

Connected to pydev debugger (build 181.4203.547)
2018-07-13 12:25:50,049 - pyvisa - DEBUG - viParseRsrcEx(98496160, 'TCPIP::192.168.85.200::INSTR', 'c_ushort(6)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x0AD26800>, <ctypes.c_char_Array_256 object at 0x0AD268F0>, <ctypes.c_char_Array_256 object at 0x0AD26940>) -> 0
2018-07-13 12:25:50,049 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - opening ...
TCPIPInstrument at TCPIP0::192.168.85.200::inst0::INSTR
2018-07-13 12:25:50,055 - pyvisa - DEBUG - viOpen(98496160, 'TCPIP::192.168.85.200::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x0AD26940>') -> 0
2018-07-13 12:25:50,055 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - is open with session 97490656
2018-07-13 12:25:50,055 - pyvisa - DEBUG - viGetAttribute(97490656, 3221159938, <ctypes.c_char_Array_256 object at 0x0AD26940>) -> 0
2018-07-13 12:25:50,057 - pyvisa - DEBUG - viWrite(97490656, b'*IDN?\r\n', 7, 'c_ulong(7)') -> 0
2018-07-13 12:25:50,057 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-13 12:25:50,058 - pyvisa - DEBUG - viRead(97490656, <ctypes.c_char_Array_20480 object at 0x0AD26940>, 20480, 'c_ulong(49)') -> 0
RIGOL TECHNOLOGIES,DP832A,DP8B195201974,00.01.16

2018-07-13 12:25:50,062 - pyvisa - DEBUG - viWrite(97490656, b'*RST\r\n', 6, 'c_ulong(6)') -> 0
2018-07-13 12:25:50,088 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH1,ON\r\n', 14, 'c_ulong(14)') -> 0
2018-07-13 12:25:50,097 - pyvisa - DEBUG - viWrite(97490656, b':SOUR1:VOLT 0.000000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:25:51,099 - pyvisa - DEBUG - viWrite(97490656, b':SOUR1:VOLT 3.300000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:25:51,108 - pyvisa - DEBUG - viWrite(97490656, b'SOUR1:CURR 0.800000\r\n', 21, 'c_ulong(21)') -> 0
2018-07-13 12:25:51,117 - pyvisa - DEBUG - viWrite(97490656, b':SOUR1:CURR:PROT 0.800000\r\n', 27, 'c_ulong(27)') -> 0
2018-07-13 12:25:51,127 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH1,ON\r\n', 14, 'c_ulong(14)') -> 0
2018-07-13 12:25:52,130 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH2,ON\r\n', 14, 'c_ulong(14)') -> 0
2018-07-13 12:25:52,138 - pyvisa - DEBUG - viWrite(97490656, b':SOUR2:VOLT 0.000000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:25:53,143 - pyvisa - DEBUG - viWrite(97490656, b':SOUR2:VOLT 3.300000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:25:53,152 - pyvisa - DEBUG - viWrite(97490656, b'SOUR2:CURR 0.800000\r\n', 21, 'c_ulong(21)') -> 0
2018-07-13 12:25:53,161 - pyvisa - DEBUG - viWrite(97490656, b':SOUR2:CURR:PROT 0.800000\r\n', 27, 'c_ulong(27)') -> 0
2018-07-13 12:25:53,171 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH2,ON\r\n', 14, 'c_ulong(14)') -> 0
2018-07-13 12:25:54,903 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH1,OFF\r\n', 15, 'c_ulong(15)') -> 0
2018-07-13 12:25:54,911 - pyvisa - DEBUG - viWrite(97490656, b':OUTP CH2,OFF\r\n', 15, 'c_ulong(15)') -> 0
2018-07-13 12:25:54,912 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - closing
2018-07-13 12:25:54,912 - pyvisa - DEBUG - viDisableEvent(97490656, 1073709055, 65535) -> 1073676291
2018-07-13 12:25:54,912 - pyvisa - DEBUG - viDiscardEvents(97490656, 1073709055, 65535) -> 1073676292
2018-07-13 12:25:54,920 - pyvisa - DEBUG - viClose(97490656,) -> 0
2018-07-13 12:25:54,920 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - is closed
2018-07-13 12:25:54,920 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - closing
2018-07-13 12:26:05,735 - pyvisa - DEBUG - viParseRsrcEx(103735736, 'TCPIP::192.168.85.200::INSTR', 'c_ushort(6)', 'c_ushort(0)', <ctypes.c_char_Array_256 object at 0x07383530>, <ctypes.c_char_Array_256 object at 0x07383620>, <ctypes.c_char_Array_256 object at 0x07383670>) -> 0
2018-07-13 12:26:05,735 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - opening ...
TCPIPInstrument at TCPIP0::192.168.85.200::inst0::INSTR
2018-07-13 12:26:05,750 - pyvisa - DEBUG - viOpen(103735736, 'TCPIP::192.168.85.200::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x07383670>') -> 0
2018-07-13 12:26:05,750 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - is open with session 102121448
2018-07-13 12:26:05,751 - pyvisa - DEBUG - viGetAttribute(102121448, 3221159938, <ctypes.c_char_Array_256 object at 0x07383670>) -> 0
2018-07-13 12:26:05,753 - pyvisa - DEBUG - viWrite(102121448, b'*IDN?\r\n', 7, 'c_ulong(7)') -> 0
2018-07-13 12:26:05,753 - pyvisa - DEBUG - TCPIP::192.168.85.200::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-07-13 12:26:05,754 - pyvisa - DEBUG - viRead(102121448, <ctypes.c_char_Array_20480 object at 0x07383670>, 20480, 'c_ulong(49)') -> 0
RIGOL TECHNOLOGIES,DP832A,DP8B195201974,00.01.16
2018-07-13 12:26:05,758 - pyvisa - DEBUG - viWrite(102121448, b'*RST\r\n', 6, 'c_ulong(6)') -> 0
2018-07-13 12:26:05,790 - pyvisa - DEBUG - viWrite(102121448, b':OUTP CH1,ON\r\n', 14, 'c_ulong(14)') -> 0
2018-07-13 12:26:05,799 - pyvisa - DEBUG - viWrite(102121448, b':SOUR1:VOLT 0.000000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:26:06,800 - pyvisa - DEBUG - viWrite(102121448, b':SOUR1:VOLT 3.000000\r\n', 22, 'c_ulong(22)') -> 0
2018-07-13 12:26:06,809 - pyvisa - DEBUG - viWrite(102121448, b'SOUR1:CURR 0.800000\r\n', 21, 'c_ulong(21)') -> 0
2018-07-13 12:26:06,819 - pyvisa - DEBUG - viWrite(102121448, b':SOUR1:CURR:PROT 0.800000\r\n', 27, 'c_ulong(27)') -> 0
[... cut ...]
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 108, in __del__
  File "C:\Program Files (x86)\Python36-32\lib\site-packages\pyvisa\resources\resource.py", line 242, in close
  File "C:\Program Files (x86)\Python36-32\lib\logging\__init__.py", line 1293, in debug
  File "C:\Program Files (x86)\Python36-32\lib\logging\__init__.py", line 1546, in isEnabledFor
TypeError: '>=' not supported between instances of 'int' and 'NoneType'

```

dlbaker2121, 2018-10-31:
I am experiencing similar warnings upon quit.  I use pyvisa command, rm. open_resource to access a Wiltron network analyzer and a Agilent power supply.  I am having no trouble running the program in Spyder Python Debugger, where I have 0 quit errors. But when I move to a Command Prompt window, where  my end user will run the code, I get similar quit "errors" to those listed.  I have tried different close commands without success.  My program functions correctly, but just ends with 20 lines of warnings.  I would prefer not to have these quit errors, any new discovers here?

MatthieuDartiailh, 2018-11-01:
I was never able to reproduce so no I did not fix it. Could you provide the output of `python -m visa info`, and the script in which you see the error ?

dlbaker2121, 2018-11-02:
Where would I find the python -m visa info?  I am a EE using Python for the 1st time, so not a computer scientist by trade.  The script is very long, 8 pages…..

khalid17890, 2018-11-02:
Hi @dlbaker2121 @MatthieuDartiailh 

I have found the solution and sorry for the delay to share it with you. 
You after you call the close function of your lan driver you need to delete the resource manager instance by the command mentioned below in the same close function.
'del self.interface1'
I hope this will get rid of the errors you're getting as well.

Kind regards,
Khalid

MatthieuDartiailh, 2018-11-02:
@khalid17890 could you post your full script I would like to understand the root issue behind this error ?

MatthieuDartiailh, 2018-11-02:
This does not tell me how you create the resource manager or when you dispose of it. I would really like to see a minimal reproducing script.