pyvisa/pyvisa-py #153: rm.list_resources() Error -- No handlers could be found for logger "pyvisa-py.usb"

cteixeira4066, 2018-08-27:
Hello. I have been messing around with PyVisa-py for a little bit, and although I've got it working on my Mac, I cannot seem to get it working on my Linux machine.. 

I'm currently running Linux with kernel: 4.15.0-33-generic and have installed both pyvisa and pyvisa-py through pip using the instructions shown here: http://python-in-the-lab.blogspot.com/2014/10/communicating-with-instruments-using.html

However, I get the following error when trying to use the list_resources() function:
```
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
No handlers could be found for logger "pyvisa-py.usb"
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
libgpib: error locking board mutex!
```
And I have the following PyVisa info:

```
Machine Details:
   Platform ID:    Linux-4.15.0-33-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Dec  4 2017 14:50:18 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR: Available via Linux GPIB (4.1.0)
      TCPIP SOCKET: Available 
```

Any help would be greatly appreciated!

MatthieuDartiailh, 2018-08-27:
I have never seen this error myself but it appears to have been previsouly reported in https://github.com/pyvisa/pyvisa-py/issues/78. Looks like uninstalling and re-installing may clean this up.

zamora18, 2018-09-13:
I also have a similar issue. I see that I have the latest pyvisa-py whereas @cteixeira4066 has 0.3.0. Not sure if that makes a difference.

My pyvisa information:
```
Machine Details:
   Platform ID:    Linux-4.15.0-34-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/jacobz/anaconda2/bin/python
   Version:        2.7.15
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          May  1 2018 23:32:55 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR: Available via Linux GPIB (< 4.0)
      TCPIP SOCKET: Available 
```

The script I am running:
```python
import visa
rm = visa.ResourceManager('@py')
rm.list_resources()
```

The output of `rm.list_resources()`:
```
No handlers could be found for logger "pyvisa-py.usb"
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
```

It seems like one "fix" in #78 was to uninstall and reinstall linux-gpib. 
On that note, I personally think it would be very helpful (although it isn't necessarily specific to pyvisa) to have a good walk-through of installing linux-gpib on Ubuntu as well as the gpib firmware for a couple USB/GPIB cables, maybe outlining which versions of linux-gpib seem to work well with pyvisa. I had (and may have) some issues with the installation that make it seem like it wasn't set up properly (going back to the "fix" of uninstalling and reinstalling). Or maybe there is one already created that you could point me to?

Thanks

MatthieuDartiailh, 2018-09-13:
Sorry I do not of any personally, but I would accept a PR to add that to pyvisa-py documentation.

zamora18, 2018-09-13:
I could contribute what I have, but I don't yet have a working system with the GPIB interface.

bdlabitt, 2019-03-21:
Same problem.
Installation:
`bruce@bruce-Bonobo-Extreme:~$ pip install pyvisa
Collecting pyvisa
  Downloading https://files.pythonhosted.org/packages/74/85/4cd8bde960e0e3efc635428b00533b93f9f220922c5d45b2801a8d96c666/PyVISA-1.9.1.tar.gz (440kB)
    100% |████████████████████████████████| 450kB 1.7MB/s 
Collecting enum34 (from pyvisa)
  Downloading https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Building wheels for collected packages: pyvisa
  Running setup.py bdist_wheel for pyvisa ... done
  Stored in directory: /home/bruce/.cache/pip/wheels/57/82/5b/f674b2e81c884b9f215f3981cd8d81fb4f5c0108aa1b9760c1
Successfully built pyvisa
Installing collected packages: enum34, pyvisa
Successfully installed enum34-1.1.6 pyvisa-1.9.1
bruce@bruce-Bonobo-Extreme:~$ pip install pyvisa-py
Collecting pyvisa-py
  Downloading https://files.pythonhosted.org/packages/7c/85/6f3170a31596d8e1ae55a108c07976da7ed2827c7a4328958fc242a88bc9/PyVISA-py-0.3.1.tar.gz (57kB)
    100% |████████████████████████████████| 61kB 1.2MB/s 
Collecting pyvisa>=1.8 (from pyvisa-py)
Collecting enum34 (from pyvisa>=1.8->pyvisa-py)
  Using cached https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Building wheels for collected packages: pyvisa-py
  Running setup.py bdist_wheel for pyvisa-py ... done
  Stored in directory: /home/bruce/.cache/pip/wheels/b0/e0/08/a9fac0d56547d74561e9ce37fb569d29e88b003f51233d4301
Successfully built pyvisa-py
Installing collected packages: enum34, pyvisa, pyvisa-py
Successfully installed enum34-1.1.6 pyvisa-1.9.1 pyvisa-py-0.3.1
bruce@bruce-Bonobo-Extreme:~$ pip install PyUSB
Collecting PyUSB
  Downloading https://files.pythonhosted.org/packages/5f/34/2095e821c01225377dda4ebdbd53d8316d6abb243c9bee43d3888fa91dd6/pyusb-1.0.2.tar.gz (54kB)
    100% |████████████████████████████████| 61kB 1.2MB/s 
Building wheels for collected packages: PyUSB
  Running setup.py bdist_wheel for PyUSB ... done
  Stored in directory: /home/bruce/.cache/pip/wheels/1f/a9/7e/d189b5030ee3a56f9b72c28281bb11d661b8ea312e28de08a5
Successfully built PyUSB
Installing collected packages: PyUSB
Successfully installed PyUSB-1.0.2
bruce@bruce-Bonobo-Extreme:~$ pip install libusb1
Collecting libusb1
Installing collected packages: libusb1
Successfully installed libusb1-1.7`

Checking if pyvisa-py is installed correctly:
`
bruce@bruce-Bonobo-Extreme:~$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.18.0-16-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.15rc1
   Compiler:       GCC 7.3.0
   Bits:           64bit
   Build:          Nov 12 2018 14:31:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
`
Using ipython:
`
bruce@bruce-Bonobo-Extreme:~$ ipython
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import visa

In [2]: rm = visa.ResourceManager('@py')

In [3]: rm.list_resources()
No handlers could be found for logger "pyvisa-py.usb"
Out[3]: ()
`

FYI, I did pip uninstall for all of the above, and a reinstall.  Same results.  I am trying to hookup a Siglent Digital Storage scope, with a USB interface.

MatthieuDartiailh, 2019-03-25:
I believe that the logger message is unrelated to the fact that no instruments are detected. The other users actually had an issue with linux-gpib which does not appear to be your case. USB requires some configuration on linux (outside of pyvisa), which is described in the docs. Could you make sure you properly configured your system ? I will also try to have a look at the logger issue.

bdlabitt, 2019-03-25:
Thanks.  What area should I look in?  (Which docs?)  I browsed through pydocs and didn't find anything.
1. libusb1?  or 
2. PyUSB?
If in "Linux" what area?  A search for usb, yields many 100's of items.  Do you mean something like modifying udev rules?  If I plug in the oscilloscope and enable USB Device USBTMC, I see it when I type lsusb.
`Bus 003 Device 003: ID f4ed:ee3a Shenzhen Siglent Co., Ltd. SDG1010 Waveform Generator (TMC mode)
`

bdlabitt, 2019-03-25:
**My issue is solved**, by adding a udev rule.
`ATTRS{idVendor}=="f4ed", ENV{ID_MM_DEVICE_IGNORE}="1"                                                                
ATTRS{idVendor}=="f4ed", MODE="0660", GROUP="adm"`
First line is to tell ModemManager to ignore this device.  (Ubuntu).  Second is to make the scope part of "adm" and to change the privileges to non-root. 

Result is now: 
`In [4]: rm.list_resources()
Out[4]: (u'USB0::62701::60986::SDS1ECDQ2R5766::0::INSTR',)
`
I totally don't understand why code tags suppress carriage returns.  Linux vs Windows?

MatthieuDartiailh, 2019-03-25:
Good to know you solved your issue. I really need to improve our docs about USB on linux but I have little experience with it. 
Could you clarify what you mean about code tags ?

bdlabitt, 2019-03-25:
GitHub code tags, they are the <> on the grey bar in the "Write" tab.  My code had carriage returns in it but GitHub rendered it as continuous lines.  Also my python code Out[4] should have been rendered on the next line.  Makes it hard to clearly convey ideas or comments, if GitHub rearranges things.  

Not pyvisa-py's problem.

It wasn't clear to me initially, that I had to install both PyUSB and libusb1.  
Perhaps you should just add a line (for linux installs) that states one may have to modify udev rules to allow non-root access?  http://ask.xmodulo.com/change-usb-device-permission-linux.html has instructions for this.

MatthieuDartiailh, 2019-03-26:
Thanks for the clarification ! and the pointer for the docs.

MatthieuDartiailh, 2019-07-22:
Fixed in #195

closed 2019-07-22 by MatthieuDartiailh

lilichen9217, 2020-02-02:
> **My issue is solved**, by adding a udev rule.
> `ATTRS{idVendor}=="f4ed", ENV{ID_MM_DEVICE_IGNORE}="1" ATTRS{idVendor}=="f4ed", MODE="0660", GROUP="adm"`
> First line is to tell ModemManager to ignore [...]
where did you add the two lines to?

lilichen9217, 2020-02-02:
Hi MatthieuDartiailh,

where did you add the two lines to?  is it "/etc/udev/rules.d/"?

lilichen9217, 2020-02-02:
Hi bdlabitt,
where did you add the two lines to? to "/etc/udev/rules.d/"?

slina0, 2020-02-02:
Hi bdlabitt,
where did you add the two lines to? to "/etc/udev/rules.d/"?