pyvisa/pyvisa-py #154: list_resources() TypeError on Python 3.7

capaulson, 2018-08-28:
The issue:

```>>> rm.list_resources()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1637, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/local/lib/python3.7/site-packages/PyVISA_py-0.3.0-py3.7.egg/pyvisa-py/highlevel.py", line 326, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/local/lib/python3.7/site-packages/PyVISA_py-0.3.0-py3.7.egg/pyvisa-py/highlevel.py", line 325, in <listcomp>
    resources = sum([st.list_resources()
  File "/usr/local/lib/python3.7/site-packages/PyVISA_py-0.3.0-py3.7.egg/pyvisa-py/sessions.py", line 107, in iter_valid_session_classes
    if issubclass(val, Session):
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/abc.py", line 143, in __subclasscheck__
    return _abc_subclasscheck(cls, subclass)
TypeError: issubclass() arg 1 must be a class
```

My current configuration, all pulled from Git:
```
Machine Details:
   Platform ID:    Darwin-17.7.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /usr/local/opt/python/bin/python3.7
   Version:        3.7.0
   Compiler:       Clang 9.1.0 (clang-902.0.39.2)
   Bits:           64bit
   Build:          Jun 29 2018 20:13:13 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

This works fine on 2.7 on my machine. Any ideas?

MatthieuDartiailh, 2018-08-28:
Not sure it will change anything but could update pyvisa-py to latest master.

capaulson, 2018-08-28:
This should be the latest master @MatthieuDartiailh. I just pulled the source yesterday.

MatthieuDartiailh, 2018-08-28:
Did you install from PyPI or directly from Github ? The current master (from Github) should indicate 0.3 for pyvisa-py version.

jboomer, 2018-08-31:
I have the same issue, using the pyvisa-py backend latest master. 

python -m visa info returns:
```

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\workspace\agilent-u2000a-py\Scripts\python.exe
   Version:        3.7.0
   Compiler:       MSC v.1914 64 bit (AMD64)
   Bits:           64bit
   Build:          Jun 27 2018 04:59:51 (#v3.7.0:1bf9cc5093)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377151778
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1377151778
         Spec. Version: 5244928
   py:
      Version: 0.3.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

```

MatthieuDartiailh, 2018-08-31:
Thanks for testing. I will investigate.

jboomer, 2018-08-31:
Added the following line before the call to 'issubclass' in sessions.py:

`logger.debug('Type of %s %s', val, type(val))`

This prints (before raising the error):

`2018-08-31 11:49:11,825 - pyvisa - DEBUG - Type of <function Session.register_unavailable.<locals>._internal at 0x000000000A997B70> <class 'function'>`

Hope this helps

MatthieuDartiailh, 2018-08-31:
Indeed yes ! The issue is that `register_unavailable` actually stores a function instead of a class in the sessions dictionary. The fix should be easy, could you test the following ?
Replace:
```python
def _internal(*args, **kwargs):
            raise ValueError(msg)

_internal.session_issue = msg
```
by 
```python
class _internal(object):
    session_issue = msg

    def __init__(self, *args, **kwargs):
        raise ValueError(msg)

```

jboomer, 2018-08-31:
Indeed this fixes the error. Now onto the next one :) (this one's with libusb though)

MatthieuDartiailh, 2018-08-31:
Could you submit a PR at some point ?

capaulson, 2018-09-01:
Thanks for the fix! Closing the issue.

closed 2018-09-01 by capaulson

MatthieuDartiailh, 2018-09-01:
The PR is now merged.