pyvisa/pyvisa-py #156: Issue with socket connection

bserra, 2018-09-03:
Hi,

I am trying to use pyvisa-py to connect to an instrument (Lakeshore 336) that I can access through a raspberry. Ping and telnet (sending *IDN?) are working properly.
Connecting through the socket package is working perfectly, but when using pyvisa, I can write to an instrument but then it doesn't reply and I end up with a timeout.
For info, the lakeshore should return the following string `LSCI,MODEL336,LSA22LU/LSA217G,2.8\r\n`
```
import visa

host = '134.171.5.180'
port = 7777

visa.log_to_screen()
rm = visa.ResourceManager('@py')    

lakeshore = rm.open_resource('TCPIP::'+str(host)+'::'+str(port)+'::SOCKET')
lakeshore.write_termination = '\r\n'
lakeshore.timeout = 5000

print (lakeshore.query('*IDN?'))
```
Results 
```
2018-09-03 15:07:31,357 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-09-03 15:07:31,360 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2018-09-03 15:07:31,380 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-09-03 15:07:31,383 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2018-09-03 15:07:31,384 - pyvisa - DEBUG - Created library wrapper for unset
2018-09-03 15:07:31,386 - pyvisa - DEBUG - Created ResourceManager with session 6472505
2018-09-03 15:07:31,387 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - opening ...
2018-09-03 15:07:31,389 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - is open with session 2998415
test
2018-09-03 15:07:31,390 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-09-03 15:07:36,402 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "visa_connection.py", line 41, in <module>
    print (lakeshore.read())
  File "/home/bserra/berryconda3/envs/test/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
  File "/home/bserra/berryconda3/envs/test/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 386, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/bserra/berryconda3/envs/test/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2018-09-03 15:07:36,472 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - closing
2018-09-03 15:07:36,473 - pyvisa - DEBUG - TCPIP::134.171.5.180::7777::SOCKET - is closed
2018-09-03 15:07:36,524 - pyvisa - DEBUG - Closing ResourceManager (session: 6472505)

```
I don't understand why everything is working with telnet and socket while pyvisa I encounter this timeout. One thing that bothers me a bit is that in the log I can't see my string being send (be it query, write or write_raw).

Any help will be highly appreciated. 

Below the result of `python -m visa info`
```
Machine Details:
   Platform ID:    Linux-4.14.62-v7+-armv7l-with-debian-9.4
   Processor:

Python:
   Implementation: CPython
   Executable:     /home/bserra/berryconda3/envs/test/bin/python
   Version:        3.6.6
   Compiler:       GCC 6.3.0 20170516
   Bits:           32bit
   Build:          Jul 29 2018 08:27:12 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

MatthieuDartiailh, 2018-09-03:
Could you try setting `read_termination to \r\n` ?

bserra, 2018-09-04:
Changed it, everything works. Thanks a lot!