pyvisa/pyvisa-py #158: Resource Manager list resources does not show USB while slush does

guruvenkat7, 2018-09-06:
Hi,

Let me first thank you for this great package. I have been using it for a while now.

The output of 'python -m visa info' on my machine gives

Machine Details:
   Platform ID:    Linux-4.4.0-128-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/afosr/anaconda3/bin/python
   Version:        3.5.1
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Dec  7 2015 11:16:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP SOCKET: Available 
--------------------------------------------

As you can see, for the 'pyvisa-py' backend, the USB libraries are installed. However, when I try to list_resources after connecting the instrument , I get the following error:

VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

However, lsusb does show the USB device with an Idvendor and idproduct. Any ideas why the device is not showing up using pyivsa-py?

I think that I maybe able to initialise the instrument directly into pyvisa using the format:

USB[board]::manufacturer ID::model code::serial number[::USB interface number][::INSTR]

What should be the entries in this? Should it be:

'USB0::idVendor::idProduct::SerialNumber::INSTR'?

The serial number I'm getting from lsusb does not seem to work. 

Let me know if more information is required and thanks in advance for the help!

Regards,
Guru

MatthieuDartiailh, 2018-09-06:
If the instrument has a front panel usually one can access the connection info from there. Also check if the instrument provide a real VISA USB interface (usbtmc) or simply a serial emulation. If you have access to a machine with Ni-VISA or Keysight VISA you can easily find out.

I know that configuring that usb interface can also be tricky. One good reading is the readme of the following project: https://github.com/python-ivi/python-usbtmc (a similar should be added to the docs).

MatthieuDartiailh, 2018-09-06:
I was not suggesting you to install that package but merely to check the readme about udev configuration. 
I am not sure I follow what you are asking about the interface ?

guruvenkat7, 2018-09-06:
I meant that initialising the instrument using visa as

instr = rm.open_resource('USB0::idVendor::idProduct::SerialNumber::INSTR')

Is there anyway to find out what should be put in the brackets above?

I know that I'm trying to find a quick fix here rather than figuring out why exactly this is happening. This is unfortunately due to time constraints. Apologies for this.

MatthieuDartiailh, 2018-09-06:
Could you try updating pyvisa-py ? the 0.3 has been in development for a while but was finally released.
What instrument are you trying to connect to ? does it have a front panel ?

guruvenkat7, 2018-10-22:
@MatthieuDartiailh Terribly sorry for the huge delay in replying to this. We managed to make it work by using Pyserial instead of VISA and by making sure that we get a USB 3.0 port to connect to the device.

The device was the Tektronix USB RSA (https://www.tek.com/spectrum-analyzer/rsa306) which does not have a front panel but rather a software for control via USB.

While I wanted to ideally make it work with pyvisa-py, I am afraid that I'll have to push that off for another day. Many apologies for that and thanks for all the help.

closed 2018-10-22 by guruvenkat7