pyvisa/pyvisa-py #165: does not discover TCPIP INST

OlafEichstaedt, 2018-10-10:
We used to communicate with an ASTERION AST-501 power supply over TCPIP using the default "@ni" backend. Since there seems to be an issue with longer command strings transmitted over TCPIP, we wanted to test the pyVIsa-py backend but had to notice that our instrument is not discovered by the resource_manager.

Is this a bug? Is it related to the device not being compatible with pyVisa-py? Or do we need to configure something else to make it work? Any suggestions on how we get closer to the root cause? 

The ni-backend shows the device as 'TCPIP0::169.254.180.2::inst0::INSTR'.

pyvisa-info
```
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      x86 Family 6 Model 42 Stepping 7, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\[...]\python36-32\python.exe
   Version:        3.6.0
   Compiler:       MSC v.1900 32 bit (Intel)
   Bits:           32bit
   Build:          Dec 23 2016 07:18:10 (#v3.6.0:41df79263a11)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: c:\windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 18874368
         Spec. Version: 5244928
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```
pip list
```
Package        Version
-------------- ---------
altgraph       0.15
beautifulsoup4 4.6.0
comtypes       1.1.4
future         0.16.0
macholib       1.9
pefile         2017.11.5
pip            10.0.1
PyInstaller    3.3.1
pypiwin32      220
PyQt5          5.10.1
pyqt5-tools    5.9.0.1.2
pyserial       3.4
PyVISA         1.9.1
PyVISA-py      0.3.1
pywifi         1.1.10
setuptools     28.8.0
sip            4.19.8
tabulate       0.8.2
```

MatthieuDartiailh, 2018-10-10:
Currently pyvisa-py cannot discover by itself instruments on TCPIP but you can still connect to your instrument if you know the address (just use the same the ni backend gives you).

tivek, 2018-12-13:
It is probably worth looking into how the python-vxi11 project [implements list_resources() and list_devices()](https://github.com/python-ivi/python-vxi11/blob/cc4671da699f1f379137dc40ffc4a302d72e6f55/vxi11/vxi11.py#L501).

anderson-pa, 2020-03-02:
I would also be interested in having this functionality. Currently working with a scope and function generator over ethernet. Same story though: from Windows using NI backend, list_resources() shows the instruments but from Ubuntu using pyvisa-py backend, they aren't listed. Connecting by explicitly passing the address works as expected.

Found this [LXI Discovery Document](https://www.lxistandard.org/members/Adopted%20Specifications/Latest%20Version%20of%20Standards_/LXI%20Standard%201.5%20Specifications/LXI%20VXI-11%20Discovery%20and%20Identification%20Extended%20Function.pdf) that specifies the method for finding network devices:

> The VXI-11 protocol should be supported by all LXI Devices for discovery purposes. If an LXI
Device supports the VXI-11 Discovery Protocol, it shall be accomplished by issuing a broadcast
RPC call on the host’s subnet. The broadcast RPC shall be to either the port-mapper itself on port
111 (querying for VXI-11 support) or the NULL procedure (procedure 0) on the Program Number
assigned to the VXI-11 Core Service (0x0607AF).

Of course, I don't know how to do that even after a lengthy search... maybe somebody with more networking experience has an answer...

commit 2e1861c5 (2021-03-08): tcpip: List resources to fix #165
closed 2022-09-15 by MatthieuDartiailh