pyvisa/pyvisa-py #169: assert_trigger() is defined twice

tivek, 2018-11-26:
https://github.com/pyvisa/pyvisa-py/blob/5f41742ca41beb0701d879558d2a5db5924a8485/pyvisa-py/highlevel.py#L246

assert_trigger() is doubly implemented by #136 and #139

MatthieuDartiailh, 2018-11-27:
Good catch ! Feel free to open a PR to fix it if you have the bandwidth.

tivek, 2018-11-27:
Sure, I can do that. I have a related question that I would like to go through with you first. When an invalid session is used, some PyVisaLibrary functions return the tuple `0, constants.StatusCode.error_invalid_object`, while others return just `constants.StatusCode.error_invalid_object`. The latter seems to be more in line with what the NI backend does. I am inclined to return just a simple return code. What do you think? If you agree, the remaining functions that return tuples should also be touched up.

MatthieuDartiailh, 2018-11-27:
At the C level in VISA, all functions return only the status. But pointers are used to pass additional information.

On the Python side, we usually return the status code and whatever value would have been returned through a pointer (the status code is usually returned last). You can have a look at the implementation of write for example (https://github.com/pyvisa/pyvisa/blob/master/pyvisa/ctwrapper/functions.py#L1852) and generally the proper signature for the library methods are defined in https://github.com/pyvisa/pyvisa/blob/master/pyvisa/highlevel.py#L374

I hope this clarifies the situation.

tivek, 2018-11-27:
Thanks @MatthieuDartiailh. The question was originally about `assert_trigger()` since the two copies had different return types. But, as far as I can tell there is more. The following return type signatures are different than specified in `VisaLibraryBase` (pyvisa master):
- `PyVisaLibrary.assert_trigger()` - should return `StatusCode`, so it seems the second of two copies is correct,
- `PyVisaLibrary.clear()` - should return `StatusCode`, returns `0, StatusCode`,
- `PyVisaLibrary.gpib_send_ifc()` - should return `StatusCode`, returns `0, StatusCode`,
- `Session.gpib_command()` - should return `int, StatusCode`, returns `StatusCode`,
- `GPIBSession.clear()` - should return `StatusCode`, returns `0, StatusCode`,
- `GPIBSession.gpib_command()` - should return `int, StatusCode`, returns `StatusCode`,
- `GPIBSession.gpib_send_ifc()` - should return `StatusCode`, returns `0, StatusCode`,
- `GPIBSession.read_stb()` - should return `int, StatusCode`, returns `int`,
- `GPIBSession.trigger()` should be renamed to `assert_trigger()`,
- `TCPIPInstrSession.lock()` - should return `str, StatusCode`, returns `StatusCode`.

This looks like a trivial refactor, but the changes are going to be over the place. What do you think? While I'm at it I can make a pull request you can review.

MatthieuDartiailh, 2018-11-27:
Yes please fix all those at once in a dedicated PR (separate from the one in which you will introduce support for events). I really wish we could have a good test suite ... Thanks again for spotting those inconsistencies !

tivek, 2018-11-27:
Thanks @MatthieuDartiailh. The question was originally about assert_trigger() since the two copies had different return types. But, as far as I can tell there is more. The following return type signatures are different than specified in VisaLibraryBase (pyvisa master):

    PyVisaLibrary.assert_trigger() - should return StatusCode, so it seems the second of two copies is correct,
    PyVisaLibrary.clear() - should return StatusCode, returns 0, StatusCode,
    PyVisaLibrary.gpib_send_ifc() - should return StatusCode, returns 0, StatusCode,
    Session.gpib_command() - should return int, StatusCode, returns StatusCode,
    GPIBSession.clear() - should return StatusCode, returns 0, StatusCode,
    GPIBSession.gpib_command() - should return int, StatusCode, returns StatusCode,
    GPIBSession.gpib_send_ifc() - should return StatusCode, returns 0, StatusCode,
    GPIBSession.read_stb() - should return int, StatusCode, returns int,
    GPIBSession.trigger() should be renamed to assert_trigger(),
    TCPIPInstrSession.lock() - should return str, StatusCode, returns StatusCode.

This looks like a trivial refactor, but the changes are going to be over the place. What do you think? While I'm at it I can make a pull request you can review.

closed 2018-12-05 by MatthieuDartiailh