pyvisa/pyvisa-py #183: USB device whose serial number cannot be read (Agilent 33220A)

Nanosystem, 2019-02-01:
Does anyone have experience connecting to Agilent instruments over USB? I am trying to control an Agilent 33220A waveform generator using pyvisa-py and pyvisa cand the serial number can't be read over a USB connection.  I get the message:

Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::2391::1031::???::0::INSTR
Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::2652::8559::???::0::RAW
Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::1027::24577::???::0::RAW
 
I tried installing the WinUSB driver (v.6.1.7600.16385) as described in issue #125 as a fix for a different instrument to no avail. I am not using any NI drivers, nor is NI installed on my system other than what is included w/ pyvisa. My code is as follows (note the commented lines for the usb.core and usb.backend were also an attempt to fix this problem, both with and without the WinUSB driver installed, also had no effect):

import visa
#visa.log_to_screen()
#import usb.core
#dev = usb.core.find()
#import usb.backend.libusb1
#backend = usb.backend.libusb1.get_backend(find_library=lambda x: "./libusb-1.0.dll")
#dev = usb.core.find(backend=backend)
resources = visa.ResourceManager('@py')
resources.list_resources()

My system information is:

C:\Users\Administrator\Desktop\tools\STD>py -m visa info
Machine Details:
   Platform ID:    Windows-10-10.0.14393-SP0
   Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python36\python.exe
   Version:        3.6.5
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 28 2018 17:00:18 (#v3.6.5:f59c0932b4)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.2.1)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2019-02-04:
Since you are on windows, you can try install a VISA implementation (National Instrument, Keysight, ...) and try reading the serial number using it. If it works, I would be interested to know what those libraries do differently. You could investigate using Wireshark to inspect the data being sent and received for pyvisa-py and the NI/Keysight/... VISA implementation.
You can also try to connect directly to the instrument since you can most likely get the usb address from it.

AndKe, 2020-03-27:
I have a similar issue, my electronic load shows always up as:

Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::3034::33145::???::0::RAW

What is the expected way to open the USB0::....::RAW instead of what I do, opening 
rm.open_resource("ASRL/dev/ttyUSB0::INSTR")  

I did try rm.open_resource("USB0::3034::33145::???::0::RAW")  , but then I get 
"ValueError: The device has no langid"?

MatthieuDartiailh, 2020-03-28:
Let me clarify the way VISA deals with USB. When connecting an instrument using a USB cable, The instrument may support three types of communication:
- USBTMC which is an IVI foundation defined standard (is USB::INSTR resource)
- USB::RAW which is a NI specific protocol
- emulate a serial resource (ASRL) over USB
Some instrument may support both USB::INSTR and USB::RAW but I know of no instrument using both a USB specific protocol and emulating serial over USB. If you can access your resource using serial, you are fine this is the resource you want and any detection issue for USB is not going to impact you.