pyvisa/pyvisa-py #185: ValueError: [Errno 60] Operation timed out

saarangp, 2019-04-19:
I get the following error when I run this code 

`VISA_ADDRESS = 'USB0::10893::5990::MY55440772::0::INSTR'
rm = visa.ResourceManager()
scope = rm.open_resource(VISA_ADDRESS)
print("Connected to " + scope.query("*IDN?"))`

> USBError                                  Traceback (most recent call last)
> /anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py in write(self, data)
>     224         try:
> --> 225             return self.usb_send_ep.write(data)
>     226         except usb.core.USBError as e:
> 
> /anaconda3/lib/python3.6/site-packages/usb/core.py in write(self, data, timeout)
>     386         """
> --> 387         return self.device.write(self, data, timeout)
>     388 
> 
> /anaconda3/lib/python3.6/site-packages/usb/core.py in write(self, endpoint, data, timeout)
>     947                 _interop.as_array(data),
> --> 948                 self.__get_timeout(timeout)
>     949             )
> 
> /anaconda3/lib/python3.6/site-packages/usb/backend/libusb1.py in bulk_write(self, dev_handle, ep, intf, data, timeout)
>     823                             data,
> --> 824                             timeout)
>     825 
> 
> /anaconda3/lib/python3.6/site-packages/usb/backend/libusb1.py in __write(self, fn, dev_handle, ep, intf, data, timeout)
>     919         if not (transferred.value and retval == LIBUSB_ERROR_TIMEOUT):
> --> 920             _check(retval)
>     921 
> 
> /anaconda3/lib/python3.6/site-packages/usb/backend/libusb1.py in _check(ret)
>     594         else:
> --> 595             raise USBError(_strerror(ret), ret, _libusb_errno[ret])
>     596 
> 
> USBError: [Errno 60] Operation timed out
> 
> During handling of the above exception, another exception occurred:
> 
> ValueError                                Traceback (most recent call last)
> <ipython-input-5-ef162dd4a6ff> in <module>()
>       4 rm = visa.ResourceManager()
>       5 scope = rm.open_resource(VISA_ADDRESS)
> ----> 6 print("Connected to " + scope.query("*IDN?"))
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in query(self, message, delay)
>     561         """
>     562 
> --> 563         self.write(message)
>     564 
>     565         delay = self.query_delay if delay is None else delay
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in write(self, message, termination, encoding)
>     221             message += term
>     222 
> --> 223         count = self.write_raw(message.encode(enco))
>     224 
>     225         return count
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py in write_raw(self, message)
>     199         :rtype: int
>     200         """
> --> 201         return self.visalib.write(self.session, message)
>     202 
>     203     def write(self, message, termination=None, encoding=None):
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa-py/highlevel.py in write(self, session, data)
>     366         # from the session handle, dispatch to the write method of the session object.
>     367         try:
> --> 368             ret = self.sessions[session].write(data)
>     369         except KeyError:
>     370             return 0, StatusCode.error_invalid_object
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa-py/usb.py in write(self, data)
>     124         send_end, _ = self.get_attribute(constants.VI_ATTR_SEND_END_EN)
>     125 
> --> 126         count = self.interface.write(data)
>     127 
>     128         return count, StatusCode.success
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py in write(self, data)
>     312                                               data[begin:end])
>     313 
> --> 314             bytes_sent += raw_write(data)
>     315 
>     316         return bytes_sent
> 
> /anaconda3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py in write(self, data)
>     225             return self.usb_send_ep.write(data)
>     226         except usb.core.USBError as e:
> --> 227             raise ValueError(str(e))
>     228 
>     229     def read(self, size):
> 
> ValueError: [Errno 60] Operation timed out

MatthieuDartiailh, 2019-04-21:
This looks awfully similar to https://github.com/pyvisa/pyvisa/issues/387.
Sadly I do not really know what is happening here. Could you post the output of `pyvisa-info` and the model of the instrument you are trying to connect to ?

anderson-pa, 2020-03-09:
I get the exact same error on Ubuntu 18.04, python 3.6 when trying to communicate with an Agilent 33521A with the 5.02 Keysight firmware. It was working before when it had the v1.14 firmware, though even then if I had a malformed command I'd have to reset the instrument to recover comms.

Here's my visa info output:

> Machine Details:
>    Platform ID:    Linux-4.15.0-88-generic-x86_64-with-Ubuntu-18.04-bionic
>    Processor:      x86_64
> 
> Python:
>    Implementation: CPython
>    Executable:     /usr/bin/python3
>    Version:        3.6.9
>    Compiler:       GCC 8.3.0
>    Bits:           64bit
>    Build:          Nov  7 2019 10:44:02 (#default)
>    Unicode:        UCS4
> 
> PyVISA Version: 1.10.1
> 
> Backends:
>    ni:
>       Version: 1.10.1 (bundled with PyVISA)
>       #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
>          found by: auto
>          bitness: 64
>          Could not get more info:
>             VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
>    py:
>       Version: 0.3.1
>       ASRL INSTR: Available via PySerial (3.4)
>       USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
>       USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
>       TCPIP INSTR: Available 
>       TCPIP SOCKET: Available 
>       GPIB INSTR:
>          Please install linux-gpib to use this resource type.
>          No module named 'Gpib'
>