pyvisa/pyvisa-py #189: Python 2 support statement

slightlynybbled, 2019-04-23:
This project clearly supports legacy Python and there are a few issues which are referencing Python 2.  As Python 2 is [quickly reaching end-of-life](https://pythonclock.org/), should there be a statement that at least references an end-of-support for Python 2 in the docs and/or readme?

As a potential contributor, I would rather my efforts not be diluted into multiple versions, so I'm a proponent of announcing end-of-life coincident with the EOL for Python 2.7.

MatthieuDartiailh, 2019-04-23:
My plans (that I forgot to advertise clearly even though it is probably mentioned in some issues) was to have a release around the summer which would be the last supporting Python 2. There are too few contributors and not good testing tools to keep supporting more versions.

slightlynybbled, 2019-04-23:
Great!  Could you add to the readme so that the expectation can be set for those unable to change?

This may be something that you decide to pick a future minor version number to not exceed.  Since you are currently at v0.3.1, then a statement such as "Versions >= 0.4.0, to be released in `<month>` will no longer directly support Python 2".  This sort of statement will allow current users of that version to put `< 0.4.0` in their current dependency documents.

MatthieuDartiailh, 2019-04-23:
Good idea. The way things are going I think the next release will be 0.4.0 and hence Python 2 support will be dropped in 0.5.0. I will update the README on Wednesday.

closed 2019-05-05 by slightlynybbled