pyvisa/pyvisa-py #192: VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360)

ambegossi, 2019-06-18:
I'm trying to use the pyvisa-py library to communicate with a SRS ds345 function generator.

After the open resource command:
rm = visa.ResourceManager('@py')
ds345 = rm.open_resource('GPIB0::6::INSTR')

I'm trying to send a command to the function generator:
ds345.write('FREQ 32760.0')

but I'm getting the following error:
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-5-b553b82905e1> in <module>
      2 
      3 # DS345 Function Generator
----> 4 ds345.write('FUNC %d' %function)
      5 ds345.write('AMPL %f' %amplitude)
      6 ds345.write('OFFS 0')

~/anaconda3/envs/qtzoenv/lib/python3.7/site-packages/pyvisa/resources/messagebased.py in write(self, message, termination, encoding)
    221             message += term
    222 
--> 223         count = self.write_raw(message.encode(enco))
    224 
    225         return count

~/anaconda3/envs/qtzoenv/lib/python3.7/site-packages/pyvisa/resources/messagebased.py in write_raw(self, message)
    199         :rtype: int
    200         """
--> 201         return self.visalib.write(self.session, message)
    202 
    203     def write(self, message, termination=None, encoding=None):

~/anaconda3/envs/qtzoenv/lib/python3.7/site-packages/pyvisa-py/highlevel.py in write(self, session, data)
    371 
    372         if ret[1] < 0:
--> 373             raise errors.VisaIOError(ret[1])
    374 
    375         return ret

VisaIOError: VI_ERROR_SYSTEM_ERROR (-1073807360): Unknown system error (miscellaneous error).

MatthieuDartiailh, 2019-06-18:
Could you post the output of `pyvisa-info` ?

Currently pyvisa-py only check explicitly for TimeoutError during a write (https://github.com/pyvisa/pyvisa-py/blob/master/pyvisa-py/gpib.py#L186). You can try to tweak the logic to get a bit more details about what happens in the gpib lib. This part of the code could definitively be improved. PR welcome.

closed 2019-06-18 by ambegossi

MatthieuDartiailh, 2019-06-18:
Why did you close @ambegossi ? If you found a solution to your problem, please post it it might help others.

ambegossi, 2019-06-18:
I was able to resolve this by configuring the gpib port on the function generator manually. It was configured to receive commands on the wrong port. Thank you for your help

MatthieuDartiailh, 2019-06-18:
Thanks for your answer !