pyvisa/pyvisa-py #193: rm.list_resources() "failed to open configuration file libgpib: invalid descriptor"

amrelk, 2019-06-24:
My NI GPIB-USB-HS can communicate with my devices just fine with `ibtest` but not with pyvisa-py.

Here are the steps to reproduce (with my system info at the end):
```
Python 3.7.3 (default, Jun 24 2019, 04:54:02) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import visa                                                                                                                                                                                                                                                                                  

In [2]: rm = visa.ResourceManager('@py')                                                                                                                                                                                                                                                             

In [3]: rm.list_resources()                                                                                                                                                                                                                                                                          
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
failed to open configuration file
libgpib: invalid descriptor
Out[3]: ()

In [4]: visa.visa_info()                                                                                                                                                                                                                                                                             
Machine Details:
   Platform ID:    Linux-5.1.14-arch1-1-ARCH-x86_64-with-arch
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 9.1.0
   Bits:           64bit
   Build:          Jun 24 2019 04:54:02 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: /opt/ni-visa/usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.3.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb.util'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb.util'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.2.1 r[1815]')
```

MatthieuDartiailh, 2019-06-24:
Since you know the address of your instrument, can you try to directly open the resource ? I would like to understand if the issue is limited to `list_resources` or is wider.
Also note that currently, pyvisa-py assumes that the board number is 0 which may break with your adapter. If you have a good solution to determine the proper board number to scan from linux-gpib feel free to open a PR.

amrelk, 2019-06-24:
After some probing with strace, it appears that pyvisa-py looks for gpib.conf at /usr/etc/gpib.conf instead of /etc/gpib.conf where it is installed. Making a symlink fixed the issue. Not sure if this counts as a bug or not.

MatthieuDartiailh, 2019-06-24:
Thanks for the update.  Could you detail a bit more how you reached that conclusion ? Pyvisa-py specifies nowhere where to look for gpib.conf so I am curious to understand where that erroneous value is coming from.