pyvisa/pyvisa-py #196: gpib.py: VI_ATTR_SEND_END_EN should call IbcEOT, not the unimplemented IbcEndBitIsNormal

dirkenstein, 2019-07-11:
This issue may be specific to using pyvisa-py with linux-gpib 4.2.0 but may not happen with NI 488.2 libararies.

Basically linux-gpib does not implement IbcEndBitIsNormal being set to 0.

However, according to the NI-488.2 documentation,  IbcEndBitIsNormal seems to relate to _receiving_ EOI not sending it.

I believe VI_ATTR_SEND_END_EN should use  IbcEOT (0x4) constant for setting send EOI or not. This is both for` ibconfig()` and `ibask()`.

MatthieuDartiailh, 2019-07-11:
Thanks for report. If you can make a PR with those changes it would be great ! I will try to look into it but I am not very familiar with linux-gpib it may take me some time. 

As a side note, if you are familiar with low-level gpib operation, I would greatly appreciate if you have time to review https://github.com/pyvisa/pyvisa-py/pull/194 that aims at improving GPIB support in Pyvisa-py.

MatthieuDartiailh, 2020-07-12:
Closed by #194

closed 2020-07-12 by MatthieuDartiailh