pyvisa/pyvisa-py #202: pyvisa-py does not list the instruments while ni backend does.

Dog-menG, 2019-08-08:
I tried the code below with NI backend:

import visa
rm = visa.ResourceManager()
print (rm.list_resources())

The output shows the connecting instrument:
('ASRL3::INSTR',)

while I changed NI backend to py backend, the output is empty. pyvisa-py installed.
import visa
rm = visa.ResourceManager('@py')
print (rm.list_resources())

The output is empty:
()

Please suggest. Thanks

MatthieuDartiailh, 2019-08-08:
Pyvisa-py relies on pyserial for listing the resources. And I am not sure why it does not detect your resource. Could you please post the output of `pyvisa-info` and give some details about the instrument you are trying to connect and the kind of serial adapter you are using if you are using any. Also in some cases listing the resource may fail but the resource can be accessed nonetheless, could you try opening the resource with the known address ?

Henry-Avicena, 2025-03-25:
Any solution to this? I am having the same issue.

MatthieuDartiailh, 2025-03-25:
As I requested after the original post, can I get more details on your setup ?

Henry-Avicena, 2025-03-25:
C:\Users\user>pyvisa-info
Machine Details:
   Platform ID:    Windows-11-10.0.26100-SP0
   Processor:      Intel64 Family 6 Model 186 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\user\AppData\Local\Programs\Python\Python312\python.exe
   Version:        3.12.2
   Compiler:       MSC v.1937 64 bit (AMD64)
   Architecture:   AMD64
   Build:          Feb  6 2024 21:26:36 (#tags/v3.12.2:6abddd9)
   Unicode:        UCS4

PyVISA Version: 1.14.1

Backends:
   ivi:
      Version: 1.14.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 26214400
         Spec. Version: 7340544
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         architecture:
            ('x86', 64)
         Vendor: National Instruments
         Impl. Version: 26214400
         Spec. Version: 7340544
   py:
      Version: 0.7.2
      ASRL INSTR: Available via PySerial (3.5)
      TCPIP INSTR: Available
         Resource discovery:
         - VXI-11: partial (psutil not installed)
         - hislip: disabled (zeroconf not installed)
      TCPIP SOCKET: Available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         No backend available
      VICP INSTR:
         Please install PyVICP to use this resource type.
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'
      GPIB INTFC:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of functionalities.
         No module named 'gpib'

I tried installing libusb as well but that did not work, I also already have pyusb installed so I wonder if this improper installation warning is related to the reason why it does not show up. Currently I am connecting to a Keysight multimeter 34461A, but I will need to connect to a Keithley source meter in the long run. I am not using any serial adapter, just a usb to usb cable if that is what you are referring to. I have not tried to connect directly to it when it does not show up yet.

`
import pyvisa
rm = pyvisa.ResourceManager()
print(f"NI Backend: {rm.list_resources()}")

rm2 = pyvisa.ResourceManager('@py')
print(f"Py Backend: {rm2.list_resources()}")
`

Output:
NI Backend: ('USB0::0x2A8D::0x1301::MY60079317::INSTR', 'TCPIP0::192.168.4.159::inst0::INSTR')
C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyvisa_py\tcpip.py:409: UserWarning: TCPIP:instr resource discovery is limited to the default interface.Install psutil: pip install psutil if you want to scan all interfaces.
  warnings.warn(
C:\Users\user\AppData\Local\Programs\Python\Python312\Lib\site-packages\pyvisa_py\tcpip.py:122: UserWarning: TCPIP::hislip resource discovery requires the zeroconf package to be installed... try 'pip install zeroconf'
  warnings.warn(
Py Backend: ('TCPIP::192.168.4.59::INSTR', 'TCPIP::192.168.4.159::INSTR')

Interestingly enough, I was able to fix this on a raspberry pi with the following links:
https://stackoverflow.com/questions/66480203/pyvisa-not-listing-usb-instrument-on-linux
https://download.tek.com/document/1KW-61463-0__Raspberry_Pi_3_Save_Lab_Space_%20Cost_Application_Note_090718.pdf
(I installed pyvisa, pyvisa-py, libusb, pyseiral, and pyusb in this setup)

MatthieuDartiailh, 2025-03-25:
As explained in pyvisa-py documentation, to use USB devices on windows you need to uninstall the USBTMC driver.

My advice is to use the ivi backend on windows (except for tcpip resources).