pyvisa/pyvisa-py #206: Timeout error breaks communication

robsan00, 2019-09-06:
I have a problem with restoring from an error case where a query runs into a timeout. I am working under Ubuntu, executing the same script under Windows using NI-visa shows no problems.

Here an example script:
```
import visa

dev_name = "USB0::0x14EB::0x0090::<censored>::INSTR"
resource_manager = visa.ResourceManager()
instrument = resource_manager.open_resource(dev_name)

print(instrument.query_ascii_values("system:error?", converter="s"))

instrument.timeout = 5000  # 5 s
try:
    print(instrument.query_ascii_values("query?", converter="f"))
except Exception as e:
    print("expected timeout", e)

try:
    print(instrument.query_ascii_values("system:error?", converter="s"))
except Exception as e:
    print("unexpected error", e)
```

Under Ubuntu using pyvisa-py I get:
```
['+0', '"No error"\n']
expected timeout VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
unexpected error VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
If I try to execute the script again, nothing works anymore, the serial communication seems to be broken:
```
Traceback (most recent call last):
  File "test_script.py", line 5, in <module>
    instrument = resource_manager.open_resource(dev_name)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1763, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 216, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1720, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 213, in __init__
    self.after_parsing()
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/usb.py", line 201, in after_parsing
    self.parsed.serial_number)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 256, in __init__
    super(USBTMC, self).__init__(vendor, product, serial_number, **kwargs)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/protocols/usbtmc.py", line 166, in __init__
    **device_filters))
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/usb/core.py", line 1249, in device_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/pyvisa-py/protocols/usbutil.py", line 191, in cm
    if not fnmatch(getattr(dev, attr).lower(), pattern.lower()):
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/home/<censored>/venv/python3/lib/python3.6/site-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```
I have to switch the device off and on again, to restore communication.
Executing the same script under Windows (using NI-visa) works out just fine (also executing it multiple times).

Executing SCPI commands like "*RST" or "*CLS" after the timeout error, don't seem to help either.

Do I have to do any special clean-up in case of a timeout (where NI-visa is more lenient than pyvisa-py)?

The device I am working with, is a Pendulum CNT-90.
Excerpt from pip freeze:
```
pyusb==1.0.2
PyVISA==1.10.0
PyVISA-py==0.3.1
```

MatthieuDartiailh, 2019-09-06:
At the moment I can only suggest you give a try to the master branch of pyvisa-py. The USB backend is a nasty beast. I am working towards way to facilitate its development however there are far from ready. Let me know how it goes.

robsan00, 2019-09-09:
Thanks for the quick reply! Functionally the master branch seems to work fine (I don't get the issues mentioned above). However, I get spammed with UnknownAttribute errors in the log:
```
2019-09-09 09:38:34,688 | ERROR    | pyvisa.set_attribute: Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
Traceback (most recent call last):
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 425, in set_attribute
    return self._set_attribute(attribute, attribute_state)
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/usb.py", line 173, in _set_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
2019-09-09 09:38:34,689 | ERROR    | pyvisa.set_attribute: Unknown attribute 1073676312 (0x3fff0018 - VI_ATTR_TERMCHAR)
Traceback (most recent call last):
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 425, in set_attribute
    return self._set_attribute(attribute, attribute_state)
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/usb.py", line 173, in _set_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676312 (0x3fff0018 - VI_ATTR_TERMCHAR)
2019-09-09 09:38:34,690 | ERROR    | pyvisa.set_attribute: Unknown attribute 1073676344 (0x3fff0038 - VI_ATTR_TERMCHAR_EN)
Traceback (most recent call last):
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 425, in set_attribute
    return self._set_attribute(attribute, attribute_state)
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/usb.py", line 173, in _set_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676344 (0x3fff0038 - VI_ATTR_TERMCHAR_EN)
2019-09-09 09:38:34,690 | ERROR    | pyvisa.get_attribute: Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
Traceback (most recent call last):
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/sessions.py", line 377, in get_attribute
    return self._get_attribute(attribute)
  File "/home/<censored>/venv/py3_pyvisa_py_dev/lib/python3.6/site-packages/pyvisa-py/usb.py", line 160, in _get_attribute
    raise UnknownAttribute(attribute)
pyvisa-py.sessions.UnknownAttribute: Unknown attribute 1073676310 (0x3fff0016 - VI_ATTR_SEND_END_EN)
```
etc.

MatthieuDartiailh, 2019-09-09:
Good ! The unknown attribute has also been reported in another thread. If you have time feel free to have a look at it. I will try to have a look at it this week if I can manage.

MatthieuDartiailh, 2019-09-11:
@robsan00 Can you check https://github.com/pyvisa/pyvisa-py/pull/207 fixes your issue ?

robsan00, 2019-09-12:
Applying #207, the log errors are gone and everything else seems to work fine as well.

MatthieuDartiailh, 2019-09-18:
I merged since it does go in the right direction even though it probably requires more work.

closed 2019-09-18 by MatthieuDartiailh