pyvisa/pyvisa-py #210: pyvisa -py serial number cannot be read

yexub, 2019-09-30:
I tried connect my device using GPIB-USB-HS, got "Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::3034::5924::???::2::RAW
()"
At 'sudo python' model, it will return '()'

And I tired to fix it by the method "  https://techoverflow.net/2019/08/09/how-to-fix-pyvisa-found-a-device-whose-serial-number-cannot-be-read-the-partial-visa-resource-name-is-usb0-0instr/ "
it will not find any USB instruments,and return ' ( ) '

>> python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.18.0-80.7.1.el8_0.x86_64-x86_64-with-centos-8.0.1905-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.8
   Compiler:       GCC 8.2.1 20180905 (Red Hat 8.2.1-3)
   Bits:           64bit
   Build:          Jul  1 2019 16:43:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

MatthieuDartiailh, 2019-09-30:
If you are using a USB to gpib adapter you should try to access your resource as GPIB not as USB. This means you will have to install linux-gpib. Also note that depending on the adapter linux gpib may not be able to list the resources but you should still be able to connect to it.

yexub, 2019-10-09:
I have installed linux-gpib. I can get the resources list, 
>>> inst1.query('*IDN?')
'KEITHLEY INSTRUMENTS INC.,MODEL 2400,685011?,C30   Mar 17 2006 09:29:29/A02  /F/E\n'
>>> inst1.read()
'+9.910000E+37,+1.000029E-03,+9.910000E+37,+1.176638E+03,+3.789200E+04\n'
>>> inst.query('*IDN?')
'KEITHLEY INSTRUMENTS INC.,MODEL 2182 ,0857394,A09  /A02  '
>>> inst.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 427, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 400, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.6/dist-packages/pyvisa-py/highlevel.py", line 350, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> 

I use read method to read data, It works on 2400, but get errors on 2182,