pyvisa/pyvisa-py #211: Having problems with large data transfers (NI vs pyvisa-py)

charkster, 2019-10-07:
After looking at successful transfers from my RIgol ds1054z (screen captures 30kbytes in size) on the NI backend, and comparing them with my Ubuntu pyvisa-py backend... I believe pyusb is messing-up somewhere. I have been looking at wireshark captures, and the Windows 10 NI captures compare very well with my Ubuntu ones with the exception being the configurations of the bulk end points (I believe). I am able to see the USBTMC data packets and they all appear proper. That leaves the configuration of the BULK-in and BULK-out endpoints.
My packet lengths are 64 to 128 bytes on pyusb and 40 to 60 bytes on Windows 10. The data payloads are the same. When I look under /dev I can see that pyusb was able to disconnect the kernel driver, but maybe something was done to the interface which messes-up my large data transfers.
Any ideas would be appreciated.

charkster, 2019-10-20:
I switched to the raspberry pi's kernel usbtmc driver and I have no issues now.
I will take a wireshark capture to see how it differs from pyvisa-py.

```
#!/bin/bash
sudo chmod a+w /dev/usbtmc0
echo ":DISP:DATA? ON,OFF,PNG" > /dev/usbtmc0
sleep 3
sudo cat /dev/usbtmc0 > rigol_kernel_usbtmc.raw
dd bs=11 skip=1 if=rigol_kernel_usbtmc.raw of=rigol_kernel_usbtmc.png
rm rigol_kernel_usbtmc.raw
```

MatthieuDartiailh, 2019-10-20:
That would be a great datapoint to have ! The usbtmc on pyvisa-py suffers from several issues but I do have currently access to the hardware not the time to actually fix it. Any support is then welcome.

closed 2025-02-21 by MatthieuDartiailh

2xB, 2025-02-21:
@charkster If you still by chance have a way to test this, could you test if with the merge of #490 this is fixed for you? It does apply to large data (in 2019 the bug I fixed occurred when transmitting > 2*1024 bytes), therefore this seems likely.

PR #490 (2025-02-21): Fix large data transfer using USBTMC
Transfer of large (e.g. binary) data did not work before due to an accidentally overwritten variable. This is solved in this commit by changing the variable name.

- [x] Closes #211
- [ ] Executed ``black . && isort -c . && flake8`` with no errors
- [ ] The change is fully covered by automated unit tests
- [ ] Documented in docs/ as