pyvisa/pyvisa-py #216: Pyinstaller packaged exe file can't find pyvisa-py

mike9755, 2019-11-30:
import pyvisa
rm = pyvisa.ResourceManager('@py')

part of code as above, it works well with direct running *.py file, but it failed when package the *.py file to executable file with pyinstaller, the wrong message as below:
"ValueError: Wrapper not found: No package named pyvisa-py"
Does anybody meet same question with me or tell me how to fix it, thank you very much

[image]

MatthieuDartiailh, 2019-12-02:
@charlesnicholson may know more since he contributed a patch a couple of years ago to simplify using pyinstaller. 
Given that the way pyvisa access pyvisa may be considered slightly non standard, pyinstaller may fail to pick up pyvisa-py as a used module. Could you force pyinstaller to package pyvisa-py ? Note that this is merely a guess I have never used pyinstaller.

mike9755, 2019-12-08:
I can't fix this problem now, so I try to use NI visa firstly, hope this problem could be fix in future

kristofgilicze, 2020-04-02:
For imports where PyInstaller unable to autodiscover imports ( mainly dynamic __import__, maybe eval() statements ) use the `hiddenimports` list in the spec file.
```python
hiddenimports = [
     'pyvisa-sim',
     'pyvisa-py'
]
```
Is there more dynamic import in PyVisa which should be listed ( i am not too familiar with the codebase) ?

MatthieuDartiailh, 2020-04-02:
In principle any optional backend, but I am not aware of any in addition to those two.

kristofgilicze, 2020-04-03:
I see 2 ways of resolving this issue:

Publishing a hook file with all the 'optional backends' as hiddenimports in the official PyInstaller repository would solve this issue for most people.
https://github.com/pyinstaller/pyinstaller/tree/develop/PyInstaller/hooks

or keep it manual as - is, maybe a docs mention.

johndefiore, 2020-08-12:
Has anyone been successful in getting Pyinstaller to generate an executable that can find Pyvisa-py?  I've tried:

hiddenimports = [ 'pyvisa-py']

in the spec file, but it doesn't seem to work.  I get the same result as the OP.  Any hints or even debug ideas would be greatly appreciated.

MatthieuDartiailh, 2020-08-12:
I have no experience with PyInstaller but it may be confused by the fact that technically `pyvisa-py` is not a valid package name. We recently ran into the same issue with mypy and renamed the installed package to `pyvisa_py` (the project name and the PyPI package keep the same name). If that is an option you can try using both pyvisa master and pyvisa-py master to see if it helps.

johndefiore, 2020-08-12:
@MatthieuDartiailh Thanks very much, that was the issue.  All is working now.

johndefiore, 2020-08-12:
@MatthieuDartiailh I'd be happy to, if I had any idea how to do that.  (Sorry, I'm an analog hardware guy, just barely can manage to get some code written and not really overly familiar with github procedures.)

MatthieuDartiailh, 2020-08-12:
I think the best place to put such information would be the FAQ. You can probably do everything from the web interface as follow:
- at https://github.com/pyvisa/pyvisa-py click on the fork button to create a copy of the repo under your own name
- in your fork go to docs/source/faq.rst and click on the pen icon to edit the file (I suggest your write your paragraph outside of the web interface and copy-paste it).
- when you are done, at the bottom of the page select  `Create a new branch for this commit and start a pull request` and then click commit
- on the next page check that the description of the PR is accurate and open the PR. You can also check that the target branch is the master branch of this repo, not of your own fork.

If you encounter any issue with the formatting don't worry I can take care of it. If this feels a bit overwhelming, feel free to send to me what you look like to add to the documentation by email and I will integrate it.

jaydoherty, 2021-07-07:
I can confirm that "hiddenimports = ['pyvisa_py']" is the correct syntax that is needed for Pyinstaller to locate the PyVISA-py backend (pyvisa 1.11.3, pyvisa-py 0.5.2, pyinstaller 4.3). And just to be very clear for anyone else out there, your spec file should look something like this:

```
block_cipher = None
a = Analysis(['minimal.py'],
     pathex=['/Developer/PItests/minimal'],
     binaries=None,
     datas=None,
     hiddenimports=['pyvisa_py'],  # <----- this is the key line
     hookspath=None,
     runtime_hooks=None,
     excludes=None,
     cipher=block_cipher)
pyz = PYZ(a.pure, a.zipped_data,
     cipher=block_cipher)
exe = EXE(pyz,... )
coll = COLLECT(...)
```

MatthieuDartiailh, 2021-07-07:
@jaydoherty would you have time to make a PR against the docs to add this information ?

Wiew1, 2022-11-08:
You must import pyvisa_py library using hiddenimport in pyinstaller command. 
My command:  `pyinstaller --onefile --windowed --hidden-import=pyvisa_py main.py`

without command --hidden-import the pyvisa library is not included and after running exe file, the message [Wrapper not found: No package named pyvisa_py´] is shown.

karelv, 2024-02-20:
```pyinstaller --onefile --windowed --hidden-import=pyvisa_py main.py```

That did the trick for me as well, thanks a lot!