pyvisa/pyvisa-py #219: [COM] Communication issue with pyvisa-py

mingdu2, 2020-02-29:
I have a python project with multiple scripts converted to simplified.exe by pyinstaller.
the following code ran into a problem.

// simplified.py
import visa
rm = pyvisa.ResourceManager('@py')

error message  is while running simplified.exe
Wrapper not found: No package named pyvisa-py
of course , running simplified.py has no issue.

Wondering if someone here may have some idea .
I  appreciate it.
Randy

MatthieuDartiailh, 2020-02-29:
I think this is not the first time a similar issue is reported and I believe it boils down to the fact that pyinstaller does not pick up pyvisa-py as a dependency of your scrip (because you do not import it explicitely). I am not familiar with Pyinstaller but if you can force it to include it in the exe it should work. If you manage to figure your issue out please consider making a PR to improve the documentation.

bilderbuchi, 2020-05-11:
Does this guide from pyinstaller docs help you? https://pyinstaller.readthedocs.io/en/stable/when-things-go-wrong.html#listing-hidden-imports

MatthieuDartiailh, 2020-05-11:
Thanks for the pointer @bilderbuchi.