pyvisa/pyvisa-py #229: pyvisa-info does not start

bilderbuchi, 2020-05-11:
Running `pyvisa-info` in a powershell terminal on Windows fails with
```
Fatal error in launcher: Unable to create process using '"d:\bld\pyvisa_1584488223356\_h_env\python.exe"  "C:\Users\buchner\AppData\Local\Continuum\miniconda3\Scripts\pyvisa-info.exe" ': The system cannot find the file specified.
```
I have confirmed that the pyvisa-info.exe is present where the error message says, but I don't have a Python installation at this D: path.

It seems to me that somehow during packaging a machine-local path for Python slipped into the configuration (I don't even have a D: drive). I could not find out how the pyvisa-info exe is created, it does not seem to use a setup.py console_script entrypoint.

Curiously, this happens with both a pyvisa-py pip installation and an installation from conda-forge. Pyvisa itself is on 1.10.1 from conda-forge.

To Reproduce
--------------
Steps to reproduce the behavior:
1. Install pyvisa-py on Windows with an Anaconda Python installation, either 0.4.0 via pip, or 0.3.1 from conda-forge (tried with both)
2. In a terminal/powershell, run `pyvisa-info`
3. Instead of diagnostic output, the above error appears

bilderbuchi, 2020-05-11:
shoot, I just realised that this does not come from pyvisa-py, but pyvisa itself. Sorry, can you transfer the issue, please? I also realised this affects pyvisa-shell, too, with the same error.

bilderbuchi, 2020-05-11:
OK, I just installed pyvisa from pip instead, and there the console tools work, so this seems to be an issue of the conda-forge package. :-/

MatthieuDartiailh, 2020-05-11:
Thanks. I will have a look at the conda package. Those script are created by setuptools and it requires some tweaking for conda to deal with them properly. I will try to fix the issue this week.

MatthieuDartiailh, 2020-05-11:
I opened a PR to fix the recipe. It just needs a review to get merged.

bilderbuchi, 2020-05-12:
Thanks for the quick fix, confirmed fixed.

closed 2020-05-12 by bilderbuchi