pyvisa/pyvisa-py #230: Resource name inconsistency between pyvisa-py and NI VISA on Windows

bilderbuchi, 2020-05-11:
On Windows 10, I have connected an USB-serial dongle. This shows as ususal as COM3 in the package manager, and also as `ASRL3::INSTR` in NI-MAX, and I can connect without problems:
```
import pyvisa
rm_ni = pyvisa.ResourceManager()
rm_ni.list_resources()
Out[4]: ('ASRL3::INSTR',)
instr = rm_ni.get_instrument('ASRL3::INSTR')
instr.close()
```
When using the pyvisa-py backend, it curiously gets listed as ASRL**COM**3::INSTR, which is not what I expected [from the docs](https://pyvisa.readthedocs.io/en/latest/introduction/names.html).
```
rm_py = pyvisa.ResourceManager('@py')
rm_py.list_resources()
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x00000220529BA1F8>
Traceback (most recent call last):
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\weakref.py", line 572, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
Out[8]: ('ASRLCOM3::INSTR',)
```
I don't know what the usb exception is about. 

I cannot connect with ASRL3::INSTR from NI-VISA (expected, I guess)
```
instrpy = rm_py.get_instrument('ASRL3::INSTR')
Traceback (most recent call last):
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\IPython\core\interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-9-16b5a564fc3a>", line 1, in <module>
    instrpy = rm_py.get_instrument('ASRL3::INSTR')
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa\resources\resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa-py\highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa-py\sessions.py", line 218, in __init__
    self.after_parsing()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa-py\serial.py", line 65, in after_parsing
    self.interface = cls(port=self.parsed.board, timeout=self.timeout, write_timeout=self.timeout)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '3': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
```
but it works with ASRLCOM3::INSTR
```
instrpy = rm_py.get_instrument('ASRLCOM3::INSTR')
instrpy.close()
```
So, I don't understand why pyvisa-py inserts this additional `COM` in the resource name?

One last thing that would be great if it could be fixed/allowed is resource names like "COM3" only. NI-VISA allows this, but pyvisa-py not. It would be helpful for the Windows users in my lab -- who typically think in "COMx" port names --  if this would work with pyvisa-py, too.
```
rm_ni.get_instrument('COM3')
Out[13]: <'SerialInstrument'('COM3')>
instr3 = rm_ni.get_instrument('COM3')
instr3.close()
instr4 = rm_py.get_instrument('COM3')
Traceback (most recent call last):
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\IPython\core\interactiveshell.py", line 3331, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-16-d6205ca0aef3>", line 1, in <module>
    instr4 = rm_py.get_instrument('COM3')
  File "C:\Users\buchner\AppData\Local\Continuum\miniconda3\lib\site-packages\pyvisa\highlevel.py", line 1753, in open_resource
    resource_pyclass = self._resource_classes[(info.interface_type, info.resource_class)]
AttributeError: 'int' object has no attribute 'interface_type'
# Please make pyvisa-py accept "COM3", too :-)
```
Version information:
```
pyvisa.__version__
Out[19]: '1.10.1'
import pkg_resources
pkg_resources.get_distribution('pyvisa-py').version
Out[24]: '0.4.0'
import serial
serial.__version__
Out[26]: '3.4'
```

**Output of `pyvisa-info`**
Sorry, this is currently not working for me, see #229

MatthieuDartiailh, 2020-05-11:
Thanks for the detailed report.

MatthieuDartiailh, 2020-05-11:
Since you figured out how to get `pyvisa-info` to work could you post those details, now ?

bilderbuchi, 2020-05-12:
```
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\buchner\AppData\Local\Continuum\miniconda3\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          May  6 2020 11:45:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 19923456
         Spec. Version: 5244928
```