pyvisa/pyvisa-py #231: Communication issue with NHR9300 using pyvisa-py 4.0

ssezgin-cpi, 2020-05-11:
Open_resource function did not exit successfully and stay within infinite loop without any success and timeout error. Same IP (TCPIP0::{nhr_ip_address}::inst0::INSTR) communication worked fine using NI-VISA.

MatthieuDartiailh, 2020-05-11:
Please post the output of `pyvisa-info`.

ssezgin-cpi, 2020-05-11:
sezer@ubuntu:~/Projects/factory-hw-test$ pyvisa-info
Machine Details:
   Platform ID:    Linux-5.3.0-51-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3.7
   Version:        3.7.7
   Compiler:       GCC 7.5.0
   Bits:           64bit
   Build:          Mar 10 2020 15:16:38 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'

sezer@ubuntu:~/Projects/factory-hw-test$

MatthieuDartiailh, 2020-05-11:
Can you try to update to pyvisa-py 0.4.0 and see if the problem persist ? Note that this is similar to #227. If you can test the workaround described there it would be great.

ssezgin-cpi, 2020-05-15:
I will and report my findings when I have access to HW again.