pyvisa/pyvisa-py #235: [COM] Communication issue with Thorlabs PM101 using pyvisa-py

avigan, 2020-05-29:
I am having trouble connecting on Windows to a Thorlabs PM101 power meter via USB. Before installing NI-VISA I would like to test with pyvisa-py. When I start `pyvisa-shell`, it seems that only the devices connected via serial are detected, but none of the USB ones. In addition to the PM101 power meter I also have a TSB01B temperature sensor (also from Thorlabs), which I think should be detected. Any help would be greatly appreciated. Information is copied below

Instrument details
------------------
* Model: Thorlabs PM101 power meter
* Communication: USB
* Link to the documentation (if available): https://www.thorlabs.com/thorproduct.cfm?partnumber=PM101

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Windows-10-10.0.10586-SP0
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\programdata\anaconda3\python.exe
   Version:        3.7.7
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          May  6 2020 11:45:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb0
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb0
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

Output of `pyvisa-shell`
-----------------------
```
> pyvisa-shell
Welcome to the VISA shell. Type help or ? to list commands.
(visa) list
( 0) ASRLCOM11::INSTR
( 1) ASRLCOM9::INSTR
( 2) ASRLCOM1::INSTR
( 3) ASRLCOM6::INSTR
( 4) ASRLCOM5::INSTR
( 5) ASRLCOM7::INSTR
(visa) exit
```

MatthieuDartiailh, 2020-05-29:
As mentioned in the documentation, on Windows, you may have to uninstall the USBTMC specific driver if it was installed automatically by Windows and re-install a generic driver. See https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows for more details.

Honestly using a big manufacturer VISA implementation is likely to be much easier and portable. NI, Keysight, Rohde&Schwarz and Tektronik all provides the VISA library free of charge (make sure you get only the runtime, development environments are not free).

avigan, 2020-05-30:
Thanks for your answer. I apologize for missing that in the documentation.

Interestingly, I have installed the Tektronik's VISA implementation and it only shows the serial devices. Still no USB. I am not sure what I am missing here. Does your comment about uninstalling the USBTMC specific driver also apply here?

MatthieuDartiailh, 2020-05-30:
Looking at the documentation and at the Write Your Own Application document, it is not obvious to me that those instrumentsd support VISA. In particular when discussing USB in the second document, the following is mentioned: `The  commands  listed  in  this  section  are  supported  by  a  USBTMC  protocol  and  can  be  used with the instrument driver TLPM.dll.`
It is hence possible that Thorlabs does something custom and does not support VISA (which is distinct from SCPI). Over serial you could use VISA since VISA does not add anything complicated on top of it.
You may also want to check if your resource does not appear as a serial resource by connecting and disconnecting it and checking if any new device pop up (in windows peripheral tools), but I doubt it is the case. 
It may be worth contacting Thorlabs for a definitive answer.

avigan, 2020-05-30:
I think there are compatible, at least for the power meter: I tested it with pyvisa and NI-VISA on my MacBook and it was working fine. So I am not sure why it is not seen on windows with TekVISA.

MatthieuDartiailh, 2020-05-30:
If you uninstalled the USBTMC driver you should install it back.

MatthieuDartiailh, 2020-07-23:
Closing for lack of feedback feel free to reopen if necessary.

closed 2020-07-23 by MatthieuDartiailh

oscargastelum, 2023-07-08:
I was able to solve this by going into the device manager, looking for the Thorlabs USB connection, and updating the driver.