pyvisa/pyvisa-py #240: Blank output when inputting rm.list_resources() in pyvisa

hiroki-oist, 2020-07-11:
Hello anyone,

Please guide me on below problem.

Code:
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
()
>>> inst = rm.open_resource(TCPIP::192.168.0.2::SOCKET")
  File "<stdin>", line 1
    inst = rm.open_resource(TCPIP::192.168.0.2::SOCKET")
                                                    ^
SyntaxError: invalid syntax

Problem:
There's sourcemeter and nanovoltmeter connected by GPIB using port number 7 and 18.
I've sent a ping to one of them and have already checked if the connection is successful.
The hat which shows where the error is was under the first colon. I have no idea why this is a SyntaxErroe...

MatthieuDartiailh, 2020-07-11:
Currently pyvisa-py does detect TCPIP instrument (I am even sure there is a way to that for socket instrument). The Syntax error comes from the forgotten " at the beginning of the instrument address.

hiroki-oist, 2020-07-15:
> Currently pyvisa-py does detect TCPIP instrument (I am even sure there is a way to that for socket instrument). The Syntax error comes from the forgotten " at the beginning of the instrument address.

Thank you! It worked!

closed 2020-07-15 by MatthieuDartiailh