pyvisa/pyvisa-py #243: USB instrument timeout wrong behavior

wscode99, 2020-07-22:
When connecting to an instrument via USB, the response time to a query is taking "timeout" time duration and not when the response is available by instrument.

For example:
> pyvisa-shell
Welcome to the VISA shell. Type help or ? to list commands.
(visa) list
( 0) USB0::2391::8199::MY49008476::0::INSTR
(visa) open 0
USB0::2391::8199::MY49008476::0::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
> (open) timeout
Timeout: 2000ms
> (open) query *IDN?
Response: Agilent Technologies,34972A,MY49008476,1.17-1.12-02-02
==>  #the query takes 2 sec to return
> (open) timeout 10000
Done
(open) query *IDN?
Response: Agilent Technologies,34972A,MY49008476,1.17-1.12-02-02
==>  #the query takes 10 sec to return
(open) timeout 30000
Done
(open) query *IDN?
Response: Agilent Technologies,34972A,MY49008476,1.17-1.12-02-02
==>  #the query takes 30 sec to return

The expected behavior is that query should return immediately when the instrument put the answer in output buffer and should not be affected by timeout settings, unless the timeout time is not long enough to the instrument answer to query.

**Output of `pyvisa-info`**
pyvisa-info
Machine Details:
   Platform ID:    Linux-5.3.0-62-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Apr 18 2020 01:56:04 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2020-07-22:
@itayperl have you observed the same since you are the most recent contributor to that backend ?

itayperl, 2020-07-22:
Hi @wscode99, can you confirm whether you're running the latest pyvisa-py master?

@MatthieuDartiailh, not after adding REN support. Such timeouts were exactly the behavior I was seeing before making my changes.

MatthieuDartiailh, 2020-07-23:
Thanks @itayperl I misremembered your issue then (I though it was a complete communication failure). So running pyvisa-py master with pyvisa master sounds good.

wscode99, 2020-07-23:
I updated pyvisa to master: PyVISA (1.10.2.dev146+g124c46b), but unable to update pyvisa-py.

When run:> pip3 install git+https://github.com/pyvisa/pyvisa-py.git#egg=pyvisa-py
Get following error: 
Could not find a version that satisfies the requirement pyvisa>1.10.1 (from pyvisa-py) (from versions: 1.3, 1.4, 1.5, 1.6, 1.6.1, 1.6.2, 1.6.3, 1.7, 1.8, 1.9.0, 1.9.1, 1.10.0, 1.10.1)
No matching distribution found for pyvisa>1.10.1 (from pyvisa-py)

When run:> pip3 install ./Downloads/pyvisa-py-master.zip
Get following error: 
Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-wq0d_fa7-build/setup.py", line 6, in <module>
        setup(use_scm_version=True)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 372, in __init__
        _Distribution.__init__(self, attrs)
      File "/usr/lib/python3.6/distutils/dist.py", line 281, in __init__
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 528, in finalize_options
        ep.load()(self, ep.name, value)
      File "/tmp/pip-wq0d_fa7-build/.eggs/setuptools_scm-4.1.2-py3.6.egg/setuptools_scm/integration.py", line 17, in version_keyword
        dist.metadata.version = _get_version(config)
      File "/tmp/pip-wq0d_fa7-build/.eggs/setuptools_scm-4.1.2-py3.6.egg/setuptools_scm/__init__.py", line 148, in _get_version
        parsed_version = _do_parse(config)
      File "/tmp/pip-wq0d_fa7-build/.eggs/setuptools_scm-4.1.2-py3.6.egg/setuptools_scm/__init__.py", line 118, in _do_parse
        "use git+https://github.com/user/proj.git#egg=proj" % config.absolute_root
    LookupError: setuptools-scm was unable to detect version for '/tmp/pip-wq0d_fa7-build'.
    
    Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
    
    For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-wq0d_fa7-build/

itayperl, 2020-07-23:
@wscode try to install them both in the same command:

```
pip install git+https://github.com/pyvisa/pyvisa.git git+https://github.com/pyvisa/pyvisa-py.git
````

MatthieuDartiailh, 2020-07-23:
I cannot reproduce locally. Maybe try to upgrade pip.

wscode99, 2020-07-24:
I was able to update when install both in the same command. Thank you @itayperl 
But the problem of response time to a query still persists. It is taking "timeout" time duration to return a response.
Please help.

Now the pyvisa-info return the following:
Machine Details:
   Platform ID:    Linux-5.4.0-42-generic-x86_64-with-Ubuntu-18.04-bionic
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.9
   Compiler:       GCC 8.4.0
   Bits:           64bit
   Build:          Jul 17 2020 12:50:27 (#default)
   Unicode:        UCS4

**PyVISA Version: 1.10.2.dev146+g124c46b**

Backends:
   ivi:
      **Version: 1.10.2.dev146+g124c46b (bundled with PyVISA)**
      Binary library: Not found
   py:
      **Version: 0.4.1.dev45+gf4ef165**
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

itayperl, 2020-07-24:
> Thanks @itayperl I misremembered your issue then (I though it was a complete communication failure). So running pyvisa-py master with pyvisa master sounds good.

Er, sorry, my bad, it was a complete failure in my case without REN control. I failed to notice that the commands did return a valid response here after the timeout. So this must be a different issue.

MatthieuDartiailh, 2020-07-24:
@itayperl are you able to reproduce @wscode99 issue ?

itayperl, 2020-07-24:
No, I'm not. The issue is probably specific to the device @wscode99 is using.
The only USBTMC device I have right now is a Thorlabs PM101 and it works flawlessly so far on the latest master.

wscode99, 2020-08-19:
Hi @MatthieuDartiailh and @itayperl, I figured out that this problem occurs when pyvisa-py is installed on Linux VM machine (using Virtualbox) where the machine OS is Windows.
When pyvisa-py is installed on machine with Linux, then it works as expected!
You can close this ticket

MatthieuDartiailh, 2020-08-19:
Thanks for the feedback @wscode99 ! I will keep open till I add a note about this in the FAQ for future reference.

closed 2020-08-19 by MatthieuDartiailh