pyvisa/pyvisa-py #245: GPIB INSTR not found after linux-gpib 4.3.3 install on raspberry pi

charkster, 2020-08-02:
I am able to talk to my GPIB devices using ibtest and I can load the python3 module "Gpib", but pyvisa-py appears to not be able to find the the "GPIB INSTR" when I run pyvisa -info. Python2.7 did not get a module created for it (linux-gpib 4.3.3 appears to not create it anymore).
I tried to talk to my instrument without seeing it listed, but that did not work. Not sure what is going on as the Gpib module is loading and working fine.

To Reproduce
THIS WORKS

on the python3 command line:
import Gpib
inst = Gpib.Gpib(0,25)
inst.write("*IDN?")
inst.read(100)
b'KEITHLEY INSTRUMENTS INC.,MODEL 2000,4109271,B01 /A02 \n'

THIS DOES NOT WORK
2. on the python3 command line:

import pyvisa
import visa
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
ke2000 = rm.open_resource('GPIB0::25::INSTR')
Traceback (most recent call last):
File "", line 1, in
File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1771, in open_resource
res.open(access_mode, open_timeout)
File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/resource.py", line 218, in open
self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
File "/home/pi/.local/lib/python3.7/site-packages/pyvisa-py/highlevel.py", line 194, in open
sess = cls(session, resource_name, parsed, open_timeout)
File "/home/pi/.local/lib/python3.7/site-packages/pyvisa-py/sessions.py", line 170, in init
raise ValueError(msg)
ValueError: Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
cannot import name 'GpibError' from 'Gpib' (/usr/local/lib/python3.7/dist-packages/Gpib.py)

Machine Details:
Platform ID: Linux-5.4.51-v7+-armv7l-with-debian-10.4
Processor:

Python:
Implementation: CPython
Executable: /usr/bin/python3
Version: 3.7.3
Compiler: GCC 8.3.0
Bits: 32bit
Build: Dec 20 2019 18:57:59 (#default)
Unicode: UCS4

PyVISA Version: 1.10.1

Backends:
ni:
Version: 1.10.1 (bundled with PyVISA)
Binary library: Not found
py:
Version: 0.4.1
ASRL INSTR: Available via PySerial (3.4)
USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
TCPIP INSTR: Available
TCPIP SOCKET: Available
GPIB INSTR:
Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
cannot import name 'GpibError' from 'Gpib' (/usr/local/lib/python3.7/dist-packages/Gpib.py)

MatthieuDartiailh, 2020-08-03:
It looks like linux-gpib is not exporting GPIBError anymore which causes the issue. @tivek you have been in contact with the Linux gpib project, could you check ?
In the meantime @charkster you can install gpib-ctypes as a workaround. It will use Linux-gpi  under the hood but bypass the python binding.

charkster, 2020-08-04:
> It looks like linux-gpib is not exporting GPIBError anymore which causes the issue. @tivek you have been in contact with the Linux gpib project, could you check ?
> In the meantime @charkster you can [...]

Thanks Matthieu. I installed gpib-ctypes and I can see my gpib instruments.
However when I `print rm.list_resources()` I am seeing 60 lines of `libgpib: invalid descriptor`

Is there any way to suppress this? It is not a large problem as I normally don't use `list_resources()`

tivek, 2020-08-04:
Hm. linux-gpib did not change their Python API since 2018, so it is likely not them.

I think something weird is going on when pyvisa tries to import GpibError. GpibError should be imported from gpib, not Gpib.

As far as the libgpib errors when gpib-ctypes is used, I can take a look when I get access to the equipment in two weeks' time.

MatthieuDartiailh, 2020-08-07:
Thanks @tivek I fixed the issue for linux-gpib.

closed 2020-08-09 by bors[bot]

charkster, 2020-08-10:
Still seeing the issue after installing pyvisa@master and pyvisa-py@master. I assume that the change got pulled into the Master branch when this issue got closed.

MatthieuDartiailh, 2020-08-10:
Sorry I meant the issue about using linux-gpib with pyvisa-py...

I will open a new issue for the spurious prints

commit a7c96258 (2020-08-09): Merge #246

246: gpib: do not import GpibError from Gpib since it lives in gpib module r=MatthieuDartiailh a=MatthieuDartiailh

- [x] Closes #245 (insert issue number if relevant)
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] Added an entry to the CHANGES file

@charkster can you confirm it fixes your issue ?
PR #246 (2020-08-09): gpib: do not import GpibError from Gpib since it lives in gpib module

- [x] Closes #245 (insert issue number if relevant)
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] Added an entry to the CHANGES file

@charkster can you confirm it fixes your issue ?