pyvisa/pyvisa-py #250: [COM] Communication issue with GPIB using pyvisa

AbhayKrS-dev, 2020-08-20:
I'm creating an automation setup with GPIB instruments on centos and I'd like to automate as much as possible using pyvisa. i have the following things installed- 

1.	Centos 7.6, Python 3.6
2.	Kernel drivers (iokerneldrivers-installer.run)
3.	Keysight Connection Expert (iolibrariessuite-installer.run)
4.	Pyvisa , pyvisa-py library
5.	Pyusb updated version 1.0.2 (with default version which comes with python we were not able to communicate with usb)

After all this i am able to communicate successfully with USB/TCPIP devices but GPIB devices are not responding although it is getting detected in the keysight connection expert. 

I read the pyvisa-py backend library document and it says we would need linux-ctypes, linux-gpib drivers for communicating with GPIB instruments but still after that i am not able to communicate with the device and get the following error-

>>> dev = rm.open_resource("GPIB1::5::INSTR")
Traceback (most recent call last):
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/sessions.py", line 130, in get_session_class
    return cls._session_classes[(interface_type, resource_class)]
KeyError: (<InterfaceType.gpib: 1>, 'INSTR')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/highlevel.py", line 192, in open
    cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/sessions.py", line 133, in get_session_class
    (interface_type, resource_class))
ValueError: No class registered for InterfaceType.gpib, INSTR

Is there anything in particular that I would need to do to be able to communicate with the GPIB devices?

MatthieuDartiailh, 2020-08-20:
If you have installed Keysight Connection Expert you do not need to use pyvisa-py but can simply rely on the default backend of pyvisa. This should be the easiest solution.

I am not sure if linux-gpib and Keysight can conflict with one another. Another option would be to pip install gpib-ctypes which would then access the Keysight drivers. But you are probably better off just using pyvisa default backend.

AbhayKrS-dev, 2020-08-21:
Yes that's true it should not require pyvisa-py , I tried the same thing on windows and there only with the help of pyvisa I am able to communicate with all the kinds of devices- GPIB/USB/TCPIP
But on centos it is not responding for GPIB, I installed gpib-ctypes but I am getting the same error as mentioned above.

AbhayKrS-dev, 2020-08-24:
Output of pyvisa-info- 

pyvisa-info
/usr/local/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-3.10.0-957.el7.x86_64-x86_64-with-centos-7.6.1810-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-39)
   Bits:           64bit
   Build:          Apr  2 2020 13:34:55 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.4.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 

Error for the code - 

>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
/usr/local/lib/python3.6/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
>>> print(rm.list_resources())
()
>>> 
>>> rm.open_resource("GPIB::INFC")
Traceback (most recent call last):
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/sessions.py", line 130, in get_session_class
    return cls._session_classes[(interface_type, resource_class)]
KeyError: (<InterfaceType.gpib: 1>, 'INSTR')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1771, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.6/site-packages/pyvisa/resources/resource.py", line 218, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.6/site-packages/pyvisa/highlevel.py", line 1725, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/highlevel.py", line 192, in open
    cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
  File "/usr/local/lib64/python3.6/site-packages/pyvisa-py/sessions.py", line 133, in get_session_class
    (interface_type, resource_class))
ValueError: No class registered for InterfaceType.gpib, INSTR
>>>

MatthieuDartiailh, 2020-08-24:
It appears that pyvisa cannot locate your installation of Keysight VISA (and similarly gpib-ctypes cannot find the gpib library). I suggest you search where did Keysight install the visa shared library (visa.so) and provide the full path to it when creating a ResourceManager. If you installed Keysight in the default location it suggested, we should add some notes in PyVISA documentation once we figure out how to fix your issue.

AbhayKrS-dev, 2020-08-24:
Thanks , I was able to open the resource when the path of pyvisa library was passed in resource manager initialization. 
Also yes, adding some documentation regarding this would be helpful in future.

MatthieuDartiailh, 2020-08-24:
Feel free to open a PR against PyVISA (this is unrelated to PyVISA-py). The best location for this information would be under configuring.rst I think. 
However we may also have some issues in our custom find_library function. Can you try the following ?
```python
import ctypes
print(ctypes.util.find_library("visa"))
```
I do not have access to a linux system right now but I will try to investigate.