pyvisa/pyvisa-py #256: PyVISA-Py: USB backend issue value error operation timed out

rakansha, 2020-09-06:
My software details for VISA are:
Machine Details:
   Platform ID:    Linux-4.4.0-28-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.11+
   Compiler:       GCC 5.3.1 20160413
   Bits:           64bit
   Build:          Apr 17 2016 14:00:29 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
## I am using py visa to connect to a Keysight 33500B Series waveform generator.I have already successfully used python VISA to connect to Tektronix oscilloscope as well as Teledyne LeCroy Waveform generator..But with Keysight I get an operation time out error on any query or write function..This is the what my input output session looks like:

> > > import visa
> > > rm=visa.ResourceManager('@py')
> > > rm.list_resources()
> > > (u'USB0::2391::11271::MY52814581::0::INSTR',)
> > > keysight=rm.open_resource('USB0::2391::11271::MY52814581::0::INSTR')
> > > keysight.query("*IDN?")
> > > Traceback (most recent call last):
> > >   File "<stdin>", line 1, in <module>
> > >   File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 401, in query
> > >     self.write(message)
> > >   File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 207, in write
> > >     count = self.write_raw(message.encode(enco))
> > >   File "/home/manish/.local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
> > >     return self.visalib.write(self.session, message)
> > >   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 271, in write
> > >     return self.sessions[session].write(data)
> > >   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
> > >     count = self.interface.write(data)
> > >   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
> > >     bytes_sent += raw_write(data)
> > >   File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
> > >     raise ValueError(str(e))
> > > ValueError: [Errno 110] Operation timed out
> > > ---

fgr1986, 2020-11-05:
Hi, same issue with Agilent u3606a.
```

pyvisa info:

Machine Details:                                                                                                                                                                                                      Platform ID:    Linux-5.4.72-v8+-aarch64-with-glibc2.17                                                                                                                                                            Processor:                                                                                                                                                                                                                                                                                                                                                                                                                         Python:                                                                                                                                                                                                               Implementation: CPython                                                                                                                                                                                            Executable:     /home/pi/miniforge3/envs/sitesting/bin/python                                                                                                                                                      Version:        3.8.6                                                                                                                                                                                              Compiler:       GCC 7.5.0                                                                                                                                                                                          Bits:           64bit                                                                                                                                                                                              Build:          Oct  7 2020 18:25:18 (#default)                                                                                                                                                                    Unicode:        UCS4                                                                                                                                                                                                                                                                                                                                                                                                               PyVISA Version: 1.10.1                                                                                                                                                                                                                                                                                                                                                                                                                Backends:                                                                                                                                                                                                             ni:                                                                                                                                                                                                                   Version: 1.10.1 (bundled with PyVISA)                                                                                                                                                                              Binary library: Not found                                                                                                                                                                                       py:                                                                                                                                                                                                                   Version: 0.3.1                                                                                                                                                                                                     ASRL INSTR: Available via PySerial (3.4)                                                                                                                                                                           USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1                                                                                                                                                           USB RAW: Available via PyUSB (1.1.0). Backend: libusb1                                                                                                                                                             TCPIP INSTR: Available                                                                                                                                                                                             TCPIP SOCKET: Available                                                                                                                                              [... cut ...]                                               GPIB INSTR:                                                                                                                                                                                                           Please install linux-gpib to use this resource type.                                                                                                                                                               No module named 'gpib'   
```

MatthieuDartiailh, 2020-11-05:
Please update to the latest pyvisa and pyvisa-py versions and report the exact error you see (since sometimes there are slight variations between systems).

fgr1986, 2020-11-05:
Min example with latest version of system (clean environment):

```
Machine Details:
   Platform ID:    Linux-5.4.72-v8+-aarch64-with-glibc2.17
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/pi/miniforge3/envs/pyvisa_test/bin/python3.8
   Version:        3.8.6
   Compiler:       GCC 7.5.0
   Bits:           64bit
   Build:          Oct  7 2020 18:25:18 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```
**Minimal example code**

```
import traceback
import pyvisa as visa

rm = visa.ResourceManager()
m = rm.open_resource('USB0::2391::19736::MY50099047::0::INSTR')

m.write_termination = '\n'
m.read_termination = '\n'
try:
    m.query('*IDN?')
except Exception as e:
    traceback.print_exc()
print('adding delay: ')
m.query_delay = .2
try:
    m.query('*IDN?')
except Exception as e:
    traceback.print_exc()

```
Output/Error:

```
Traceback (most recent call last):
  File "test.py", line 11, in <module>
    m.query('*IDN?')
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

adding delay:

Traceback (most recent call last):
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/protocols/usbtmc.py", line 256, in write
    return self.usb_send_ep.write(data)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/usb/core.py", line 406, in write
    return self.device.write(self, data, timeout)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/usb/core.py", line 977, in write
    return fn(
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/usb/backend/libusb1.py", line 837, in bulk_write
    return self.__write(self.lib.libusb_bulk_transfer,
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/usb/backend/libusb1.py", line 938, in __write
    _check(retval)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/usb/backend/libusb1.py", line 602, in _check
    raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 17, in <module>
    m.query('*IDN?')
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 638, in query
    self.write(message)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 543, in write
    written, status_code = self.sessions[session].write(data)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/usb.py", line 179, in write
    count = self.interface.write(data)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/protocols/usbtmc.py", line 436, in write
    bytes_sent += raw_write(data)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/protocols/usbtmc.py", line 258, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

```

MatthieuDartiailh, 2020-11-05:
Can you try to read just a single byte of the answer ? I would like to know if reading fail altogether or if we have a termination issue.

```python
import pyvisa as visa

rm = visa.ResourceManager()
m = rm.open_resource('USB0::2391::19736::MY50099047::0::INSTR')

m.write_termination = '\n'
m.read_termination = '\n'
m.write('*IDN?')
while True:
     print(m.read_bytes(1))
```

fgr1986, 2020-11-05:
@MatthieuDartiailh 

This is the output to your code:
```

Traceback (most recent call last):
  File "test.py", line 10, in <module>
    print(m.read_bytes(1))
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 371, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

MatthieuDartiailh, 2020-11-05:
Nothing printed before the error message ?

fgr1986, 2020-11-05:
No, that is all the output

MatthieuDartiailh, 2020-11-05:
Can you try a command that would produce a visible change in the instrument state ? We need to figure out if we cannot read the answer or if the instrument is not answering because it did not properly receive the message.

fgr1986, 2020-11-05:
```
>>> import pyvisa as visa                                                                                                                                                                                                                                                                                                                                                                 >>>                                                                                                                                                                                          >>> rm = visa.ResourceManager()                                                                                                                                                              >>> m = rm.open_resource('USB0::2391::19736::MY50099047::0::INSTR')                                                                                                                          >>>                                                                                                                                                                                          >>>                                                                                                                                                                                          >>> m.write_termination = '\n'                                                                                                                                                               >>> m.read_termination = '\n'                                                                                                                                                                >>> m.write('CONF:VOLT:DC')                                                                                                                                                                  13     
>>> 
```

MatthieuDartiailh, 2020-11-05:
Do you see the mode changing on the instrument ? If you send `"OUTP 1"` does the output turn on (assuming it was off) ?

fgr1986, 2020-11-05:
Unfortunately I am not at the lab. Will check it tomorrow. (Sorry about the inconvenience)

MatthieuDartiailh, 2020-11-05:
No problem. Debugging those kind of issues is always a pain and I would really like to make pyvisa-py better but I do not have that much open-source time those days.

fgr1986, 2020-11-06:
Hi @MatthieuDartiailh , After a few hard-resets on the device: write works (I can see it controlling the instrument, though the ERROR REMOTE appears in the display. Also, query works but the same ERROR Remote message appears in the display.

fgr1986, 2020-11-06:
```
`import pyvisa as visa

rm = visa.ResourceManager()
m = rm.open_resource('USB0::2391::19736::MY50099047::0::INSTR')

m.write_termination = '\n'
m.read_termination = '\n'
m.write('CONF:VOLT:DC')
m.write('sense:voltage:dc:range 10V')

print(m.query('*IDN?'))

m.write('MEAS:VOLT:DC?')
while True:
    print(m.read_bytes(1))`
```

Produces:

```
Agilent Technologies,U3606A,MY50099047,02.00-03.00-03.00
Traceback (most recent call last):
  File "test.py", line 16, in <module>
    print(m.read_bytes(1))
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 371, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/pi/miniforge3/envs/pyvisa_test/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

```

MatthieuDartiailh, 2020-11-06:
This is the point where things get really complicated... I am still confused by how some USB devices works flawlessly while some other like your cannot behave properly. 

If you want to keep digging, try to use Wireshark to spy on the transferred data using PyVISA-py and possibly using NI or Keysight implementation if you can install it. There must be something different. I will try to go again through the code and the USBTMC specification and see if we missed anything but it may take me some time.

fgr1986, 2020-11-06:
I will try to sniff it, though due to the actual circumstances I will not be able to access physically the instrument everyday.
Will use a normal x86 machine to use a different backend and see the differences.

In any case, thanks a lot for your support and all the work you put on the project.

MatthieuDartiailh, 2020-11-06:
You are welcome and I am sorry I cannot offer you a better solution.

fgr1986, 2021-03-01:
Hi @MatthieuDartiailh 
I am back at the lab, so probably can get the wireshark traces you asked for.
This issue also happens to keysight b2912a (actually, even before, on the instr.write):
```

pyvisa_py/protocols/usbtmc.py in write(self, data)
    256             return self.usb_send_ep.write(data)
    257         except usb.core.USBError as e:
--> 258             raise ValueError(str(e))
    259 
    260     def read(self, size):

ValueError: [Errno 110] Operation timed out
```
can you tell me what traces you need from wireshark?

thanks!

MatthieuDartiailh, 2021-03-01:
Ideally I would like to see everything from the opening of the instrument till the error. And if you can get the same in a working environment that would be great since I could compare both.

CompThing, 2021-03-01:
Hi Matthieu,
I am not sure if the attachment will help with this issue. I captured
some traces, that I think were for this issue in November 2019.
However, I did not get to discuss it before the contract that was using
the Signal Generator ended in January 2020.
The traces might be too old with an old version of pyvisa-py.
I think there are traces of the error with pyvisa-py and no problems
with usbtmc.
Hope it helps and is not a distraction with it simply being too old,
Michael Wilkinson
Compute Thing Limited

fgr1986, 2021-03-01:
Hi Matthieu, just send you an email with the traces.
Thanks

MatthieuDartiailh, 2021-03-01:
Thanks. However I cannot give you a timeline for this. I am a bit overwhelmed at the moment.

fgr1986, 2021-03-01:
Of course, and thanks again for your time!

ferdymercury, 2021-03-08:
Maybe this helps: https://github.com/python-ivi/python-usbtmc/pull/50

MatthieuDartiailh, 2021-03-09:
Thanks for sharing @ferdymercury .I honestly do not remember of I did such changes for pyvisa-py I have a vague recollection but I am not sure. Could you make a PR ? I am a bit under water at the moment.

mwcmwc12, 2022-02-08:
We are also running into this issue with a Keysight 33500B Series waveform generator. Has there been farther work on this? Is there anything we can do to help. Using python-usbtmc has no problems.

I took a look at the python-usbtmc fix but have no idea what the heck they are doing.

mwcmwc12, 2022-02-09:
The problem seems to lie with these two lines:
https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/protocols/usbtmc.py#L292
and the line right under it.

C/P here for clarity:

```
self.usb_dev.reset()
self.usb_dev.set_configuration()
```

Commenting out both of them and the timeout error disappears with the 33500B. We tested with a Keysight DMM and don't see a problem there either.

If you comment out just the reset, then you get a resource busy error. If you comment out just set_configuration, you get the time out error again. You must comment out both.

I have no idea what this impacts, but from reading some libusb docs, they say: 
**"You cannot change/reset configuration if your application has claimed interfaces. It is advised to set the desired configuration before claiming interfaces."**

I am not sure if this has anything to do with it? Maybe resetting and then trying to set configuration is going out of order. You don't reset, but set_configuration is already called once in the USBRAW class init, which USBTMC class calls
https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/protocols/usbtmc.py#L216

I am going to test with a non-Keysight device and see if anything weird happens

MatthieuDartiailh, 2022-02-09:
I must say I have no time for PyVISA beyond answering to people issues. If you can figure out how to at least mitigate this issue I will happily review a PR. 

It is perfectly possible that the logic got messed up at one point when refactoring those two classes and that a lack of deep enough understanding of libusb caused the issue. If you do make a PR, please add that kind of information as comment so that we do not regress in the future.

mwcmwc12, 2022-02-09:
Ok, let me do some additional testing with non-Keysight devices and see if any weird problems occur with commenting out those two lines. If things look good, I will create a PR for this problem.

pfjarschel, 2022-02-22:
Sorry to barge in this issue, but if it helps, I was having a very similar issue with a Tektronix oscilloscope (TBS1062). Just commenting out those two lines did not work, but if I also comment lines 216 and 221 (actually, the entire try/except blocks), it worked fine, and no problem was observed with other instruments from other manufacturers.

Line 221 was:
`self.usb_dev.set_interface_altsetting()`

It seems there is some kind of issue with the configuration settings and some very specific instruments.

MatthieuDartiailh, 2022-02-22:
Thanks for your input @pfjarschel . More feedback is always welcome.

Looking a bit at different things related libusb, it looks like we could avoid the situation you describe by calling set altsetting only if the interface has multiple settings which we can check. Since you have hardware available to test, could you make a PR or alternatively test one if I can find some time to make one ?

mwcmwc12, 2022-02-23:
Sorry for the silence, we have been swamped at work and couldn't work on this any further. Some added testing revealed this:
If we just comment out the two lines as previously stated:

`self.usb_dev.reset()
self.usb_dev.set_configuration()`

This works for the Keysight/Agilent Function generators, but won't work on our Tektronix MSO44 scope. Will cause a timeout problem. When we get time we can look into @pfjarschel 's fix as well when we have time.

MatthieuDartiailh, 2022-02-23:
Sorry, I didn't mean to blame you. I have often seen cases of people disappearing, so I just took the opportunity to have more people on board.

I think that setting the alt setting only when relevant as done here https://github.com/google/gousb/pull/38/files may really help. 

It is great you have two instruments to test this way; we may get a good fix rather than moving the issue around some more.

pfjarschel, 2022-03-03:
Hi, sorry for "disappearing" as well!
I had a short vacation, but now that I'm back, I'll try this!