pyvisa/pyvisa-py #261: TCP/IP connection problems

siriusy, 2020-09-07:
I try to use Pyvisa and Pyvisa-py to control several instruments through Ethernet. Everything works well except an Anritsu Signal Generator named MG3694C. 
I got the error :

    [Command: python3 -u /home/quantum/Downloads/MG3694C.py]
    Traceback (most recent call last):
      File "/home/quantum/Downloads/MG3694C.py", line 11, in <module>
        print(inst.query('*IDN?'))
      File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 558, in query
        self.write(message)
      File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 223, in write
        count = self.write_raw(message.encode(enco))
      File "/usr/local/lib/python3.6/dist-packages/pyvisa/resources/messagebased.py", line 201, in write_raw
        return self.visalib.write(self.session, message)
      File "/usr/local/lib/python3.6/dist-packages/pyvisa-py/highlevel.py", line 373, in write
        raise errors.VisaIOError(ret[1])
    pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

My code is simple, and the same code (different address) works well to all other instruments :

    import visa
    import numpy as np
    rm = visa.ResourceManager('@py')
    ipaddr = 'TCPIP0::192.168.0.254::INSTR'
    inst = rm.get_instrument(ipaddr)
    print(inst.query('*IDN?'))

I spent a lot of time searching online try to find a solution, but failed. The problem seems to be in the buffer size, like in [https://github.com/pyvisa/pyvisa/issues/174](url). However, the definition of the set_buffer function in pyvisa-py seems has not been finished. As follow:
    
    #/usr/local/lib/python3.6/dist-packages/pyvisa/highlevel.py
    def set_buffer(self, session, mask, size):
        """Sets the size for the formatted I/O and/or low-level I/O communication buffer(s).

        Corresponds to viSetBuf function of the VISA library.

        :param session: Unique logical identifier to a session.
        :param mask: Specifies the type of buffer. (Constants.VI_READ_BUF, .VI_WRITE_BUF, .VI_IO_IN_BUF, .VI_IO_OUT_BUF)
        :param size: The size to be set for the specified buffer(s).
        :return: return value of the library call.
        :rtype: :class:`pyvisa.constants.StatusCode`
        """
        raise NotImplementedError

More details of my computer:

    Machine Details:
       Platform ID:    Linux-4.15.0-39-generic-x86_64-with-Ubuntu-18.04-bionic
       Processor:      x86_64
    
    Python:
       Implementation: CPython
       Executable:     /usr/bin/python3
       Version:        3.6.7
       Compiler:       GCC 8.2.0
       Bits:           64bit
       Build:          Oct 22 2018 11:32:17 (#default)
       Unicode:        UCS4
    
    PyVISA Version: 1.10.0.dev0
    
    Backends:
       ni:
          Version: 1.10.0.dev0 (bundled with PyVISA)
          Binary library: Not found
       py:
          Version: 0.3.1
          ASRL INSTR: Available via PySerial (3.4)
          USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
          USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
          TCPIP INSTR: Available 
          TCPIP SOCKET: Available 
          GPIB INSTR:
             Please install linux-gpib to use this resource type.
             No module named 'gpib'

I'm new to visa. So how can I fix this issue? Can anyone help?

mnemocron, 2021-01-19:
I came here because I switched from USB to Ethernet due to missing USB drivers for ARM (Raspberry Pi) from Keysight.
So I am on the Raspberry Pi trying to connect to the Keysight 34465A without success.

@siriusy thank your for the hint with `vxi11` this one works like a charm

@MatthieuDartiailh see my code below, maybe it helps for debugging

It is not the same error. I get `Connection refused`
Note that I had to come up with the address `'TCPIP0::192.168.1.221::5000::SOCKET'` myself, because I don't have a connection manager on the raspberry pi. The IP is correct and the Keysight manual states port 5000 as standard.

- `pyvisa` `1.11.3`
- `pyvisa-py` `0.5.1`
- `python` `3.7.3`

### pyvisa variant

```python
python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa as visa
>>> rm = visa.ResourceManager()
>>> ADR = 'TCPIP0::192.168.1.221::5000::SOCKET'
>>> dmm = rm.open_resource(ADR)
>>> dmm.query('*IDN?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 638, in query
    self.write(message)
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa_py/highlevel.py", line 543, in write
    written, status_code = self.sessions[session].write(data)
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa_py/tcpip.py", line 625, in write
    size = self.interface.send(block)
ConnectionRefusedError: [Errno 111] Connection refused
```

### vxi11 variant

```python
python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import vxi11
>>> instr = '192.168.1.221'
>>> instr = vxi11.Instrument('192.168.1.221')
>>> print(instr.ask("*IDN?"))
Keysight Technologies,34465A,MY54505614,A.02.14-02.40-02.14-00.49-02-01
```

MatthieuDartiailh, 2021-01-19:
Can you try the following ? This will be closer to the vxi11 example since INSTR resources on TCPIP uses the vxi11 protocol by default. 
```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> dmm = rm.open_resource("TCPIP::192.168.1.221::INSTR")
>>> dmm.query('*IDN?')
```

mnemocron, 2021-01-19:
Hey, that worked! 
Thank you :D

MatthieuDartiailh, 2021-01-19:
Happy to help !

closed 2021-01-27 by MatthieuDartiailh