pyvisa/pyvisa-py #262: USB connection - Keithley works, SRS Lock-in timeout

Rromotpd, 2020-09-07:
I want to control both devices: Keithley multimeter and SRS Lock-in with Raspberry Pi. The issue is that Keithley works perfect - I can write commands / read data from device, but no matter what I do, my SRS Lock-in timeout in every write command.

```
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager("@py")
2019-04-29 15:41:01,053 - pyvisa - DEBUG - SerialSession was correctly imported.
2019-04-29 15:41:01,165 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2019-04-29 15:41:01,191 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2019-04-29 15:41:01,204 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2019-04-29 15:41:01,207 - pyvisa - DEBUG - Created library wrapper for unset
2019-04-29 15:41:01,212 - pyvisa - DEBUG - Created ResourceManager with session 6482890
>>> rm.list_resources()
('ASRL/dev/ttyAMA0::INSTR', 'USB0::46342::8192::003398::0::INSTR', 'USB0::1510::8464::8008202::0::INSTR')
############################# KEITHLEY MULTIMETER ####### 
>>> k=rm.open_resource('USB0::1510::8464::8008202::0::INSTR')
2019-04-29 15:41:23,342 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - opening ...
2019-04-29 15:41:23,631 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - is open with session 3820438
>>> k.query("*IDN?")
2019-04-29 15:41:51,899 - pyvisa - DEBUG - USB0::1510::8464::8008202::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
'KEITHLEY INSTRUMENTS INC.,MODEL 2110,8008202,02.02-02-01\n\x00\x00\x00'
######################################################
############################# SRS LOCK-IN ####### 
>>> s=rm.open_resource('USB0::46342::8192::003398::0::INSTR')
2019-04-29 15:42:20,745 - pyvisa - DEBUG - USB0::46342::8192::003398::0::INSTR - opening ...
2019-04-29 15:42:21,038 - pyvisa - DEBUG - USB0::46342::8192::003398::0::INSTR - is open with session 2920480
>>> s.query("*IDN?")
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 225, in write
    return self.usb_send_ep.write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/home/pi/.local/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 563, in query
    self.write(message)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa/resources/messagebased.py", line 201, in write_raw
    return self.visalib.write(self.session, message)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 368, in write
    ret = self.sessions[session].write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/usb.py", line 126, in write
    count = self.interface.write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 314, in write
    bytes_sent += raw_write(data)
  File "/home/pi/.local/lib/python3.5/site-packages/pyvisa-py/protocols/usbtmc.py", line 227, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
######################################################
```

Any suggestions how can I establish connection? So far I've tried:
1. Change read_termination and write_termination for: \r \n \r\n, None for Lock-in, 
2. Switch USB cables,
3. Reset Lock-in to default settings

Nothing works, I'm clueless..

Connection issues are only on Raspberry - on PC with Windows and NI-VISA drivers installed, my scripts for SRS Lock-in work as they should. But it is so weird that Keithley works perfect on both Raspberry and PC.

mnemocron, 2021-01-18:
I am having a similar issue with a Keysight device. 
The code from my Windows installation does not work on a Raspberry Pi installation.
Another device works fine on both plattforms.

On Windows I do not use the `pyvisa-py` backend and the Keysight DMM works. I would conclude that it is not a device / cable issue.

- Keysight 34465A (Voltmeter)    - does not work
- Manson NTP 6531 (Power Supply) - works perfectly

|          | Windows 10 | Raspberry Pi |
|----------|------------|--------------|
| python version | 3.8.5 (Anaconda) | 3.7.3 |
| pyvisa version | 1.11.3 |  1.11.3 |
| pyvisa-py version | - | 0.5.1 |
| pyvisa-py backend | `visa.ResourceManager('@ivi')` | `visa.ResourceManager('@py')`
| NTP 6531 | mansonlib.py works | pyvisa or mansonlib.py works |
| Keysight 34465A | pyvisa works | **does not work** |

#### Code on Windows 10 (Spyder / Anaconda)

- works
- the Keysight DMM does display "Hello"

```python
import pyvisa as visa

VISA_ADDRESS = 'USB0::0x2A8D::0x0101::MY54505614::0::INSTR'

resourceManager = visa.ResourceManager()
print(resourceManager.list_resources())
dmm = resourceManager.open_resource(VISA_ADDRESS)

dmm.write('*IDN?')
idn = dmm.read()
print(idn)
dmm.write('DISP:TEXT "Hello"')
```

#### Code on Raspberry Pi

- does not work
- read timeout
- Keysight DMM does not display any text

```python
>>> import pyvisa as visa
>>> visa.log_to_screen()
>>> resourceManager = visa.ResourceManager("@py")
2021-01-18 13:50:08,639 - pyvisa - DEBUG - Created ResourceManager with session 6689986
>>> VISA_ADDRESS = 'ASRL/dev/ttyAMA0::INSTR'
>>> dmm = resourceManager.open_resource(VISA_ADDRESS)
2021-01-18 13:50:18,937 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - opening ...
2021-01-18 13:50:18,960 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - is open with session 5702442
>>> dmm.write('*IDN?')
2021-01-18 13:51:06,553 - pyvisa - DEBUG - Serial.write b'*IDN?\r\n'
7
>>> idn = dmm.read()
2021-01-18 13:51:20,106 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2021-01-18 13:51:22,125 - pyvisa - DEBUG - ASRL/dev/ttyUSB0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/pi/.local/lib/python3.7/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>>> dmm.write('DISP:TEXT "Hello"')
2021-01-18 13:51:51,421 - pyvisa - DEBUG - Serial.write b'DISP:TEXT "Hello"\r\n'
19
>>> dmm.write('DISP:TEXT "Hello"\r')
2021-01-18 13:52:02,846 - pyvisa - DEBUG - Serial.write b'DISP:TEXT "Hello"\r\r\n'
20
```

MatthieuDartiailh, 2021-01-18:
From your Windows example it appears that your device uses the USBTMC protocol. However on Linux, you access to it as a simple serial instrument which is incorrect. Does Pyvisa-py detects your USB devices when you list the available resources ? Did you read the documentation regarding the edition of the udev rules which is often required for USB devices on Linux ?

mnemocron, 2021-01-18:
The examples from [keysight.com](https://www.keysight.com/main/editorial.jspx?cc=CH&lc=ger&ckey=2798637&nid=-35489.384515&id=2798637) use a serial class device too. How can you tell from the Windows example that it is USBTMC ?

And the resource manager finds both devices

```
resourceManager.list_resources()
```

```
('ASRL/dev/ttyUSB0::INSTR', 'ASRL/dev/ttyAMA0::INSTR')
```

where `USB0` is the NTP-6531 and `AMA0` is the Keysight 34465A

MatthieuDartiailh, 2021-01-19:
The fact that the VISA address on Windows is `USB0::0x2A8D::0x0101::MY54505614::0::INSTR` tells me it is a USBTMC device rather than a serial emulation over USB. I do not have much experience with USBTMC on Linux but I do know tat they cannot be accessed as serial resources. 
Can you use Connection Expert on Linux ? If so can you use use to communicate with your device and if yes with what address ?

mnemocron, 2021-01-19:
Oh man.. drivers. Thanks a lot for the hint.
So I assume this device (or newer Keysight devices in general) cannot be controlled with `pyvisa` __and__ USB __and__ ARM architecture?

---

>  ### Does IO Libraries Suite for Linux support 32-bit Linux OSes or Linux built on the ARM CPU architecture?
> No, it does not. IO Libraries Suite for Linux is only built to work on 64-bit Linux OSes and the types of these OSes are listed in the ReadMe. ARM CPU architecture is not supported and thus IO Libraries Suite cannot be installed on these examples of Linux OSes or hardware: Raspbian and other Raspberry Pi Linux OSes, Ubuntu MATE. The suggestion for using Raspberry Pi to communicate with instruments is to use the common TCP/IP-Socket based libraries with instruments that have the LAN interface.

[keysight FAQ](https://www.keysight.com/main/editorial.jspx?cc=CH&lc=ger&ckey=3013040&nid=-34356.800136&id=3013040)

MatthieuDartiailh, 2021-01-19:
Technically you do need a Keysight specific driver, but simply a generic USBTMC driver. However I do not know if one exists for Linux on ARM.

My suggestion to use Connection Expert was simply a possible way to debug the system.

mnemocron, 2021-01-26:
Quick follow up. I managed to get it working with the `usbtmc` python library.

```python
import usbtmc
instr = usbtmc.Instrument(0x2a8d, 0x0101)
instr.ask("*IDN?")
```

MatthieuDartiailh, 2021-01-26:
So even though PyVISA-py fail to list your instrument it can still be accessed through USBTMC. Can you then try the same using pyvisa-py and the address `USB::0x2A8D::0x0101::MY54505614::INSTR` which is the generic version of the address you ued on Windows ?

mnemocron, 2021-01-26:
I see, this is what worked on the Windows originally.

> ```python
> VISA_ADDRESS = 'USB0::0x2A8D::0x0101::MY54505614::0::INSTR'
> ```

I just tried it again on the Raspberry Pi. Oddly enough, now the resource manager actually lists the device correctly.
This made me curious and I uninstalled `usbtmc` and tried again. It still works.

```python
import pyvisa as visa
rm = visa.ResourceManager()
rm.list_resources()
# ('ASRL/dev/ttyAMA0::INSTR', 'USB0::10893::257::MY54505614::0::INSTR')
dmm = rm.open_resource('USB0::10893::257::MY54505614::0::INSTR')
dmm.query('*IDN?')
# 'Keysight Technologies,34465A,MY54505614,A.02.14-02.40-02.14-00.49-02-01\n'
dmm.close()
dmm = rm.open_resource('USB::0x2A8D::0x0101::MY54505614::INSTR')
dmm.query('*IDN?')
# 'Keysight Technologies,34465A,MY54505614,A.02.14-02.40-02.14-00.49-02-01\n'
```

MatthieuDartiailh, 2021-01-26:
Did you anything special when installing the usbtmc package ? I am wondering if it may have touch a configuration file or something.