pyvisa/pyvisa-py #263: Not getting complete data with read_raw()

torquil, 2020-09-07:
Hi! I'm using the development version of pyvisa and pyvisa-py together with a Siglent SDS1104X-E oscilloscope via USB on Debian Linux Unstable. I'm trying to retrieve the channel 1 waveform from the scope. When I used lxi-tools via ethernet this worked fine, but it caused my scope to freeze up quite often.

The command to request the waveform is:

oscilloscope.write('C1:WF? ALL')

which uses the SCPI command that worked fine with lxi-tools.

However, when I subsequently invoke read_raw(), I'm not getting the complete data. I only get around 60-70 bytes. By running read_raw() many times, I seem to get a sensible amount of data corresponding to the number of displayed points, but I'm not sure if the data is completely correct, because I'm having some trouble extracting variable values from the second or later segments.

When running many times manually, I have to manuall test each returned segment for the correct termination characters for the Siglent, which is b'\n\n'

So my question is: why is read_raw() stopping early, and what can I do to make it receive all the data?

The scope firmware is 8.1.6.1.26.

Best regards,
Torquil Sørensen

sigurdsvela, 2022-06-24:
Was this issue ever solved? I'm having the exact same issue (WF retrieval using pyvisa-py on my SDS1104)

MatthieuDartiailh, 2022-06-24:
Could you open a new issue for this with all the details requested in the template ?

WillMatthews, 2022-06-25:
As in @torquil 's original issue message, I also I tried using `read_raw`, but that itself had an issue where it halted prematurely (#318). I'm not sure if it's relevant here, but I will look at updating that issue in the next few days to help make progress.

MatthieuDartiailh, 2022-06-25:
The USB issue is a pain but I do not have the bandwidth to look at it. Does reading one byte at a time using read_bytes helps ?

sigurdsvela, 2022-06-26:
When trying `read_bytes(1)` the program just hung forever on the first call.

However, when using `read_bytes(1)` with TCP i get the entire message(Which seems weird).  
read_raw still crashes though with error `VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

Worth noting that everything seems to be working perfectly with the vxi11 python package.

Details: Mac Pro M1, Python 3.9, pyvisa@1.12.0, pyvisa-py@0.5.3

MatthieuDartiailh, 2022-06-26:
I kind of remember the read_bytes(1) issue... I will try to look. However could you tell me what the message actually ends on since you claim it is complete ? Does it by any chance does not end on a termchar ? That would explain read_raw erroring.

sigurdsvela, 2022-06-26:
Checked now, and the message ends with a single newline.  
I'm not quite sure what the default termchar is, but adding adding `inst.read_termination = '\n'` yielded the same error.

sigurdsvela, 2022-06-29:
Soo, I'm not sure whats happening, but I'm no longer getting the timeout.
While using the scope I used the "default" button to reset to default settings. I'm not sure if thats the reason, but thats the only thing I can think of that has changed.

I am, however, getting what looks like scrambled data when using read_raw over usb.

This is what I get when using TCP (Correct. Matches what is on the display)
[image]

And this is what I get when I try to use USB
[image]

When using USB I also get the following warning

    <...PATH...>/pyvisa_py/protocols/usbtmc.py:115: UserWarning: Unexpected MsgID format. Consider updating the device's firmware. See https://github.com/pyvisa/pyvisa-py/issues/20
    warnings.warn(

<details>
<summary>Code</summary>

    import pyvisa
    from matplotlib import pyplot as plt

    rm = pyvisa.ResourceManager('@py')

    sds = rm.open_resource('USB0::62700::60984::SDSMMDBD2R4502::0::INSTR')
    #sds = rm.open_resource('TCPIP::192.168.42.70::INSTR')

    sds.write("C1:WF? DAT2")
    result = bytearray(sds.read_raw())

    # Result mapping per the Siglent SDS Remote Programming manual
    result = [r if r <= 128 else r - 255 for r in result]

    plt.plot(result)

</details>

MatthieuDartiailh, 2022-06-29:
This may be a combination of you reading binary data using read_raw which you should not do (please look at the documentation of read_binary_data and if your instrument uses a weird format ask here) and the TCPIP not honoring the termination character somehow. I will try to have a look at the TCPIP issue, in the meantime could you see if things works better when you use a method ensuring you get all the binary data even though it contains the termchar ?

sigurdsvela, 2022-06-30:
For read_binary_values I get a `Pipe Error`

<details>
    <summary>
        Traceback
    </summary>

    USBTimeoutError                           Traceback (most recent call last)
    File ./envs/lib/python3.9/site-packages/pyvisa_py/protocols/usbtmc.py:464, in USBTMC.read(self, size)
        463 try:
    --> 464     resp = raw_read(recv_chunk + header_size + max_padding)
        465     response = BulkInMessage.from_bytes(resp)
    
    File ./envs/lib/python3.9/site-packages/pyvisa_py/protocols/usbtmc.py:271, in USBRaw.read(self, size)
        269     size = 1
    --> 271 data = self.usb_recv_ep.read(size, self.timeout).tobytes()
        273 return data
    
    File ./envs/lib/python3.9/site-packages/usb/core.py:423, in Endpoint.read(self, size_or_buffer, timeout)
        411 r"""Read data from the endpoint.
        412 
        413 The parameter size_or_buffer is either the number of bytes to
       (...)
        421 For details, see the Device.read() method.
        422 """
    --> 423 return self.device.read(self, size_or_buffer, timeout)
    
    File ./envs/lib/python3.9/site-packages/usb/core.py:1029, in Device.read(self, endpoint, size_or_buffer, timeout)
       1027     buff = util.create_buffer(size_or_buffer)
    -> 1029 ret = fn(
       1030         self._ctx.handle,
    ...
        603     else:
    --> 604         raise USBError(_strerror(ret), ret, _libusb_errno[ret])
        606 return ret
    
    USBError: [Errno 32] Pipe error
</details>

The instrument does use, what looks to me like, kinda weird formating. But I'm not sure whats common.

Per the manual
> The head of message: C1:WF ALL. These are followed by the string #900000070, the beginning of a binary block in which nine ASCII integers are used to give the length of the block (70 bytes).After the length of block, is beginning of wave data. At the last meet ―0A 0A‖, means the end of data.
> [image]

So it uses ascii values to give the length of the data in the header of the message. That part is however not included with the WF? DAT2 command, which only returns the data.

The message should end on hex 0A 0A, double newline. Maybe thats the issue? That 0A is showing up in the data?
I cant see in the manual any mention of how it is "escaped" if it shows up in the data. As in, what if the waveform data contains 0x0A0A at some point?

I think the most pressing issue might be that read_bytes is not working(Over USB it gives the same pipe error, and over TCP it returns the entire message) If that worked, I would be able to implement any weird formatting myself.

MatthieuDartiailh, 2022-06-30:
Some instrument omit the term char after a binary block there is no standard (I have seen different instruments from the same manufacturer behave differently). 

In the case of `WF? DAT2`, there is not even a IEEE header ? If so then you have to assume the same size as `WF? DAT1` and read all the data manually (if read_bytes fail you can use read_raw repeatedly till you get all the data).

My suggestion to alter the term char (which is actually `read_termination` sorry) was simply so that read_binary_values read all the characters and does not leave a feedline unread.

dernasherbrezon, 2024-06-30:
Found the similar issue with Rigol DS1054Z. (PyVISA==1.14.1). When I open device like this:

```python
_rigol = _resource_manager.open_resource('USB0::0x1AB1::0x04CE::DS1ZA253502774::INSTR', write_termination='\n', read_termination='\n')
```

PyVisa correctly splits output by first '\n', but if second '\n' found in the stream, it will output it during ```read_raw```. Can it be caused by two consecutive \n\n in the stream?