pyvisa/pyvisa-py #264: pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

mdhayath64, 2020-09-07:
import pyvisa
rm=pyvisa.ResourceManager()
sg=rm.open_resource('TCPIP0::192.168.1.254::inst0::INSTR')
sg.query('*IDN?')

when i use these code iam gettin dat error

amritphy, 2022-03-09:
Is there any solution provided on this? I am trying to communicate with the Keysight oscilloscope to extract the waveform from it. I am successfully able to extract the measurement table using a query but querying for the waveform always raises this error.
I have tried the following:
```
import pyvisa as visa
rm=visa.ResourceManager()
scope=rm.open_resource('TCPIP::192.168.0.100::INSTR')
x = scope.query('WAVeform:XYFormat:ASCii:XDATa?')
```
This raises the following error
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/astria/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/home/astria/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/astria/anaconda3/lib/python3.6/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/astria/anaconda3/lib/python3.6/site-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/home/astria/anaconda3/lib/python3.6/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

```
In fact, no query that asks about the waveform goes through. like - `scope.query('WAVeform:STATus?')`
Please guide what's going wrong.

MatthieuDartiailh, 2022-03-09:
Can you post the exact model of your instrument and a link to its manual if possible ? Can you also post the output of pyvisa-info ?

amritphy, 2022-03-10:
Hi, 
The exact model of the information are as follows
```
Keysight DSO-S 404A 4GHz, 4 Analog channel 20GSa/s and 10-bit ADC (Infiniium S-series)
The link to the manual is : https://www.keysight.com/zz/en/assets/9018-70007/user-manuals/9018-70007.pdf?success=true
```

the pyvisa-info results in the following.
```
Machine Details:
   Platform ID:    Linux-3.10.0-862.el7.x86_64-x86_64-with-centos-7.5.1804-Core
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/astria/anaconda3/bin/python
   Version:        3.6.5
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          Apr 29 2018 16:14:56 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```
Please let me know if you need any further info

MatthieuDartiailh, 2022-03-10:
You may be hitting a bug in the VXI-11 implementation of pyvisa-py. Could you try using a raw socket connection on port 5025 ?

mongolracer, 2022-03-10:
I have the same issue right now. My instrument answers only to *IDN?, but all other commands results in the error:

VI_ERROR_IO (-1073807298): Could not perform operation because of I/O error.

Must be a bug ?

MatthieuDartiailh, 2022-03-11:
For Keysight instrument you can open the resource as a RAW sockect using the following format of resource name `TCPIP::{your instrument IP}::5025::SOCKET`.

Could you also post a minimal script exhibiting the issue ? I may be able to see if you are doing something off. 

@mongolracer  can you also post the output of `pyvisa-info` along with your instrument model and a link to its documentation ?

MatthieuDartiailh, 2022-03-11:
@amritphy checking the programming manual for the infinium S series it appears that waveform command are expected to work only when following a DIGitize command. Do you issue a DIGitize command in your script ?

amritphy, 2022-03-11:
> For Keysight instrument you can open the resource as a RAW sockect using the following format of resource name `TCPIP::{your instrument IP}::5025::SOCKET`.
>
> Could you also post a minimal script [...]

The script that I am using is as follows
```
import pyvisa as visa
import numpy as np
import matplotlib.pyplot as plt

rm = visa.ResourceManager('@py')
scope = rm.open_resource('TCPIP::192.168.0.100::INSTR')
print(scope.query('*IDN?'))
scope.timeout = 30000 #30 seconds

# Setting source as Channel 1
scope.write('DATA:SOU CH2')
scope.write('DATA:WIDTH 1')
scope.write('DATA:ENC ASCI')

scope.write('DATA:RESOLUTION FULL')

ymult = float(scope.query('WFMPRE:YMULT?')) # y-axis least count
yzero = float(scope.query('WFMPRE:YZERO?')) # y-axis zero error
yoff = float(scope.query('WFMPRE:YOFF?')) # y-axis offset
xincr = float(scope.query('WFMPRE:XINCR?')) # x-axis least count
xoff = float(scope.query('WFMP:PT_OFF?')) # x-axis offset
xzero = float(scope.query('WFMPRE:XZERO?')) # x-axis least count

print('Completed getting axis information')

ADC_wave = scope.query_ascii_values('CURV?', container=np.array)
print(ADC_wave)
#data
volts = (ADC_wave - yoff) * ymult + yzero
time = np.arange(xzero, xincr * len(Volts) - xoff + xzero, xincr)

plt.plot(time,volts)

```