pyvisa/pyvisa-py #267: 'Resource' object has no attribute 'write'

BDuons, 2020-09-23:
Hi,
I've just updated to the latest version of pyvisa and now get the following error:

```
Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:37:02) [MSC v.1924 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> rm.list_resources()
('ASRLCOM3::INSTR', 'ASRLCOM4::INSTR')
>>> my_instrument = rm.open_resource('ASRLCOM3::INSTR')
>>> print(my_instrument.query('*IDN?'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Resource' object has no attribute 'query'
```

I've noted this unusual output:
```
>>> print(my_instrument)
Resource at ASRLCOM3::INSTR

```
expected behaviour on readthedocs is along the lines of:
`<GPIBInstrument('GPIB::14')>`

*pyvisa-info*
PS C:\Users\mainuser\Documents\Python\sm1800> pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\mainuser\appdata\local\programs\python\python38\python.exe
   Version:        3.8.3
   Compiler:       MSC v.1924 64 bit (AMD64)
   Bits:           64bit
   Build:          May 13 2020 22:37:02 (#tags/v3.8.3:6f8c832)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2020-09-23:
I transferred the issue @BDuons. The new error you are seeing can come from many different things including improperly configured termination characters, or improper serial parameters (baud_rate etc). Please have a look at https://pyvisa.readthedocs.io/en/latest/introduction/communication.html to get it right.

Note that not all instrument supports `*IDN?` and furthermore they may require different configuration and as a consequence this is far from a reliable method.

MatthieuDartiailh, 2020-09-23:
I created #269 that should address that issue. If anybody that can check it works on a windows it would be great.

MatthieuDartiailh, 2020-09-23:
Also note that with https://github.com/pyvisa/pyvisa/pull/547 in, you should not need to specify the resource class anymore.

greyltc, 2020-09-25:
> Also note that with pyvisa/pyvisa#547 in, you should not need to specify the resource class anymore.
When can we expect this is in a stable release? This bug broke all my stuff.

MatthieuDartiailh, 2020-09-25:
I imagine and I am sorry about that. I am planning to release the bugfix next Wednesday and if people manage to test #269 before then I will also make a pyvisa-py release.

greyltc, 2020-09-25:
Cool, thanks!

commit fcead3fd (2020-09-30): Merge #269

269: serial: trim COM from serial ports and add it when opening r=MatthieuDartiailh a=MatthieuDartiailh

- [x] Closes #267
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] Added an entry to the CHANGES file
PR #269 (2020-09-30): serial: trim COM from serial ports and add it when opening
- [x] Closes #267
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] Added an entry to the CHANGES file
closed 2020-09-30 by bors[bot]