pyvisa/pyvisa-py #270: [COM] Communication issue with Anritsu MS4644B using USB

ercanal, 2020-09-23:
Instrument details
------------------
* Model: Anritsu MS4644B
* Communication: USB
* Link to the documentation (if available): [Programming Manual](https://dl.cdn-anritsu.com/en-us/test-measurement/files/Manuals/Programming-Manual/10410-00322Y.pdf)

Output of `pyvisa-info`
-----------------------

I am trying to use Anritsu MS4644B from raspberry PI with pyvisa-py but I can't communicate with the device except opening the connection

```
pi@raspberrypi:~ $ .local/bin/pyvisa-info
Machine Details:
   Platform ID:    Linux-4.19.118-v7+-armv7l-with-debian-10.4
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Dec 20 2019 18:57:59 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.0

Backends:
   ivi:
      Version: 1.11.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.0
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

python script with debug logging  enabled
```
>>> import logging
>>> logging.basicConfig(format='%(levelname)s:%(message)s', level=logging.DEBUG)
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
DEBUG:SerialSession was correctly imported.
DEBUG:USBSession and USBRawSession were correctly imported.
DEBUG:TCPIPSession was correctly imported.
DEBUG:GPIBSession was not imported No module named 'gpib'.
DEBUG:Created library wrapper for py
DEBUG:Created ResourceManager with session 2328881
>>> address = "ASRL/dev/ttyAMA0::INSTR"
>>> inst = rm.open_resource(address)
WARNING:There is no class defined for (<InterfaceType.unknown: -1>, None). Using Resource
DEBUG:ASRL/dev/ttyAMA0::INSTR - opening ...
DEBUG:ASRL/dev/ttyAMA0::INSTR - is open with session 1058748
>>> print(inst.query("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Resource' object has no attribute 'query'
```

attribute list from pyvisa-shell
```
pi@raspberrypi:~ $ .local/bin/pyvisa-shell

Welcome to the VISA shell. Type help or ? to list commands.

(visa) list
( 0) ASRL/dev/ttyAMA0::INSTR
(visa) open 0
ASRL/dev/ttyAMA0::INSTR has been opened.
You can talk to the device using "write", "read" or "query".
The default end of message is added to each message.
(open) attr
+---------------------------+------------+----------------------------+-------------------------+
|         VISA name         |  Constant  |        Python name         |           val           |
+---------------------------+------------+----------------------------+-------------------------+
|   VI_ATTR_INTF_INST_NAME  | 3221160169 |                            |    VI_ERROR_NSUP_ATTR   |
|      VI_ATTR_INTF_NUM     | 1073676662 |      interface_number      |    VI_ERROR_NSUP_ATTR   |
|     VI_ATTR_INTF_TYPE     | 1073676657 |       interface_type       |    InterfaceType.asrl   |
|  VI_ATTR_MAX_QUEUE_LENGTH | 1073676293 |                            |    VI_ERROR_NSUP_ATTR   |
|     VI_ATTR_RSRC_CLASS    | 3221159937 |       resource_class       |          INSTR          |
| VI_ATTR_RSRC_IMPL_VERSION | 1073676291 |   implementation_version   |    VI_ERROR_NSUP_ATTR   |
|  VI_ATTR_RSRC_LOCK_STATE  | 1073676292 |         lock_state         |    VI_ERROR_NSUP_ATTR   |
|    VI_ATTR_RSRC_MANF_ID   | 1073676661 |                            |    VI_ERROR_NSUP_ATTR   |
|   VI_ATTR_RSRC_MANF_NAME  | 3221160308 | resource_manufacturer_name |    VI_ERROR_NSUP_ATTR   |
|     VI_ATTR_RSRC_NAME     | 3221159938 |       resource_name        | ASRL/dev/ttyAMA0::INSTR |
| VI_ATTR_RSRC_SPEC_VERSION | 1073676656 |        spec_version        |    VI_ERROR_NSUP_ATTR   |
|     VI_ATTR_TMO_VALUE     | 1073676314 |          timeout           |           2000          |
|     VI_ATTR_USER_DATA     | 1073676298 |                            |    VI_ERROR_NSUP_ATTR   |
+---------------------------+------------+----------------------------+-------------------------+
(open)
```

MatthieuDartiailh, 2020-09-23:
Thanks for your report. You should get an instance of SerialInstrument not a bare Resource. I will try to investigate what is happening. In the meantime you can force pyvisa to use a SerialInstrument as follow:

``` python
import pyvisa
from pyvisa.resources.serial import SerialInstrument
rm = pyvisa.ResourceManager()
inst = rm.open_resource("ASRLCOM3::INSTR, resource_pyclass=SerialInstrument)
```

MatthieuDartiailh, 2020-09-23:
I figured out the root cause of your issue and will try to address it shortly, you can follow the discussion at #269

MatthieuDartiailh, 2020-09-23:
Also note that with https://github.com/pyvisa/pyvisa/pull/547 in, you should not need to specify the resource class anymore.

MatthieuDartiailh, 2020-10-01:
Fixed in the latest bugfix releases

closed 2020-10-01 by MatthieuDartiailh