pyvisa/pyvisa-py #272: sending ints into pyserial's write() function is not good?

greyltc, 2020-09-27:
My serial instrument communication is broken and I think I've just figured out why.

https://github.com/pyvisa/pyvisa-py/blob/06511146938631a7c8984b5c33405b14bb28a171/pyvisa_py/serial.py#L203
^ in my setup, this line sends `int`s into pyserial's `write()` function.

[The docs for that](https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.write) don't mention that sending an int to that is okay.

Have a look at the count value that's coming back. It's wrong. I haven't gone so far as to sniff the actual bus to see if anything is being sent on it, but none of my serial instruments are working today so that's some evidence that serial writes are broken.

I'm not able to follow the need for the `iter_bytes` and `SerialTermination.last_bit` and `end_out` stuff above (I guess I've never used any of those features?). I thought all the termination character stuff was handled at a higher level. I assume whatever that is, it's driving the desire to iterate out serial comms messages one byte at a time. Anyway, I think it's bad for several reasons to write one byte at a time, so I suggest sending the whole message in one go unless there's a very good reason not to.

greyltc, 2020-09-27:
I've made https://github.com/pyvisa/pyvisa-py/pull/273 which contains a fix that works for me.

MatthieuDartiailh, 2020-09-28:
Sorry I messed that up when removing the old Python 2 compat and I forgot how iterating on bytes work... testing serial resources is always a bit of hassle for me sorry for the inconvenience.

commit 281887d9 (2020-09-29): Merge #277

277: serial: fix writing to serial resources r=MatthieuDartiailh a=MatthieuDartiailh

- [x] Closes #272 
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [ ] Added an entry to the CHANGES file
commit 12e95968 (2020-09-29): Merge #277

277: serial: fix writing to serial resources r=MatthieuDartiailh a=MatthieuDartiailh

- [x] Closes #272 
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [ ] Added an entry to the CHANGES file
PR #277 (2020-09-29): serial: fix writing to serial resources
- [x] Closes #272 
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [ ] Added an entry to the CHANGES file
closed 2020-09-29 by bors[bot]