pyvisa/pyvisa-py #274: GPIB interface_clear call

greyltc, 2020-09-27:
In the past, I thought I was able to do
```
sm = rm.open_resource("GPIB0::24::INSTR")
sm.send_ifc()
```
Should that work? Or is there another way I should be doing that?

I know that it's not strictly proper to send the IFC signal to an instrument object, but I thought that under the hood that got redirected to the actual GPIB bus controller object that the instrument is connected to so that this worked.

MatthieuDartiailh, 2020-09-28:
VISA does not support send_ifc on INSTR resources (actually the current behavior of PyVISA is misleading and will be fixed). You need to access the interface object using 
```python
ifc = rm.open_resources("GPIB0::INTFC")
```

MatthieuDartiailh, 2020-09-28:
See https://github.com/pyvisa/pyvisa/pull/550 for what I mean about fixing PyVISA.

closed 2020-10-01 by MatthieuDartiailh