pyvisa/pyvisa-py #278: gpib_control_ren can't be called on INTFC objects

greyltc, 2020-09-30:
``` python
$ cat -n smoltest.py 
     1	import pyvisa
     2	rm = pyvisa.ResourceManager('@py')
     3	vl = rm.visalib
     4	
     5	d = rm.open_resource('GPIB0::24::INSTR')
     6	print(d.query('*IDN?').strip())
     7	d.write('*RST')
     8	d.close()
     9	
    10	b = rm.open_resource('GPIB0::INTFC')
    11	vl.gpib_control_ren(b.session, pyvisa.constants.VI_GPIB_REN_ADDRESS_GTL)
    12	b.close()
```
```bash
$ python smoltest.py 
KEITHLEY INSTRUMENTS INC.,MODEL 2420,1123724,C30   Mar 17 2006 09:29:29/A02  /H/J
Traceback (most recent call last):
  File "smoltest.py", line 11, in <module>
    vl.gpib_control_ren(b.session, pyvisa.constants.VI_GPIB_REN_ADDRESS_GTL)
  File "/usr/lib/python3.8/site-packages/pyvisa_py/highlevel.py", line 325, in gpib_control_ren
    return self.handle_return_value(
  File "/usr/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_OPER (-1073807257): The given session or object reference does not support this operation.
```

MatthieuDartiailh, 2020-09-30:
As per VISA documentation INTFC only support the following modes for control_ren,
                constants.VI_GPIB_REN_ASSERT,
                constants.VI_GPIB_REN_DEASSERT,
                constants.VI_GPIB_REN_ASSERT_LLO

greyltc, 2020-09-30:
`.command()` can't even be called on INSTRs, right?

MatthieuDartiailh, 2020-09-30:
command can only be called on INTFC indeed.

greyltc, 2020-09-30:
`gpib_control_ren()` can call `.command()` in three places in the code that's in the master branch right now.

Only one of those places can be reached when it's called on an `INTFC` object.
So the other two cases must be error cases then (they both seem to be trying to be doing 0x01 commands).

Seems like there should be a bus level GTL allowed on INTFC objects (issues `controller.command(b"\x01")`) and a device level GTL (calls `interface.ibloc()`). Should those two things be in `gpib_control_ren()`? I don't know.

MatthieuDartiailh, 2020-09-30:
Regarding the interface you can emit a global GTL, it is simply not covered as part of control ren in the VISA specification. 

Regarding the current master branch, I agree it is heavily incorrect and comes from my misunderstanding of some concepts at the time of writing and my incapacity to test the changes. I have some time tonight and I will try to provide a correct version for you to test. In the meantime I will release pyvisa 1.11.1 and pyvisa-py 0.5.1 since the breakage on serial resources are very serious. I will include the fixes to control ren in a 0.5.2.

MatthieuDartiailh, 2020-10-01:
For reference this is what the specification says:
[image]
Spending some time reading the linux-gpib source code I am understanding better how to do that. I pushed a bare-bone implementation (incomplete enum of commands and no support for secondary address) but if you can test it if you would be great. I have wanted to add better support for GPIB commands to pyvisa for some time and it seems this might just give me enough insight to do it.