pyvisa/pyvisa-py #281: Timeout Error on query

aboddeti, 2020-10-09:
I am remotely controlling a Keysight Power supply, model# E3649A using pyvisa on a Raspberry Pi. 
Until recently, it worked without any error. 

Unfortunately, currently I am getting a timeout error. 

Below is the information of my pyvisa
`
Machine Details:
   Platform ID:    Linux-4.19.66-v7+-armv7l-with-debian-10.1
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.16
   Compiler:       GCC 8.2.0
   Bits:           32bit
   Build:          Apr  6 2019 01:42:57 (#default)
   Unicode:        UCS4

PyVISA Version: 1.10.1

Backends:
   ni:
      Version: 1.10.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.1
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

`
and my python script is below:

`mport time
import visa

rm = visa.ResourceManager()
res = rm.list_resources()
print(res)

inst = rm.open_resource("ASRL/dev/ttyUSB0::INSTR")
time.sleep(1)
print("Connected to" + inst.query('*IDN?'))
`

MatthieuDartiailh, 2020-10-17:
That seems odd in particular since you are not running very recent versions of either pyvisa or pyvisa-py. I would suggest going through the usual debugging steps detailed here https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#getting-the-instrument-configuration-right, taking into account that serial instruments have some extra parameters (baud_rate, etc). I would also suggest you to update to the most recent version of pyvisa and pyvisa-py.

aboddeti, 2020-10-17:
Issue solved, updated to latest version of pyvisa.

closed 2020-10-17 by aboddeti