pyvisa/pyvisa-py #284: ASRLCOM name for serial devices and lost "com" alias

charlesbaynham, 2020-11-02:
When using ni-visa on Windows, pyvisa's `rm.list_resources()` lists devices as e.g. `ASRL3::INSTR` and can open them using `rm.open_resource("COM3")`. Using the pyvisa-py backend instead, devices are listed as `ASRLCOM3::INSTR` and cannot be opened using their COM alias.

What are these "ASRLCOM3" devices?

To Reproduce
--------------

```
Python 3.5.6 |Anaconda, Inc.| (default, Aug 26 2018, 16:05:27) [MSC v.1900 64 bi
t (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> rm = pyvisa.ResourceManager('@py')
>>> rm.open_resource("com3")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\ProgramData\Miniconda3\envs\ptb_debug\lib\site-packages\pyvisa\highle
vel.py", line 1753, in open_resource
    resource_pyclass = self._resource_classes[(info.interface_type, info.resourc
e_class)]
AttributeError: 'int' object has no attribute 'interface_type'
```

**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 63 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     D:\ProgramData\Miniconda3\envs\ptb_debug\python.exe
   Version:        3.5.6
   Compiler:       MSC v.1900 64 bit (AMD64)
   Bits:           64bit
   Build:          Aug 26 2018 16:05:27 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      #1: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.3.1
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP SOCKET: Available
      TCPIP INSTR: Available
      ASRL INSTR: Available via PySerial (3.4)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
```

Thanks in advance!

MatthieuDartiailh, 2020-11-02:
Please update to the most recent version of PyVISA and PyVISA-py in which serial resources are properly listed as ASRLx::INSTR. Note however that currently the automatic aliasing to COMx is currently not supported. You are welcome to submit a PR in that direction if you are so inclined however.

charlesbaynham, 2020-11-02:
Thanks for the help!

MatthieuDartiailh, 2020-11-02:
I am going to assume it now works and close. Feel free to reopen if needed.

closed 2020-11-02 by MatthieuDartiailh

charlesbaynham, 2020-11-02:
Yep, can confirm that it's just a version problem.