pyvisa/pyvisa-py #285: RCP sessions remotely close after 5 minutes

Handfeger, 2020-11-05:
When not using the device for 5 Minutes it closes the connection resulting in a `BrokenPipeError: [Errno 32] Broken pipe` when trying to send queries. I run this code in a Linux Docker container on a Windows host using the WSL2 backend.

To Reproduce
--------------
```python
import pyvisa as visa
import time

rm = visa.ResourceManager('@py')
raw_inst = rm.open_resource('TCPIP0::10.0.1.21::inst0::INSTR')

raw_inst.query(':OUTP:STAT?')
time.sleep(60*6)
raw_inst.query(':OUTP:STAT?')
```

I already tried to set the socket to `keep_alive` by inserting `self.sock.setsockopt( socket.SOL_SOCKET, socket.SO_KEEPALIVE, 1)` into `pyvisa-py.protocols.rcp.RawTCPClient` inside the connect method with no success.

**Output of `pyvisa-info`**
```
/usr/local/lib/python3.8/site-packages/gpib_ctypes/gpib/gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Linux-4.19.128-microsoft-standard-x86_64-with-glibc2.2.5
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/local/bin/python
   Version:        3.8.6
   Compiler:       GCC 8.3.0
   Bits:           64bit
   Build:          Oct 13 2020 20:37:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.1

Backends:
   ivi:
      Version: 1.11.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.1.0). Backend: N/A
      USB RAW: Available via PyUSB (1.1.0). Backend: N/A
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available
```

MatthieuDartiailh, 2020-11-05:
I have never observed this before. Could you try reproducing either directly on your Windows machine or on a Linux machine ? Sometimes virtual machines can behave in a surprising way.

Handfeger, 2020-11-05:
The code was working fine using a Windows machine with the National Instruments ivi adapter. I tried installing pyvisa-py on there which resulted in version conflicts for pyvisa that I couldn't solve yet. I will update you on this matter.

MatthieuDartiailh, 2020-11-05:
If you install the latest pyvisa-py, you need the latest pyvisa. Let me know if you run into more issues.

Handfeger, 2020-11-09:
Ok I testet it now on Windows and it works fine. So it is most likely related to Linux or the Virtual Machine. I still need to understand why this does not happen to all sockets. If anyone finds this Problem in future [this resource](https://tech.xing.com/a-reason-for-unexplained-connection-timeouts-on-kubernetes-docker-abd041cf7e02) might help. I will also share my fix when I have found the error finally.

MatthieuDartiailh, 2020-11-09:
Thanks for sharing your findings @Handfeger

Handfeger, 2020-11-23:
Ok so this is the fault of [Docker for Windows](https://github.com/docker/for-win/issues/2639) and there seems to be no way to fix it right now. Only way I found to fix this issue right now is to catch the broken pipe error and reestablish the connection and then to repeat the visa command.
This happens to all devices, connected to any type of socket.

MatthieuDartiailh, 2020-11-23:
Thanks for the follow up ! Hopefully it will help others.

Handfeger, 2021-01-06:
Found a way to tell Unix sockets to send keep alive packets to the client, so you never run into the 5 minute timeout. This should also bring more stability to other systems, which they didn't need.

PR #292 (2021-10-05): Fixing TCP/IP droping in from Dockercontainers
This fixes #285

I ran the tests and believe that other installations should not be affected (but not sure about windows). I am not sure if it is possible to write a test for this as you need to wait 5 minutes inside a docker container to experience this issue. I am happy to provide a test if anybody has an idea how it could be tested.

Another option to make this a option that can be accessed via highlevel function. I have found no way to access the sockets from outside. Maybe another option would be to expose the socket somehow to highlevel acce [... cut]
closed 2021-10-05 by MatthieuDartiailh