pyvisa/pyvisa-py #289: [COM] Communication issue with several instruments using py

blackbit42, 2020-11-24:
I am running a setup on a i686 system (soekris net5501) with Debian Buster and a NI PCI board with a tnt4882 chip.

Software used:
PyVISA 1.11.3 (installed via pip3)
PyVISA-py 0.5.1 (installed via pip3)
linux-gpib 4.3.3 (built on the system)

Instrument details
------------------
* Model: HP 54200A (oscilloscope)
* Communication: GPIB (address 1)
* Link to the documentation: [HP 54200A, D Operating, Programming & Service.pdf](https://doc.xdevs.com/doc/HP_Agilent_Keysight/HP%2054200A,%20D%20Operating,%20Programming%20&%20Service.pdf)

Instrument details
------------------
* Model: HP 66311A (power supply)
* Communication: GPIB (address 5)
* Link to the documentation: [HP 66311A Users guide.pdf](http://www.a0101.com/upfile/200727112146.pdf)

Instrument details
------------------
* Model: HP 54520C (oscilloscope)
* Communication: GPIB (address 7)
* Link to the documentation: [HP 545420C Progammer's Reference.pdf](https://doc.xdevs.com/doc/HP_Agilent_Keysight/HP%2054520C,%2022C,%2040C,%2042C%20Programmer.pdf)

Output of `pyvisa-info`
-----------------------
```
/usr/local/lib/python3.7/dist-packages/visa.py:23: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  FutureWarning,
Machine Details:
   Platform ID:    Linux-4.19.0-12-686-i586-with-debian-10.6
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.3')
      GPIB INTFC: Available via Linux GPIB (b'4.3.3')

```

Disclaimer: I am new to GPIB and still try to find my feet. When I run a test, only a single instrument is connected to the PCI board, simply because I only have a single GPIB cable at the moment.

I can talk to the 66311A power supply in principle. It enumerates in the pyvisa shell and reacts to certain commands. E.g., I can request the IDN string, can enable and disable the output and can print fancy text on the display.
Though, I get 'Rmt Addr Err' written on the display whenever I talk to it. Something is apparently wrong.

The 54200A scope doesn't want to talk to me at all, even though at some point I saw it in pyvisa 'list' as '( 0) GPIB0::1::INSTR'. When I try to talk to it, I see 'REM MTA' on the instrument screen and get 'gpib kernel: tnt4882: write bus error' in the kernel ring buffer.

The 54520C scope doesn't want to talk to me either, I cannot get it to enumerate it with its address. When I try to talk to it, I see 'QUERY UNTERMINATED' on the instrument screen. **Is this the smoking gun?** Apparently traffic arrives at the instrument somehow but isn't understood.

I tried various things in regard to read_termination, write_termination ('\n', '\r', '\0') and timeout as indicated by the documentation, to no avail.

Some more observations:

I get these messages in the kernel ring buffer. Not sure if that gives an indication what's wrong:
```
Nov 23 01:51:49 gpib kernel: tnt4882: read timed out
Nov 23 07:05:10 gpib kernel: tnt4882: write bus error
Nov 23 07:09:05 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:09:05 gpib kernel: tnt4882: read timed out
Nov 23 07:09:50 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:09:50 gpib kernel: tnt4882: read timed out
Nov 23 07:12:34 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:12:34 gpib kernel: tnt4882: read timed out
Nov 23 07:16:38 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:16:38 gpib kernel: tnt4882: read timed out
Nov 23 07:17:35 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:17:35 gpib kernel: tnt4882: read timed out
Nov 23 07:18:30 gpib kernel: tnt4882: read interrupted
Nov 23 07:18:30 gpib kernel: tnt4882: read interrupted
Nov 23 07:21:31 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:21:31 gpib kernel: tnt4882: read timed out
Nov 23 07:22:22 gpib kernel: tnt4882: read interrupted
Nov 23 07:22:22 gpib kernel: tnt4882: read interrupted
Nov 23 07:23:21 gpib kernel: tnt4882: minor 0 read timed out
Nov 23 07:23:21 gpib kernel: tnt4882: read timed out
```

At several occasions I have run into timeouts when a query was successfully received by the instrument and a response should have been provided, even though I have set read_termination to ''.

At this point I do not really know how to debug the problem further. Any recommendations and pointers welcome.

MatthieuDartiailh, 2020-11-24:
The write bus error is not something I have encountered before. Given that it seems that the low-level communication is meeting with some issues, I would suggest you to report directly to linux-gpib since I do not think there is anything obvious to do at pyvisa-py level.

blackbit42, 2020-11-24:
For whatever reason I can talk to the 54542C now. I haven't changed anything.
I am closing this for now.

closed 2020-11-24 by blackbit42