pyvisa/pyvisa-py #293: Unknown attribute ResourceAttribute.suppress_end_enabled (VI_ATTR_SUPPRESS_END_EN)

cpagravel, 2021-01-27:
I'm not entirely sure how I can reproduce this issue. I have an automated test that runs a lot of extra layers and I'm not able to reproduce the issue outside of using the device in the test. 

I'm using an ITECH- IT7321 device. It has a USB port for communication.

To Reproduce
--------------
Steps to reproduce the behavior:

I started debugging this issue but I haven't found a great explanation of what VI_ATTR_SUPPRESS_END_EN is. Could someone provide some details about this? It might help me isolate the issue to make something reproducible.

In my code I have a separate thread polling three measurements from the device. This occurs every 0.1 seconds.

`instr.query(":MEASURE:CURRENT:AC?;:MEASURE:VOLTAGE:AC?;:MEASURE:POWER:AC?")`

In another thread I am running additional threads for communicating with serial devices unrelated to pyvisa.

The error I have is below:

```
01-26 21:00:06.748 ERROR Unknown attribute ResourceAttribute.suppress_end_enabled (0x3fff0036 - VI_ATTR_SUPPRESS_END_EN)
Traceback (most recent call last):
  File "/home/gtvchrome/.virtualenvs/p3test/lib/python3.7/site-packages/pyvisa_py/sessions.py", line 669, in get_attribute
    return self._get_attribute(attribute)
  File "/home/gtvchrome/.virtualenvs/p3test/lib/python3.7/site-packages/pyvisa_py/usb.py", line 207, in _get_attribute
    raise UnknownAttribute(attribute)
pyvisa_py.sessions.UnknownAttribute: Unknown attribute ResourceAttribute.suppress_end_enabled (0x3fff0036 - VI_ATTR_SUPPRESS_END_EN)
```

**Output of `pyvisa-info`**
Note I tried downloading and compiling libusb1 from source to get the newest version in case that may help.
```
Machine Details:
   Platform ID:    Linux-4.19.0-13-amd64-x86_64-with-debian-10.7
   Processor:      

Python:
   Implementation: CPython
   Executable:     /home/gtvchrome/.virtualenvs/p3test/bin/python
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           64bit
   Build:          Jul 25 2020 13:03:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.1.0). Backend: libusb1
      USB RAW: Available via PyUSB (1.1.0). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

MatthieuDartiailh, 2021-01-27:
This attribute appears to be accessed in `read` but has not been implemented for the resource. Looks like I made a mistake. Can you add `"SUPPRESS_END_EN"` to the list at https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/usb.py#L87, alternatively you could comment out the line https://github.com/pyvisa/pyvisa-py/blob/main/pyvisa_py/usb.py#L141 and the following ones ? If you can confirm the first fix works for you I would love to get a PR  for it.

cpagravel, 2021-01-27:
Thanks for the quick response. I'll try adding `"SUPPRESS_END_EN"` to the list you mentioned on L87 and re-running.

cpagravel, 2021-01-27:
The first fix seemed to work, thanks for sparing the time! I'd be happy to throw up a PR.

PR #294 (2021-01-29): fix #293 - Handle SUPPRESS_END_EN in usb.py
Add SUPPRESS_END_EN to list of attributes handled in USBSessions

- [x] Closes #293  (insert issue number if relevant)
- [o] Executed ``black . && isort -c . && flake8`` with no errors
Passes black and flake8. `isort -c` has issue:
ERROR: pyvisa-py/docs/source/conf.py Imports are incorrectly sorted.
ERROR: pyvisa-py/pyvisa_py/__init__.py Imports are incorrectly sorted.
Not fixing this because pre-commit msg requests to limit scope of change
- [ ] The change is fully covered by automated unit tests
Unit tests are in pyvisa repo. Though none of them co [... cut]
commit 0f7dd7c5 (2021-01-29): Merge pull request #294 from cpagravel/u/chrisgravel/add_supress_end_en

fix #293 - Handle SUPPRESS_END_EN in usb.py
commit e0f9020f (2021-01-29): fix #293 - Handle SUPPRESS_END_EN in usb.py
closed 2021-01-29 by MatthieuDartiailh