pyvisa/pyvisa-py #300: [COM] Communication issue with Tektronix MSO64B using query_binary_values

rainswor, 2021-03-31:
I'm using the fastframe feature from tektronix and I'd like to read all the frames using CURVE?

I have the scope setup to read 10000 points and frames 1 to 750.

>>> scope.query('data?')
'REF1;RIBINARY;CH1;1;10000;1;750;1;1\n'

If i do

>>> scope.write('CURVE?')
8
>>> raw_data=scope.read_raw()
>>> len(raw_data)
7506000

Each frame is 10k bytes +7 bytes for the header +1byte for the frame separator (';'). So 10008*750=7506000

However, when I do 

>>> data=mitek.scope.query_binary_values('CURVE?',datatype='b')
>>> len(data)
10000

Is there a way for query_binary_values to grab the full output of CURVE? It seems to just grab the first frame. If you are wondering why I am not just using read_raw. It is also giving me strange results. The raw data that comes back for some of the frames seems to have gibberish in between bytes e.g.

\x03\x03\x03\x03\x03\x03\x03\x03\x04\x05\x05\x06\x07\x08\t\t\t\x08\x08\x07\x07\x08\x08\x08\x07\x07\x07\x07\x07\x07\x07\x06\x06\x06\x06\x06\x07\x07\x07\x06\x06\x06\x06\x07\x07\x07\x07\x07\x08\x08\t\n\x0b\x0b\x0b\n\n\n\n\n\t\t\t\t\t\t\n\x0b\x0c\x0c\x0c\r\x0f\x11\x12\x14\x15\x15\x16\x18\x1a\x1b\x1d\x1e\x1e\x1f !#$%&&\'(()*+,,----./012345679:;<==========>?@AAAABCDDDDDDDEFFFFFEEEEEDCA@@??>=<;:::98653333445555677888889:;;;;:998764310.-+)\'%" \x1c\x18\x15\x13\x12\x11\x10\x0f\x0e\r\x0c\x0b\x0b\n\t\x08\x08\x07\x07\x07\x07\x07\x07\x07\x07\x07\x07\x07\x07\x07

Instrument details
------------------
* Model: MSO64B
* Communication: TCPIP
* Link to the documentation (if available):

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Linux-3.10.0-1160.11.1.el7.x86_64-x86_64-with-redhat-7.9-Nitrogen
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.6.8
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-44)
   Bits:           64bit
   Build:          Nov 10 2020 07:30:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

MatthieuDartiailh, 2021-03-31:
query_binary_values is tailored to return a single binary block if in your case you need multiple such blocks and you know exactly how much data you expect you can use read_bytes and manually convert your data. Also I do not thing you get gibberish it is just that bytes in the ASCII range are likely represented by their code in the ASCII table.

rainswor, 2021-03-31:
Thank you! You comment gave me the extra understanding to fix my issue. I was trying to use .split(';') as this is the frame separator. This didn't work as sometimes other values were represented by ';'. Counting the bytes instead fixed my problem and I either use read_bytes as you said or just used read_raw and slice in size blocks I need.

MatthieuDartiailh, 2021-04-01:
I would advise against using read_raw since it will stop if it encounters the termination character which can exist in binary blocks. read_bytes does not suffer from the same issue (save if you explicitly ask for it).