pyvisa/pyvisa-py #301: VISA connections using ::SOCKET

byzantic, 2021-04-19:
I'm having problems connecting to instruments when using VISA SOCKET connection strings. INSTR type connection strings are OK, but I have one instrument that will only connect via SOCKET.

So if I connect to myy nice shiny new Keithley 6500 using INSTR, all is fine:
```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> inst = rm.open_resource("TCPIP::10.1.110.148::INSTR")
>>> inst.query("*IDN?")
'KEITHLEY INSTRUMENTS,MODEL DMM6500,04493036,1.7.3c\n'
```
But if I try using SOCKET, I just get a timeout:
```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> inst = rm.open_resource("TCPIP::10.1.110.148::5025::SOCKET")
>>> inst.query("*IDN?")
...
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```
**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Linux-5.4.0-72-lowlatency-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.5
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Jan 27 2021 15:41:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
```

MatthieuDartiailh, 2021-04-19:
You probably need to set the proper termination characters. The following is a reasonable starting point and may be enough:
```python
inst.write_termination = "\n"
inst.read_termination = "\n"
```

byzantic, 2021-04-19:
Fantastic! Thanks for your very rapid response.

I can confirm that fixes the issue reported here .. i.e. this wasn't actually an issue with pyvisa at all.

Sigh. Now the only problem is that I have to find out how to force those settings down from the PyMeasure adapter layer .. but that's another story.

Closing this because it wasn't really an issue.

closed 2021-04-19 by byzantic

MatthieuDartiailh, 2021-04-19:
You are welcome